Koenkk/zigbee2mqtt #2095: zigbee2mqtt stops running after days or hours

domo86, 2019-10-07:
Hello,
I have installed Zigbee ( 6.1.0 )  and CC2531  ( firmware 20190608)  ) Stack_Home_1.2 Default.
3 sensor Aqara + 2 temperature sensor Aqara.

It works like a charm, BUT after some hours, but usually after 1 or 2 days, the sensors are not taken in account by the system.

There is no bad logs generated by zigbee2mqtt, it just stops...

Stopping and restarting zigbee2mqtt is not enough (error). it is needed to unplug/replug then reset the CC2531, then stop/start zigbee2mqtt.
then, it runs till the next time.....

I have to add that the Raspberry has a UPS Pico HAT, but I don' t think it can affect it.

If some of you could help me on this, because I found nothing on theweb.

Koenkk, 2019-10-09:
Can you run zigbee2mqtt with `DEBUG=* npm start` and provide the logging when it stops working?

DubhAd, 2019-10-09:
FWIW I had/have the [same problem](https://github.com/Koenkk/Z-Stack-firmware/issues/105) that was mitigated through disabling the graphing. I have re-enabled availability and reporting, and I don't get a hard lock up like that any more.

domo86, 2019-10-09:
Thnaks DubhAd, what do you mean by "disabling the graphing" ?

domo86, 2019-10-09:
Koenkk,
I have done: cd /opt/zigbee2mqtt ( where the package.json is )
then DEBUG=* npm start, but I have the following errors:

`pi@HomeAss:/opt/zigbee2mqtt $ DEBUG=* npm start

> zigbee2mqtt@1.6.0 start /opt/zigbee2mqtt
> node index.js

  serialport/bindings loading LinuxBinding +0ms
  zigbee2mqtt:info 10/9/2019, 6:38:21 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-10-09.18-38-19'
  zigbee2mqtt:debug 10/9/2019, 6:38:21 PM Removing old log directory '/opt/zigbee2mqtt/data/log/2019-10-07.19-40-53'
  zigbee2mqtt:debug 10/9/2019, 6:38:21 PM Loaded state from file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:debug 10/9/2019, 6:38:21 PM Saving state to file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:info 10/9/2019, 6:38:21 PM Starting zigbee2mqtt version 1.6.0 (commit #e26ad2a)
  zigbee2mqtt:info 10/9/2019, 6:38:21 PM Starting zigbee-shepherd
  zigbee2mqtt:debug 10/9/2019, 6:38:21 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/opt/zigbee2mqtt/data/database.db","coordBackupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  zigbee-herdsman:znp:log Opening with /dev/ttyACM0 and {"baudRate":115200,"rtscts":true} +0ms
  serialport/stream opening path: /dev/ttyACM0 +0ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +10ms
  serialport/stream Binding #open had an error [Error: Error Resource temporarily unavailable Cannot lock port] +15ms
  serialport/stream close attempted, but port is not open +14ms
events.js:174
      throw er; // Unhandled 'error' event
      ^

Error: Port is not open
    at SerialPort.close (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/stream.js:425:29)
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/znp/znp.js:107:37)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/znp/znp.js:7:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/znp/znp.js:3:12)
    at SerialPort.serialPort.open (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/znp/znp.js:103:45)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/stream.js:197:14)
    at binding.open.then.err (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/stream.js:241:12)
Emitted 'error' event at:
    at errorOrDestroy (internal/streams/destroy.js:107:12)
    at SerialPort.onerror (_stream_readable.js:733:7)
    at SerialPort.emit (events.js:198:13)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/stream.js:199:10)
    at process.nextTick (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/stream.js:204:31)
    at process._tickCallback (internal/process/next_tick.js:61:11)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-10-09T16_38_21_556Z-debug.log
`

domo86, 2019-10-09:
Hello Koenkk,
This is what I see ( but I'm not sure that the origin issus is the same, but the effect is the same)

  zigbee2mqtt:info 10/9/2019, 6:45:13 PM MQTT publish: topic 'homeassistant/sensor/0x00158d0004240bc7/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/baie_sud","json_attributes_topic":"zigbee2mqtt/baie_sud","name":"baie_sud_linkquality","unique_id":"0x00158d0004240bc7_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0004240bc7"],"name":"baie_sud","sw_version":"Zigbee2mqtt 1.6.0","model":"Aqara door & window contact sensor (MCCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 10/9/2019, 6:45:13 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"e26ad2a","coordinator":20190608,"log_level":"debug","permit_join":false}'
  serialport/bindings/poller error [Error: bad file descriptor] +3m
  serialport/stream binding.read error [Error: bad file descriptor] +3m
  serialport/stream disconnected [Error: bad file descriptor] +4ms
  serialport/stream #close +4ms
  serialport/binding-abstract close +3m
  serialport/stream _read queueing _read for after open +2ms
  serialport/bindings/poller Stopping poller +28ms
  serialport/bindings/poller Destroying poller +2ms
  serialport/stream binding.close finished +5ms
  zigbee-herdsman:znp:log Serialport closed +3m

I would like to launch the DEBUG=* npm start as a deamon, how can I do and where will be the output log?

00punisher, 2019-10-16:
Hi,

I have upgraded zigbee2mqtt to v1.6.0 and CC2531 coordinator to firmware 20190608 Stack_Home_1.2 Default, running in a Raspberry 3b.

I have 3 CC2531 routers, several Xiaomi sensors, 2 Hue bulbs with their respective dimmer controls.

Everything works fine, but after some hours/days, routers turn offline and sensors and others stop working. No info in the logs.

I have to stop zigbee2mqtt, unplug and plug the CC2531 coordinator and start z2m again.

Up to the upgrade, I hadn't had this kind of issue. It was really stable. Nothing else changed.

I run into this thread and it seems the same issue. I will try the debug option, but I don't know when it stops working again.

Any help/suggestion would be highly appreciated. 

Thanks!

Koenkk, 2019-10-16:
Output with debug should help to pinpoint the issue.

00punisher, 2019-10-16:
Just enabled debug logging for zigbee2mqtt in configuration.yaml. Anything else while waiting it fails?

domo86, 2019-10-22:
Just for information, my problem seems solved only by changing the power supply ( 2.5A instead of 2 A ). Thanks for your support.

00punisher, 2019-10-24:
Hi, 
So far so good. Anyway I checked the power supply and it is 3A.
I will keep you posted just in case.
Thanks!

00punisher, 2019-11-12:
Hi guys,

No more issues/problems so far. It is true that I have restarted Z2M fewer times than before. 

On the other hand, some routers (cc2531 and cc2530) with the latest firmware stopped suddenly after some days working perfectly. I remove them and re-pair them and they start working again, but I think it has nothing to do with this.

Thanks!

00punisher, 2019-11-13:
Hi all,

Sorry to say that 40min ago, my system notified via Telegram that Z2M stopped.

I will check logs once at home this evening and let you know.

00punisher, 2019-11-13:
Hi,

The log has no error. Simply routers became offline suddenly. All of them and at the same time. Here it is the exact moment:

[...]
11/13/2019, 11:00:48 AM - debug: Ping 0x00158d0002168aec (basic)
11/13/2019, 11:00:48 AM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1}}' of device 'lumi.plug' (0x00158d0002168aec) of endpoint 1
11/13/2019, 11:00:48 AM - debug: Successfully pinged 'Enchufe1'
11/13/2019, 11:00:48 AM - info: MQTT publish: topic 'zigbee2mqtt/Enchufe1/availability', payload 'online'
11/13/2019, 11:00:53 AM - debug: Ping 0xccccccfffe3a7ef0 (default)
11/13/2019, 11:01:01 AM - debug: Ping 0x00124b0018e1a916 (default)
11/13/2019, 11:01:03 AM - debug: Failed to ping 'Ikea1'
11/13/2019, 11:01:03 AM - info: MQTT publish: topic 'zigbee2mqtt/Ikea1/availability', payload 'offline'
[...]

:-(

00punisher, 2019-11-14:
BTW, to get it back working I had to stop the service, unplug and plug the C2531 coordinator and start Z2M again.

Koenkk, 2019-11-15:
I would recommend disabling the availability feature (it could make the stick crash).

regissg, 2019-12-13:
Hello. I have the exact same problem with zigbee2mqtt that stops running after days or hours of working perfectly well and the need to unplug/replug CC2531 and restart zigbee2mqtt to get out of that situation.
I have activated the Debug mode to provide more details but what should I do to disable the availability feature?

regissg, 2020-01-10:
All has worked well for the past 28 days but this morning I had again the issue of zigbee2mqtt not working. I had again to restart the zigbee2mqtt service and reboot the CC2531.

Here is the log extract I have with the "log_level: debug"

```debug 2020-01-10T07:39:39: Received Zigbee message from 'salon/temp', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":5103}' from endpoint 1 with groupID 0
info  2020-01-10T07:39:39: MQTT publish: topic 'zigbee2mqtt/salon/temp/battery', payload '86'
debug 2020-01-10T07:39:39: Received Zigbee message from 'salon/temp', type 'attributeReport', cluster 'msPressureMeasurement', data '{"16":10052,"20":-1,"measuredValue":1005}' from endpoint 1 with groupID 0
info  2020-01-10T07:39:39: MQTT publish: topic 'zigbee2mqtt/salon/temp/battery', payload '86'
info  2020-01-10T07:39:39: MQTT publish: topic 'zigbee2mqtt/salon/temp/voltage', payload '2975'
info  2020-01-10T07:39:39: MQTT publish: topic 'zigbee2mqtt/salon/temp/voltage', payload '2975'
info  2020-01-10T07:39:39: MQTT publish: topic 'zigbee2mqtt/salon/temp/temperature', payload '20.35'
info  2020-01-10T07:39:39: MQTT publish: topic 'zigbee2mqtt/salon/temp/temperature', payload '20.35'
info  2020-01-10T07:39:39: MQTT publish: topic 'zigbee2mqtt/salon/temp/humidity', payload '51.03'
info  2020-01-10T07:39:39: MQTT publish: topic 'zigbee2mqtt/salon/temp/humidity', payload '51.03'
info  2020-01-10T07:39:39: MQTT publish: topic 'zigbee2mqtt/salon/temp/pressure', payload '1004'
info  2020-01-10T07:39:39: MQTT publish: topic 'zigbee2mqtt/salon/temp/pressure', payload '1005'
info  2020-01-10T07:39:39: MQTT publish: topic 'zigbee2mqtt/salon/temp/linkquality', payload '31'
info  2020-01-10T07:39:39: MQTT publish: topic 'zigbee2mqtt/salon/temp/linkquality', payload '31'
debug 2020-01-10T07:39:40: Received MQTT message on 'zigbee2mqtt/Prise/set' with data 'OFF'
debug 2020-01-10T07:39:40: Publishing 'set' 'state' to 'Prise'
info  2020-01-10T07:39:41: MQTT publish: topic 'zigbee2mqtt/Prise/state', payload 'OFF'
debug 2020-01-10T07:40:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T07:45:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T07:50:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T07:55:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T08:00:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T08:05:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T08:10:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T08:15:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T08:20:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T08:25:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T08:30:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T08:35:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T08:40:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T08:45:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T08:50:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T08:55:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T09:00:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T09:05:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T09:10:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T09:15:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T09:20:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T09:25:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T09:30:30: Saving state to file C:\Program Files\zigbee2mqtt\data\state.json
debug 2020-01-10T09:32:48: Received MQTT message on 'zigbee2mqtt/Biblio/set/brightness' with data '255.0'
debug 2020-01-10T09:32:48: Publishing 'set' 'brightness' to 'Biblio'
debug 2020-01-10T09:32:52: Received MQTT message on 'zigbee2mqtt/Biblio/set/brightness' with data '0.0'
debug 2020-01-10T09:32:52: Publishing 'set' 'brightness' to 'Biblio'
error 2020-01-10T09:32:54: Publish 'set' 'brightness' to 'Biblio' failed: 'Error: SRSP - AF - dataRequest after 6000ms'
info  2020-01-10T09:32:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'brightness' to 'Biblio' failed: 'Error: SRSP - AF - dataRequest after 6000ms'","meta":{"friendly_name":"Biblio"}}'
error 2020-01-10T09:33:00: Publish 'set' 'brightness' to 'Biblio' failed: 'Error: SRSP - AF - dataRequest after 6000ms'
info  2020-01-10T09:33:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'brightness' to 'Biblio' failed: 'Error: SRSP - AF - dataRequest after 6000ms'","meta":{"friendly_name":"Biblio"}}'```

all worked ok up to 9:25 and then zigbee2mqtt was not responsive to any zigbee button action (not even logging error for them). However zigbee2mqtt was responsive to mqtt orders coming from Alexa via Openhab (see at 9:32:48). Even though zigbee2mqtt logged an error at 09:32:54 the action to set brightness to 255 did work ok.

I have also attached the state.json file but it seems that the file is overwritten every 5mn so it does not show its state at the time zigbee2mqtt was not working but after I restarted the service
[state.zip](https://github.com/Koenkk/zigbee2mqtt/files/4045015/state.zip)

Koenkk, 2020-01-11:
It seems that the stick has crashed, I would recommend switching to https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing which improves stability

closed 2020-03-18 by stale[bot]

regissg, 2020-03-23:
Hello, I keep having from time to time (could be weeks) the same problem where zigbee2mqtt is not working and where I need to unplug/plug the CC2531 and then restart the zigbee2mqtt service on windows.
The issue is blocking for me as I now use zigbee2mqtt to control my home heating system.

The CC2531 is running the 20190619 version and my windows 10 PC is running zigbee2mqtt version 1.7.1.
The CC2531 is connected to a powered usb hub with a 3 meter cable to the PC.

Am I the only one to have this stability problem and is there a way to recover it automatically (ie detect the failure situation and restart cc2531 and zigbee2mqtt service without human intervention)?