Koenkk/zigbee2mqtt #2098: A '/' slash is no longer accepted in friendly_name

sjorge, 2019-10-08:
I'm running dev#8d97262d39022067e22ac199f18e87f8d2c38e44, and I can no longer rename a device to something with a '/' in the friendly_name.

I had all my devices named 'room/device_name' so it was easy to subscribe to a single room via mqtt. Now renaming fails when it includes a '/' in the friendly_name.

I wanted to replace a trådfri bulb with a hue one to see if I like them more, but wanted to give the hue bulb the same name the old trådfri bulb had before I removed it from the network.

sjorge, 2019-10-08:
Update 2: Found the actual issue!

So I removed the lamp via the remove command, it worked and it was removed from the database and no updates came when I used the lamp normally. (So it was out of the network)

However it was not removed from configuration.yaml so there was a duplicate friendly_name, the failure message was not super clear.

Perhaps also if the intent to leave the device in configuration.yaml, the friendly_name should be reset to the ieeeaddr?

Koenkk, 2019-10-09:
We should indeed discuss if a device should be removed from `configuration.yaml` in case the device is removed from the network. 

It would be nice if others could also give their input regarding this (as I think there are pros/cons for both)

sjorge, 2019-10-09:
At least the error message should maybe make it more clear that the rename failed because of a duplicate name. As that was unclear and also the device didn;t show up in the devices list, so that was not something that was imediatly obvious.

DubhAd, 2019-10-09:
I think it would be useful to have an option in the remove to change the behaviour. The question is whether the default should be to clean up, or not clean up.

Alternatively in OpenZWave you can _Remove_ (which unpairs and removes the node) or _Replace_ (which unpairs the old device and replaces it - potentially with the same device). Maybe similar logic here would work?

carldebilly, 2019-10-10:
@DubhAd The _ZWave replace_ is a feature of the Z-Wave protocol.  I think it's not possible to do the equivalent in Zigbee.

@Koenkk Maybe when we remove a device using the `zigbee2mqtt/bridge/config/remove` we should have a way to specify something like: `{friendly_name:"my device", cleanup: true}`

DubhAd, 2019-10-11:
@carldebilly 
> The _ZWave replace_ is a feature of the Z-Wave protocol. I think it's not possible to do the equivalent in Zigbee

Doesn't mean that it's not possible to replicate the _user experience_ though

sjorge, 2019-10-11:
Which was also more or less what I was trying to do manually, as I was used to swapping out my test stuff via open-zwave in the same way.

Koenkk, 2019-10-11:
What about this:

- Add an extra attribute `disabled` (default `false`)
```
devices:
  '0x00158d0001d82999':
    disabled: true
    friendly_name: 'my_occupancy_sensor'
    retain: true
```
- When `disabled` it is treated like it's not in `configuration.yaml` (so no hass discovery; won't respond to MQTT commands, nothing is published)
- Upon removal we automatically add this attribute 

This should give us best of both worlds:
- The MQTT removal api stays the same
- Configuration is kept (in case you want to enable it later)
- `friendly_name` can be re-used

Koenkk, 2019-10-11:
In that case zigbee2mqtt should refuse to start or not rename it via mqtt (as it should guarantee that friendly names of all enabled entities (devices/groups) are unique).

sjorge, 2019-10-11:
Sounds good.

carldebilly, 2019-10-11:
@Koenkk That's a good direction.  We'll need:

- Add a `disabled` information from result of `zigbee2mqtt/bridge/config/devices/get`
- Add a topic to enable & disable a device (can we disable a device when it is up and connected to network?)

My goal is to be able to handle those disabled devices in Zigbee2MqttAssistant :-)

sjorge, 2019-10-11:
I would assume to 'enable' one again, one would need to allow joining and have the device send some network traffic.

Simply toggling enable/disable is not going to work for devices that properly leave a network.

sjorge, 2019-12-10:
I think with force_remove, this is now resolved.

closed 2019-12-10 by sjorge