Koenkk/zigbee2mqtt #2099: Sudo chown (Installing Zigbee2mqtt)

dthporro, 2019-10-08:
Hi people, i'm following this guide to get working zigbee2mqtt:
https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html
I'm already in step 2 "installing" but i get an error when writing:

sudo chown -R pi:pi /opt/zigbee2mqtt

and i get

chown: invalid user: «pi:pi»

Found an old post with the same problem, but didn't found any solution. I know it's something about MY user, maybe should i replace "pi:pi" for my user and group? but i dont know any of them.

The old post with the same problem.
https://github.com/Koenkk/zigbee2mqtt/issues/987

Thank you!

didiht, 2019-10-09:
Hi,

can you try:

sudo chown -R pi /opt/zigbee2mqtt

and

sudo chgrp -R pi /opt/zigbee2mqtt

What happens?

dthporro, 2019-10-09:
ok thank you for your answer, but i passed the night trying things and finally the solution was to write:

sudo chown -R root:root /opt/zigbee2mqtt

instead of:

sudo chown -R pi:pi /opt/zigbee2mqtt

Until there everithing is ok, but now i have another problem.

I've paired 1 device (osram plug) to zigbee2mqtt:

Oct 09 11:49:50 hassbian npm[12152]:   zigbee2mqtt:info 10/9/2019, 11:49:50 AM New device 'Plug 01' with address 0x7cb03eaa00b0c107 connected!
Oct 09 11:49:50 hassbian npm[12152]:   zigbee2mqtt:info 10/9/2019, 11:49:50 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x7cb03eaa00b0c107","meta":{"modelID":"Plug 01"}}'
Oct 09 11:49:51 hassbian npm[12152]:   zigbee2mqtt:info 10/9/2019, 11:49:51 AM Successfully configured 0x7cb03eaa00b0c107 (0x7cb03eaa00b0c107)
Oct 09 11:49:52 hassbian npm[12152]:   zigbee2mqtt:info 10/9/2019, 11:49:52 AM MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa00b0c107', payload '{"state":"ON","linkquality":18}'
Oct 09 11:49:56 hassbian npm[12152]:   zigbee2mqtt:info 10/9/2019, 11:49:56 AM MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa00b0c107', payload '{"state":"ON","linkquality":5}'
Oct 09 11:49:58 hassbian npm[12152]:   zigbee2mqtt:info 10/9/2019, 11:49:58 AM Device incoming...
Oct 09 11:49:58 hassbian npm[12152]:   zigbee2mqtt:info 10/9/2019, 11:49:58 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'

But, i can't see it in home assistant, i'm a little lost there, because i'm following this guide:

https://www.zigbee2mqtt.io/integration/home_assistant

And the part of editing zigbee2mqtt configuration.yaml it's done, but i can't find the path to edit home assistant configuration.yaml. Will keep looking for it, i think that is what i am missing.

closed 2019-12-15 by stale[bot]