Koenkk/zigbee2mqtt #2100: Problems binding hue dimmer to hue bulb

sjorge, 2019-10-08:
# Bug Report
## What happened

I tried (and succeeded?) to bind my new hue dimmer to a new hue bulb... this is logged:
```
zigbee2mqtt:info  2019-10-08T16:00:41: Successfully bound cluster 'genScenes' from 'bedroom/remote_desk_lamp' to 'bedroom/desk_lamp'
zigbee2mqtt:info  2019-10-08T16:00:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_bind","message":{"from":"bedroom/remote_desk_lamp","to":"bedroom/desk_lamp","cluster":"genScenes"}}'
zigbee2mqtt:info  2019-10-08T16:00:42: Successfully bound cluster 'genOnOff' from 'bedroom/remote_desk_lamp' to 'bedroom/desk_lamp'
zigbee2mqtt:info  2019-10-08T16:00:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_bind","message":{"from":"bedroom/remote_desk_lamp","to":"bedroom/desk_lamp","cluster":"genOnOff"}}'
zigbee2mqtt:info  2019-10-08T16:00:42: Successfully bound cluster 'genLevelCtrl' from 'bedroom/remote_desk_lamp' to 'bedroom/desk_lamp'
zigbee2mqtt:info  2019-10-08T16:00:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_bind","message":{"from":"bedroom/remote_desk_lamp","to":"bedroom/desk_lamp","cluster":"genLevelCtrl"}}'
```

So it looks successful, however clicking buttons on the dimmer remote doesn't do anything.

## What did you expect to happen
Clicking a button to turn on/off or change the brightness.

## How to reproduce it (minimal and precise)
1. join a hue bulb (in this case a E14 RGB candle)
2. join a hue dimmer
3. bind the bulb to the dimmer

## Debug Info
zigbee2mqtt version:  `{"version":"1.6.0","commit":"8d97262"}`
CC253X firmware version: `{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190425}`

sjorge, 2019-10-08:
In the original bind issue it mentions that I probably need to unbind the coordinator from the hue dimmer.

I get ```zigbee2mqtt:error 2019-10-08T16:08:17: Failed to call 'DeviceBind' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Target is unknown
    at DeviceBind.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/deviceBind.js:32:9)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:320:44)
    at process._tickCallback (internal/process/next_tick.js:68:7))```

When using ieeeaddr from the devices list:
```json
{"ieeeAddr":"0x00124b001938a7e5","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190425","lastSeen":1570550791789}
```
I tried both '0x00124b001938a7e5' and 'Coordinator' as payload.

Koenkk, 2019-10-09:
The correct payload is `coordinator` (with lower `c`). Anyway `Coordinator` is also allowed in the latest dev branch.

sjorge, 2019-10-09:
I just copied the friendly_name as as, which had it with a capital C.

The flow is... unbind coordinator, bind bulb?... will test once I get home.

sjorge, 2019-10-09:
This worked fine (I did not update yet, but using the lower c)!

@Koenkk It looks like I still see the button press events in zigbee2mqtt even with the coordinator unbound... is there a way we do not automatically unbind the coordinator after joining?

closed 2019-10-09 by sjorge

Koenkk, 2019-10-09:
I think without it it doesnt work but need to check

gelsas, 2020-04-24:
Hello,
I tried to bind my hue dimmer to my hue bulb but get these errors:

```
[15:48:40] openhabian@openhab:~$  mosquitto_pub -t zigbee2mqtt/bridge/bind/HueDimmerSwitchWandlampe -m HueRGBWandlampe
[15:49:19] openhabian@openhab:~$ sudo journalctl -u zigbee2mqtt.service -f
-- Logs begin at Thu 2019-02-14 11:11:58 CET. --
Apr 24 15:49:29 openhab npm[27796]: zigbee2mqtt:error 2020-04-24 15:49:29: Failed to bind cluster 'genScenes' from 'HueDimmerSwitchWandlampe' to 'HueRGBWandlampe' (Error: Bind 0x00178801020e4808/1 genScenes from '0x001788010278badc/11' failed (Error: AREQ - ZDO - bindRsp after 10000ms))
Apr 24 15:49:29 openhab npm[27796]: zigbee2mqtt:info  2020-04-24 15:49:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_bind_failed","message":{"from":"HueDimmerSwitchWandlampe","to":"HueRGBWandlampe","cluster":"genScenes"}}'
Apr 24 15:49:39 openhab npm[27796]: zigbee2mqtt:error 2020-04-24 15:49:39: Failed to bind cluster 'genOnOff' from 'HueDimmerSwitchWandlampe' to 'HueRGBWandlampe' (Error: Bind 0x00178801020e4808/1 genOnOff from '0x001788010278badc/11' failed (Error: AREQ - ZDO - bindRsp after 10000ms))
Apr 24 15:49:39 openhab npm[27796]: zigbee2mqtt:info  2020-04-24 15:49:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_bind_failed","message":{"from":"HueDimmerSwitchWandlampe","to":"HueRGBWandlampe","cluster":"genOnOff"}}'
Apr 24 15:49:49 openhab npm[27796]: zigbee2mqtt:error 2020-04-24 15:49:49: Failed to bind cluster 'genLevelCtrl' from 'HueDimmerSwitchWandlampe' to 'HueRGBWandlampe' (Error: Bind 0x00178801020e4808/1 genLevelCtrl from '0x001788010278badc/11' failed (Error: AREQ - ZDO - bindRsp after 10000ms))
Apr 24 15:49:49 openhab npm[27796]: zigbee2mqtt:info  2020-04-24 15:49:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_bind_failed","message":{"from":"HueDimmerSwitchWandlampe","to":"HueRGBWandlampe","cluster":"genLevelCtrl"}}'

[15:58:22] openhabian@openhab:~$ mosquitto_pub -t zigbee2mqtt/bridge/unbind/HueDimmerSwitchWandlampe -m coordinator
[15:58:26] openhabian@openhab:~$ sudo journalctl -u zigbee2mqtt.service -f
-- Logs begin at Thu 2019-02-14 11:11:58 CET. --
Apr 24 15:58:36 openhab npm[27796]: zigbee2mqtt:error 2020-04-24 15:58:36: Failed to unbind cluster 'genScenes' from 'HueDimmerSwitchWandlampe' to 'Coordinator' (Error: Unbind 0x00178801020e4808/1 genScenes from '0x00124b001ca1aefd/1' failed (Error: AREQ - ZDO - unbindRsp after 10000ms))
Apr 24 15:58:36 openhab npm[27796]: zigbee2mqtt:info  2020-04-24 15:58:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_unbind_failed","message":{"from":"HueDimmerSwitchWandlampe","to":"Coordinator","cluster":"genScenes"}}'
Apr 24 15:58:46 openhab npm[27796]: zigbee2mqtt:error 2020-04-24 15:58:46: Failed to unbind cluster 'genOnOff' from 'HueDimmerSwitchWandlampe' to 'Coordinator' (Error: Unbind 0x00178801020e4808/1 genOnOff from '0x00124b001ca1aefd/1' failed (Error: AREQ - ZDO - unbindRsp after 10000ms))
Apr 24 15:58:46 openhab npm[27796]: zigbee2mqtt:info  2020-04-24 15:58:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_unbind_failed","message":{"from":"HueDimmerSwitchWandlampe","to":"Coordinator","cluster":"genOnOff"}}'
Apr 24 15:58:56 openhab npm[27796]: zigbee2mqtt:error 2020-04-24 15:58:56: Failed to unbind cluster 'genLevelCtrl' from 'HueDimmerSwitchWandlampe' to 'Coordinator' (Error: Unbind 0x00178801020e4808/1 genLevelCtrl from '0x00124b001ca1aefd/1' failed (Error: AREQ - ZDO - unbindRsp after 10000ms))
Apr 24 15:58:56 openhab npm[27796]: zigbee2mqtt:info  2020-04-24 15:58:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_unbind_failed","message":{"from":"HueDimmerSwitchWandlampe","to":"Coordinator","cluster":"genLevelCtrl"}}'

```

martdj, 2022-03-13:
I know this is an old thread, but I'm facing the same problem. I've successfully bound my Hue dimmer to a group which contains 4 Hue lamps. However, when I press any button, nothing happens. I do see the dimmer commands come by in the log.
info  2022-03-13 12:20:47: Successfully bound cluster 'genLevelCtrl' from 'Tafellamp Dimmer' to 'Eetkamerlamp'
info  2022-03-13 12:21:14: Successfully bound cluster 'genScenes' from 'Tafellamp Dimmer' to 'Eetkamerlamp'
info  2022-03-13 12:21:14: Successfully bound cluster 'genOnOff' from 'Tafellamp Dimmer' to 'Eetkamerlamp'
info  2022-03-13 12:21:15: Successfully bound cluster 'genLevelCtrl' from 'Tafellamp Dimmer' to 'Eetkamerlamp'

info  2022-03-13 12:32:10: MQTT publish: topic 'zigbee2mqtt/Tafellamp Dimmer', payload '{"action":"off_press","action_duration":null,"battery":17.5,"linkquality":116,"update":{"state":"idle"},"update_available":null}'
info  2022-03-13 12:32:10: MQTT publish: topic 'zigbee2mqtt/Tafellamp Dimmer', payload '{"action":"","action_duration":null,"battery":17.5,"linkquality":116,"update":{"state":"idle"},"update_available":null}'
info  2022-03-13 12:32:10: MQTT publish: topic 'zigbee2mqtt/Tafellamp Dimmer/action', payload 'off_press'
info  2022-03-13 12:32:10: MQTT publish: topic 'zigbee2mqtt/Tafellamp Dimmer', payload '{"action":"off_press_release","action_duration":null,"battery":17.5,"linkquality":112,"update":{"state":"idle"},"update_available":null}'
info  2022-03-13 12:32:10: MQTT publish: topic 'zigbee2mqtt/Tafellamp Dimmer', payload '{"action":"","action_duration":null,"battery":17.5,"linkquality":112,"update":{"state":"idle"},"update_available":null}'
info  2022-03-13 12:32:10: MQTT publish: topic 'zigbee2mqtt/Tafellamp Dimmer/action', payload 'off_press_release'

Was this problem solved for the author? What in the end was the solution?