Koenkk/zigbee2mqtt #2102: New Device CALEX Smart Wall Switch - RGB remote - missing messages

LarsHanses, 2019-10-08:
Bug report:
- I was able to pair the device:
- If applicable, provide steps how to reproduce the problem.
- Provide a log with `log_level` set to `debug`, see https://www.zigbee2mqtt.io/configuration/configuration.html
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.

# Bug Report
## I was able to pair the new device.
Device added to devices.js:

{
        zigbeeModel: ['Smart Wall Switch'],
        model: '421782',
        vendor: 'Calex',
        description: 'This Zigbee wall switch can control up to 20 lamps and remember different scenarios. With this you create the desired atmosphere in multiple rooms, with multiple lamps! The remote has a range of 30 meters and can easily be attached to the wall.',
	supports: 'RGB color, white, on, off',
	fromZigbee: [fz.calex_sws],
        toZigbee:[],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            ];

            execute(device, actions, callback);
        }, 

I see messges for all buttons: On, off, brightness up/down, lightingcolorCtrl
But I do not see messages by selecting the RGB ring. LED on remote is flashing, while touching the RGB color ring.

Battery is new, remote close to antenna

## What did you expect to happen
See messages for the RGB color setting
[Calex_SmartWallSwitch_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3704109/Calex_SmartWallSwitch_log.txt)

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: Starting zigbee2mqtt version 1.6.0 (commit #e26ad2a)
CC253X firmware version: Coordinator firmware version: '20190619'

Koenkk, 2019-10-09:
Can you post the logging when running zigbee2mqtt with `DEBUG=* npm start` and pressing the RGB ring?

LarsHanses, 2019-10-09:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3709050/log.txt)
[Calex_SmartWallSwitch_log_Debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/3709054/Calex_SmartWallSwitch_log_Debug.txt)

With Debug information I saw the payload. And I saw devInterview first time.

tb-killa, 2019-10-11:
In your log message we got this two messages:

``
zigbee-shepherd:af Message has unsupported cmdID: colorLoopSet

zigbee-shepherd:af Message has unsupported cmdID: enhancedMoveToHueAndSaturation
``
We need them to get the needing informations you want to get.

tb-killa, 2019-10-11:
BTW. See here https://github.com/Koenkk/zigbee2mqtt/issues/2038
On new dev everything is implemented (herdsman).

Koenkk, 2019-10-12:
@tb-killa thanks for mentioning this!

@LarsHanses on the latest zigbee2mqtt dev branch you should get `No converter` messages for this.

closed 2019-12-18 by stale[bot]

VincentSC, 2020-03-25:
Problem with this device under Phoscon/Deconz is that it prefers a light-link, and it was stealing lights from my bridge.

Here it does not connect to the network, but is shown in the devices list and network map.

@LarsHanses Can you maybe share a copy of the manual? I'm wildly guessing what was resetting and what was pairing, as I lost the little booklet.

VincentSC, 2020-03-26:
Got a reply from Calex: [image]

VincentSC, 2020-03-26:
Did a reset and then pair it close to the antenna. It gets to announce, but does not pair. So not sure how others managed to pair. So when pressing a button it triggers an announce