Koenkk/zigbee2mqtt #2103: Latest coordinator version drops my xiaomi devices, not able to repair

CypherMK, 2019-10-08:
So I updated my CC2531 a week ago to: CC2531_DEFAULT_20190608.zip.
After a few days it dropped my temperature sensor, few days later the motion sensor in the living room and now again a few days later a motion sensor in the hall. The only way to fix this, is completely remove the device and pair from scratch. Before I had the 20190315 MAX STABILITY version running without issues.

On a side note, where can I download the 20191503 MAX STABILITY version. Can't find it anywhere.  I really need it.

CypherMK, 2019-10-08:
https://github.com/Koenkk/Z-Stack-firmware/issues/86 
Found 20191503 here

wimpie007, 2019-10-09:
is it possible to report if the problem is solved with 20191503?
Thanks.

CypherMK, 2019-10-09:
Ok I'll report back. Untill now no issues. 20191503 also seems to respond slightly faster.

CypherMK, 2019-10-12:
Still no issues

CypherMK, 2019-10-16:
Still no issues with 20190315. So is it safe to say that CC2531_DEFAULT_20190608 firmware has issues? @wimpie007 @Koenkk

Koenkk, 2019-10-16:
Did you also check https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing?

CypherMK, 2019-10-16:
No I didn't but I will try. What is the difference?

wimpie007, 2019-10-16:
for me, the source routing firmware was less stable then the normal one. Maybe this has to do with the fact that some routers (lights with fysical wall button) sometimes disappear from the network.
The non-source routing version is now ok with me.

Koenkk, 2019-10-16:
@CypherMK see https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/README.md#im-a-zigbee2mqtt-users-which-one-should-i-use. `20190315` was also a source routing firmware, so I would expect a comparable experience from `20190619`

CypherMK, 2019-10-18:
Installed 20190619 today, so lets see how it goes.

CypherMK, 2019-10-27:
No issues with 20190619. So all good now.

closed 2019-10-27 by CypherMK