Koenkk/zigbee2mqtt #2104: Is there any way to stop an running transition?

blauhorn, 2019-10-09:
Hello and many thanks for the very, very good work here with zigbee2mqtt. 
I have a question:

is there any way to stop an running transition?
The background is that I'm trying to dim up a Tradfri bulb by a brightness-hold on the E1524 and by releasing the same button I'm trying to stay on the dimm value. The two devices are only coupled via a FHEM event handler.
At the moment the transition is executed until the end.

Translated with www.DeepL.com/Translator

Thankyou

## Debug Info
zigbee2mqtt version: 1.60
CC253X firmware version: CC2530_CC2591_SOURCE_ROUTING_20190619

watchmeexplode5, 2019-10-13:
I just wanted to chime and say I'm having this issue as well.

I run a sunrise light script which call up a 600 second transition. Everything works great but I am unable to stop/halt the light transition midway through. If I attempt to send another mqtt set command it will change the light briefly (1-2 seconds), but then return to prior state in the transition. 

Currently I have to let it run through the whole transition before I can use the light again.

watchmeexplode5, 2019-10-19:
Just wanted to let you guys know I found a way of "fixing" this issue for my setup.

I saw another user post an issue about new transitions halting the previous running transition. So to halt the hanging transition all I had to do was add a {"transition": 1} to my command that is meant to halt the hanging state.  

Now the commands work exactly how I would like. More of a work around than a fix but you might be able to apply something like that to fix your current issue @blauhorn

Eschricht, 2019-10-21:
> Just wanted to let you guys know I found a way of "fixing" this issue for my setup.
>
> I saw another user post an issue about new transitions halting the previous running transition. So to halt the [...]

Could you please elaborate on this? I could not get this to work, instead the transition completes more or less instantly with full brightness instead of halting the brightness level at the current state.

blauhorn, 2019-10-21:
unfortunately I had no opportunity to test it yet.  Will probably be this week

Koenkk, 2019-10-21:
I suspect this differs per device; hue bulb could handle this differently than e.g. IKEA bulbs.

closed 2019-12-27 by stale[bot]