Koenkk/zigbee2mqtt #2107: My network map looks...weird. Is my network the cause of slow response times?

timdonovanuk, 2019-10-09:
Motion sensors on my network sometimes are very slow to respond. Is there an issue with my network? Does it normally look so crazy like this, with lots of devices?

[image]

Koenkk, 2019-10-09:
Can you check with the latest dev branch?

bochelork, 2019-10-10:
At least you have a lot of lines! Here I only have one end device with lines. Actually this device is offline since I last started zigbee2mqtt....
[image]

carldebilly, 2019-10-10:
@bochelork did you try to compare with the map from https://github.com/yllibed/Zigbee2MqttAssistant ?

bochelork, 2019-10-10:
> @bochelork did you try to compare with the map from https://github.com/yllibed/Zigbee2MqttAssistant ?

Haven't installed assistant yet. Running zigbee2mqtt on my synology (Realtek RTD1293 CPU). Can I install the assistant on a different machine?

carldebilly, 2019-10-10:
You can install it anywhere, as soon it can connect to the MQTT server...

james-fry, 2019-10-11:
@carldebilly - just tried Z2MA map and it really nice :) great work!
(I have some enhancement suggestions such as colors, but I think its a great start)

I have exactly the same missing connections (from end devices to router) in all the maps, hence I assume it must be down to the data being provided by Z2M to the renderer:

[image]

[image]

[image]

carldebilly, 2019-10-11:
We can see there's no "link back" from routers to coordinator.

bochelork, 2019-10-11:
[image]

closed 2019-12-17 by stale[bot]