Koenkk/zigbee2mqtt #2109: MQTT message: zigbee2mqtt/bridge/state offline

Dim3333, 2019-10-09:
Hello,

My log in Domoticz

`
2019-10-09 20:54:20.049 (ZigBee) Connected to MQTT Server: 127.0.0.1:1883
2019-10-09 20:54:20.049 (ZigBee) MQTT CLIENT ID: 'Domoticz_1570647259_6329'
2019-10-09 20:54:20.049 (ZigBee) MqttClient::subscribe
2019-10-09 20:54:20.049 (ZigBee) onMQTTSubscribed
2019-10-09 20:54:20.100 (ZigBee) MQTT message: zigbee2mqtt/bridge/state offline
2019-10-09 20:54:20.100 (ZigBee) Zigbee2mqtt bridge is offline
2019-10-09 20:54:20.100 (ZigBee) MQTT message: zigbee2mqtt/bridge/config {'version': '1.6.0', 'commit': 'e26ad2a', 'coordinator': 20190608, 'log_level': 'info', 'permit_join': True}
2019-10-09 20:54:20.100 (ZigBee) Zigbee2mqtt log level is info
2019-10-09 20:54:20.100 (ZigBee) Joining new devices is enabled on the zigbee bridge
`
Can you help me ?
My sensor (Xiaomi Temperature) is not detected.

Thanks.

Dim3333, 2019-10-09:
The problem was write access to the port. It's OK with the command `sudo chown [USER] [PORT]`

I don't know why I have to write acces at each reboot. Is it possible to avoid it ????

lolorc, 2019-10-10:
use an udev rules to that the serial port always have to same name + the correct permission.
achieve here with : ```SUBSYSTEM=="tty", ATTRS{idVendor}=="0451", ATTRS{idProduct}=="16a8", ATTRS{serial}=="YOURSERIAL", SYMLINK="ttyUSB.CC2531-01", OWNER="pi"``` in /etc/udev/rules.d/

closed 2019-12-16 by stale[bot]

phenomeus, 2020-02-15:
this is SOOO important to get into FAQ or installation documentation! lost some time to find this out… 

so after reboot the LED was off and of course zigbee2mqtt wasn't starting because the device couldn't be access from pi or any other user other than root. 

so this is needed every time after reboot: sudo chown [USER] [PORT]

and as @lolorc stated, this can be fixed by a rule in udev.

first: udevadm info -a -n /dev/ttyACM0 | grep 'serial'
get the serial to your device "YOURSERIAL"

sudo nano /etc/udev/rules.d/99-usb-serial.rules

add this line: 
SUBSYSTEM=="tty", ATTRS{idVendor}=="0451", ATTRS{idProduct}=="16a8", ATTRS{serial}=="YOURSERIAL", SYMLINK="ttyUSB.CC2531-01", OWNER="pi"

modify your zigbee2mqtt config to adjust maybe new SYMLINK name:
nano /opt/zigbee2mqtt/data/configuration.yaml

…
serial:
  port: /dev/ttyUSB.CC2531-01
…

@Koenkk maybe you can fix this important information somewhere?

Koenkk, 2020-02-16:
@phenomeus this can be added here: https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/information/FAQ.md#verify-that-the-user-you-run-zigbee2mqtt-as-has-write-access-to-the-port

Could you make a pr?