Koenkk/zigbee2mqtt #2112: Configurable reporting interval, or battery-saving settings for secondary sensors

airdrummingfool, 2019-10-10:
Is there any way to configure reporting intervals for certain characteristics?

For example, I have a battery-powered Iris 3320-L contact sensor that also supports temperature reporting. I've successfully added code to `zigbee-herdsman-converters` to enable the temperature reporting, but it is reporting _every minute_. This greatly increases battery usage, to the point where I would rather just not receive temperature updates at all.

I see that `configureReporting.temperature()` uses the following code:
```js
    temperature: async (endpoint) => {
        const payload = [{
            attribute: 'measuredValue',
            minimumReportInterval: 0,
            maximumReportInterval: repInterval.HOUR,
            reportableChange: 0,
        }];
        await endpoint.configureReporting('msTemperatureMeasurement', payload);
    },
```

Because this device is battery-powered, and temperature isn't the primary focus, the minimum report interval for temperature should be much higher - something on the magnitude of 15 minutes in my opinion. However, I know that some devices may warrant much more frequent temperature updates. Thus, it would be nice to be able to pass variables to the configuration code. Ideally, perhaps users could pick these for themselves, but I think having it configurable per-device would be a good first step.

Alternatively, how would you feel about a new `temperature_slow` or `temperature_batterysaver` `configureReporting` key that was used for battery-powered devices where temperature reporting was not the primary function? This would allow temperature devices to continue receiving quick updates, but provide a slower, battery-saving alternative for devices that report temperature as a secondary feature.

I'd be happy to come up with a PR for either one, if you think it is an acceptable idea.

airdrummingfool, 2019-10-13:
That's only slightly better - I still got 11 reports in 11 minutes at one point, and 18 over a 30 minute period. It sees like a `reportableChange` of `1` means that it will report a change of 0.01ºC (which makes sense, because temperature is transmitted in hundredths of a ºC). In my case (and, I'd assume, the case of many users of devices that provide temp as a secondary feature), that is still far too granular. I'd prefer something around 0.25-0.5ºC. Would `reportableChange: 25` be acceptable?

```
zigbee2mqtt:info  2019-10-13T02:44:20: MQTT publish: topic 'zigbee2mqtt/0x000d6f00057cd178', payload '{"contact":true,"linkquality":60,"temperature":19.47}'
zigbee2mqtt:info  2019-10-13T02:53:52: MQTT publish: topic 'zigbee2mqtt/0x000d6f00057cd178', payload '{"contact":true,"linkquality":70,"temperature":19.62}'
zigbee2mqtt:info  2019-10-13T02:54:52: MQTT publish: topic 'zigbee2mqtt/0x000d6f00057cd178', payload '{"contact":true,"linkquality":70,"temperature":19.65}'
zigbee2mqtt:info  2019-10-13T02:55:52: MQTT publish: topic 'zigbee2mqtt/0x000d6f00057cd178', payload '{"contact":true,"linkquality":73,"temperature":19.67}'
zigbee2mqtt:info  2019-10-13T02:56:52: MQTT publish: topic 'zigbee2mqtt/0x000d6f00057cd178', payload '{"contact":true,"linkquality":73,"temperature":19.68}'
zigbee2mqtt:info  2019-10-13T03:00:53: MQTT publish: topic 'zigbee2mqtt/0x000d6f00057cd178', payload '{"contact":true,"linkquality":70,"temperature":19.7}'
zigbee2mqtt:info  2019-10-13T03:01:53: MQTT publish: topic 'zigbee2mqtt/0x000d6f00057cd178', payload '{"contact":true,"linkquality":63,"temperature":19.69}'
zigbee2mqtt:info  2019-10-13T03:02:53: MQTT publish: topic 'zigbee2mqtt/0x000d6f00057cd178', payload '{"contact":true,"linkquality":68,"temperature":19.68}'
zigbee2mqtt:info  2019-10-13T03:03:54: MQTT publish: topic 'zigbee2mqtt/0x000d6f00057cd178', payload '{"contact":true,"linkquality":68,"temperature":19.66}'
zigbee2mqtt:info  2019-10-13T03:04:54: MQTT publish: topic 'zigbee2mqtt/0x000d6f00057cd178', payload '{"contact":true,"linkquality":70,"temperature":19.65}'
zigbee2mqtt:info  2019-10-13T03:05:54: MQTT publish: topic 'zigbee2mqtt/0x000d6f00057cd178', payload '{"contact":true,"linkquality":70,"temperature":19.63}'
zigbee2mqtt:info  2019-10-13T03:06:54: MQTT publish: topic 'zigbee2mqtt/0x000d6f00057cd178', payload '{"contact":true,"linkquality":52,"temperature":19.62}'
zigbee2mqtt:info  2019-10-13T03:07:54: MQTT publish: topic 'zigbee2mqtt/0x000d6f00057cd178', payload '{"contact":true,"linkquality":52,"temperature":19.61}'
zigbee2mqtt:info  2019-10-13T03:08:54: MQTT publish: topic 'zigbee2mqtt/0x000d6f00057cd178', payload '{"contact":true,"linkquality":63,"temperature":19.6}'
zigbee2mqtt:info  2019-10-13T03:09:54: MQTT publish: topic 'zigbee2mqtt/0x000d6f00057cd178', payload '{"contact":true,"linkquality":68,"temperature":19.59}'
zigbee2mqtt:info  2019-10-13T03:10:55: MQTT publish: topic 'zigbee2mqtt/0x000d6f00057cd178', payload '{"contact":true,"linkquality":63,"temperature":19.57}'
zigbee2mqtt:info  2019-10-13T03:11:55: MQTT publish: topic 'zigbee2mqtt/0x000d6f00057cd178', payload '{"contact":true,"linkquality":63,"temperature":19.56}'
```

Koenkk, 2019-10-13:
Yes 25 is acceptable

airdrummingfool, 2019-10-15:
25 has been working fine for my sensors.

Do you want a PR that updates the `reportableChange` value for the `configureReporting.temperature()` method to 25? This would affect all devices already using that bind, so the configKey might need to be bumped. Or should a new method be created that has `reportableChange` set to 25, something like `configureReporting.temperature_batterysaver()`?

Koenkk, 2019-10-15:
> value for the configureReporting.temperature() method to 25

yes, but don't bump the configure key for now. This will fail mostly for temperature sensors as they are battery powered. Meaning that users that experience issues related to this should just re-pair.

closed 2019-12-14 by Koenkk