Koenkk/zigbee2mqtt #2113: Xiaomi MiJia Honeywell smoke detector add support for features

gonzasmts, 2019-10-10:
Hello,

it would be nice if we are able to use the following features, using zigbee2mqtt as we can use in the app. 

- set the sensitivity (i want to add one in the kitchen)
- execute self-test
- Tandem mode (not required for my case)

also i don't know if this one is possible, but an option to mute the alarm after it get’s triggered, without need to click the button?

someone already did some work on this, #420, but it seams to be stopped.

will this be functional?

Best regards

carldebilly, 2019-10-11:
You're not the only one to have it....  In my case, I want to set the sensitivity to maximum to detect cigarette usage...

Koenkk, 2019-10-11:
Please sniff the traffic when doing this via the xiaomi gateway, then we can add it.

gonzasmts, 2019-10-14:
i don't have one, so i can't help with that.

michaelbouma, 2019-10-14:
Found this link https://github.com/foz333/Xiaomi/blob/master/devicetypes/bspranger/xiaomi-mijia-honeywell-fire-detector.src/xiaomi-mijia-honeywell-fire-detector.groovy which looks like this guy decoded the send and receive protocol.

kovaga, 2019-10-31:
Ok, so I have connected my two smoke detectors to Xiaomi gateway, but I dont see an option to activate the sound alarm on the second smoke detector if the first detector is triggered. I can only see the possibility to enable "Linkage hub alarm" (The gateway is linked to send an alarm when smoke is detected)So the gateway/Hub would also start an audio signal when the sensor gets triggered.

julien-w, 2019-11-03:
I'm really interested by this feature. I have a gateway, this detector and a zigbee2mqtt CC2531, i can do some tests.
With the gateway I had the smoke denstity value (ppm) too...

Koenkk, 2019-11-03:
- Sensitivy and selftest is supported now, documentation: https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/devices/JTYJ-GD-01LM_BW.md#sensitivity
- If with tandem mode you mean 'linked mode'; the name might be confusing but AFAIK this doesn't allow for smoke detectors to be linked to each other; what it does: it puts the gateway into alarm mode once the smoke detector goes off. (as @kovaga already explained)
- @julien-w I guess you are talking about the gas sensor and not the smoke one, `gas_density` is already supported.
- Muting the alarm is not possible via the Xiaomi gateway, so I expect this is not possible.

closed 2019-11-07 by Koenkk

altbau9944, 2019-11-07:
Hello,

I have successfully installed Zigbee2Mqtt with the help of these pages and get all the values I want.

Currently I am working with the MiJia Honeywell smoke detector (JTYJ-GD-01LM/BW). 

Here I want to read out the air density (particle density in the air). Via the Xiaomi gateway I could already read the values, but with the Zigbee2Mqtt this is not possible.

Is that possible and if yes, how can I do that?

Koenkk, 2019-11-07:
Where can I find this in the xiaomi app? I searched but couldnt find it.

altbau9944, 2019-11-07:
If I rememer right, it is Mi Home app, or Xiaomi Home in the google app store.

I'm using openhab and have integrated the xiaomi gateway device into OH in the past. Over OH I was able to read the air density value.

Koenkk, 2019-11-07:
I have that app but cannot find this option (I think it's only available for the gas sensor and not the smoke). If I'm wrong, please provide me step by step instructions on where I can see it.

altbau9944, 2019-11-07:
Hi,

I've sold my Xiaomi gateway already, because I use this solution here. That's why I can't provide an step by step instruction. Sorry 

Maybe this article from the Openhab Community provide some more information.
https://community.openhab.org/t/xioami-honeywell-smoke-detector-issue/48610

This is the same smoke detector I'm using right now. And the particle density is pubklished by the MiJia Honeywell smoke detector.

altbau9944, 2019-11-08:
Good morning,

Take a look at this video: https://www.youtube.com/watch?v=zj8yW1Nj7oE
Sorry in German…

I did exactly the same in the past.
The value I want to publish over mqtt is: Particle Density

Cheers
Altbau9944

kovaga, 2019-11-08:
@Koenkk @altbau9944 
Just checked and apparently there density is amongst the channels in the openhab mihome smoke sensor.

        <thing-type id="smoke">
                <supported-bridge-type-refs>
                        <bridge-type-ref id="bridge" />
                </supported-bridge-type-refs>
                <label>Xiaomi Smoke Sensor</label>
                <channels>
                        <channel id="density" typeId="density"></channel>
                        <channel id="alarm" typeId="smoke_alarm"></channel>
                        <channel id="status" typeId="status"></channel>
                        <channel id="batteryLevel" typeId="system.battery-level" />
                        <channel id="lowBattery" typeId="system.low-battery" />
                </channels>
                <config-description-ref uri="thing-type:mihome:device" />
        </thing-type>

Need to figure out now how are they getting this value. I will repair my smokedetector to bridge and try to sniff it out.

kovaga, 2019-11-08:
The piece of code at 
/bundles/org.openhab.binding.mihome/src/main/java/org/openhab/binding/mihome/internal/handler/XiaomiSensorSmokeHandler.java

indicates that **density** is not always present in the JSON reported by the smoke alarm
```
@Override
    void parseReport(JsonObject data) {
        super.parseReport(data);
        if (data.has(DENSITY)) {
            Integer density = data.get(DENSITY).getAsInt();
            updateState(CHANNEL_DENSITY, new DecimalType(density));
        }
    }

```

Could it be possible that density is only reported and included if it is above some threshdhold value.
anyway, if in doubt sniff it out. :) something to do for the weekend.

Koenkk, 2019-11-08:
Thanks, I figured out how to get it, supported in latest dev branch now.

altbau9944, 2019-11-09:
Sounds great!!

Now I have to change to dev branch and use zigbee-herdsman-converters...
- A project for next week....

Thanks for your support!!
Great job and support

kovaga, 2019-11-10:
updated to the latest dev branch, but i am not seeing density report at all.

MQTT publish: topic 'zigbee2mqtt/sens_smoke1', payload '{"battery":100,"voltage":3195,"linkquality":57,"last_seen":1573393389829,"smoke":true,"sensitivity":"high"}'

do all of them report it or some dont?

Koenkk, 2019-11-10:
@kovaga it can take up to 2 hours before it is reported.

kovaga, 2019-11-11:
@Koenkk thanks, it has appeared.

ru320, 2020-02-22:
@Koenkk and all others.

is it now possible to trigger the alarm for this device?
for my understanding, every zigbee device is listening the messages of each other device.
What if we send the "Smoke" message into the Zigbee network. Not with the set command. Similar to the message which the smoke detector would use but with a another ID. Then maby the smoke detector would listen to this message triggers the alarm.

So my question is? how can i send such a message into the zigbee network?

Koenkk, 2020-02-22:
AFAIK this is not possible

ru320, 2020-02-22:
Thank you die the very quick answer.
Bty thank you for your system, this is awesome

Anjerlaan, 2020-08-29:
I also bought 3 of these
When I publish
```
zigbee2mqtt/Rookmelder_beneden/set {"smoke": "true"}
zigbee2mqtt/Rookmelder_beneden/set {"smoke": "false"}
```
Both give the error:
```
No converter available for 'smoke' (true)
No converter available for 'smoke' (false)
```

I guess there are no options to start or stop the alarm with zigbee messages?

Koenkk, 2020-08-29:
@Anjerlaan unfortunately not! (also doesn't seem to be supported with the official gateway).

Anjerlaan, 2020-08-29:
Hmmm...that is a pity.....I should have read this issue first before buying them :-(
So the zigbee functionality is only capable to notify, set sensitivity and perform a self test...
Quite poor to my opinion.

I don't have the official gateway.
Might it be added in newer firmware, or was this finding already based on the latest firmware?

ru320, 2020-08-29:
For me it was the same.  I wanted to use the fire alarm as a siren too.  as far as I know, a second fire alarm can trigger the first.  maybe it works if you set up two and then trigger the first with smoke and record the signal.  but at least I now have a fire alarm that also alerts me on my cell phone.  Overall, I can only say thank you again for the great zigbee2mqtt program

Anjerlaan, 2020-11-22:
> For me it was the same. I wanted to use the fire alarm as a siren too. as far as I know, a second fire alarm can trigger the first. maybe it works if you set up two and then trigger the first with [...]

With node-red (contrib-castv2), I push audio mp3 messages to my 3 Google home mini devices if one of the smoke alarm devices in Domoticz are triggered.

kevincaradant, 2021-01-03:
Hi

Oh damn I didn't see this thread before to buy them ... 

So If I understand, Xiaomi doesn't allow us to control the smoke sensor (siren ON / OFF) like the Develco SMSZB-120 for example ? 

It's really painfully for my setup but my fault, I didn't check that before to buy x)

s3frank, 2021-02-06:
Hi all,

Is it possible to disable the detector via an API call?
My use case is that my wife cooks sometimes some very strong soy based things and the fumes that come from that seem to trigger the alarm. I would like to be able to disable / enable the detector via API call so that I can give her a button which would turn it off, start a timer and turn it on again 15 minutes later.
But I don't see a "stop sniffing" type option exposed.

My alternative will be to put a networked relay before the power source....but I am hoping it can be done on the device.

Cheers,
Frank

kovaga, 2021-02-06:
> Hi all,
> 
> Is it possible to disable the detector via an API call?
> My alternative will be to put a networked relay before the power source....but I am hoping it can be done on the device.
> 
> Cheers,
> Frank

I think the only option is to set the sensitivity of the device to low.

kevincaradant, 2021-02-06:
Hi @kovaga and others :)

I was wondering if you have also false positive ? Twice, I have two different sensors which report false positive. The first time it was because my vaccum cleaner did some dust explused from the engine of the vaccum (paint texture) and the second time it was because i took a hot shower and the condensation (humidity in the air) has trigger the sensors. 

So it's not only smoke fire which are triggered by these fire sensors ? :)

kovaga, 2021-02-06:
I have three of those units, all set to high sensitivity. Didn't have a false positive so far. 
However, had a spot on detection, when I left a plastic bread board on the electric stove and turned on the wrong burner :)

duncho1, 2023-01-03:
Hi guys, I'd like to connect MiJia Honeywell smoke detector to Aqara M1S hub in EU region. I managed to do so, but seems it's not fully supported as I can't set automation nor see smoke detector status, even not able to rename it. Is it possible to change region on MiJia Honeywell smoke detector from China to EU?
Many thanks.