Koenkk/zigbee2mqtt #2116: [feature request, website] Comments in supported devices

juanjux, 2019-10-11:
Adding here since the website doesn't seem to support adding issues.

It would be great to have comments/user contributed notes below every device in the supported devices pages. This way people could share tips, workarounds, and other things, like PHP's documentation.

james-fry, 2019-10-11:
I think the preferred way to do this is by submitting a PR with the additional details/corrections:
https://github.com/Koenkk/zigbee2mqtt.io/pulls

Maybe though it would be nice to use the z2m discourse forum for ongoing discussion though?
Every supported device listed on the website:
https://www.zigbee2mqtt.io/information/supported_devices.html
could have a link to a topic in the device support category:
https://zigbee2mqtt.discourse.group/c/Need-help-supporting-a-new-device-This-is-the-place

juanjux, 2019-10-11:
@james-fry, I'm a programmer but not a web or js one so the result of a PR authored by me could be suboptimal.

The Discourse thread per device, however, sounds like a good one, as then the devices page could just have the link.

juanjux, 2019-10-11:
PS: I volunteer to create an "official" thread for every device and make a PR with the links if you're interested. It probably would need a new forum category where thread creation is limited.

james-fry, 2019-10-11:
> but not a web or js one

The website is defined in markdown as far as I know (as oppsed to html/js etc).
E.g. https://raw.githubusercontent.com/Koenkk/zigbee2mqtt.io/master/docs/devices/046677476816.md

> PS: I volunteer to create an "official" thread for every device and make a PR with the links if you're interested. It probably would need a new forum category where thread creation is limited.

Thanks ! :)
I agree re new category btw.
You'll need @Koenkk support for it though...

Koenkk, 2019-10-12:
I've created a new category for this: https://zigbee2mqtt.discourse.group/c/Share-your-finding-of-already-supported-devices

I feel it doesn't make sense to do this for every device; perhaps only for very popular devices? Also I feel that e.g. information required to get it working with zigbee2mqtt should be in the documentation rather than buried in a thread. However I agree it would be nice if users can share their experience/show-off their integration.

closed 2019-12-27 by stale[bot]