Koenkk/zigbee2mqtt #2119: Cant start after running upgrade to 1.6

fefris, 2019-10-11:
**Same issue as seen by other previously i think for me after running upgrade, have not changed the config file.**

error below when running sudo systemctl start zigbee2mqtt after running upgrade. Was all working before upgrade. Have tried upgrade procedure twice.

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.9.0
3 info using node@v10.16.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.6.0~prestart: zigbee2mqtt@1.6.0
6 info lifecycle zigbee2mqtt@1.6.0~start: zigbee2mqtt@1.6.0
7 verbose lifecycle zigbee2mqtt@1.6.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.6.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.6.0~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.6.0~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.6.0~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.6.0~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.6.0 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:198:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:198:13)
13 verbose stack     at maybeClose (internal/child_process.js:982:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.6.0
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.14.70-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.16.0
19 verbose npm  v6.9.0
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.6.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.6.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

**Similarly after npm start this error**

internal/modules/cjs/loader.js:638
    throw err;
    ^

Error: Cannot find module 'zigbee-herdsman/dist/deprecated/ziee'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Module.require (internal/modules/cjs/loader.js:690:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/zapp/cie.js:6:14)
    at Module._compile (internal/modules/cjs/loader.js:776:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:787:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-10-11T20_42_36_463Z-debug.log

_Originally posted by @fefris in https://github.com/Koenkk/zigbee2mqtt/issues/1956#issuecomment-541216573_

fefris, 2019-10-11:
**tried this fix https://github.com/Koenkk/zigbee2mqtt/issues/1892#issuecomment-527837064 which didnt help. 

also moved to dev branch which seems to have changed the error slightly.** 

Oct 12 08:41:38 raspberrypi npm[4996]:   zigbee2mqtt:info 10/12/2019, 8:41:38 AM Starting zigbee2mqtt version 1.6.0 (commit #e26ad2a)
Oct 12 08:41:38 raspberrypi npm[4996]:   zigbee2mqtt:info 10/12/2019, 8:41:38 AM Starting zigbee-shepherd
Oct 12 08:41:56 raspberrypi npm[4996]:   zigbee2mqtt:info 10/12/2019, 8:41:56 AM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: SRSP - SYS - osalNvRead after 6000ms)
Oct 12 08:42:15 raspberrypi systemd[1]: Stopping zigbee2mqtt...
Oct 12 08:42:31 raspberrypi systemd[1]: Stopped zigbee2mqtt.
Oct 12 08:57:16 raspberrypi systemd[1]: Started zigbee2mqtt.
Oct 12 08:57:17 raspberrypi npm[5301]: > zigbee2mqtt@1.6.0 start /opt/zigbee2mqtt
Oct 12 08:57:17 raspberrypi npm[5301]: > node index.js
Oct 12 08:57:23 raspberrypi npm[5301]: zigbee2mqtt:info  2019-10-11T21:57:23: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-10-12.08-57-19'
Oct 12 08:57:23 raspberrypi npm[5301]: zigbee2mqtt:info  2019-10-11T21:57:23: Starting zigbee2mqtt version 1.6.0 (commit #ba92f73)
Oct 12 08:57:23 raspberrypi npm[5301]: zigbee2mqtt:info  2019-10-11T21:57:23: Starting zigbee-herdsman...
Oct 12 08:57:35 raspberrypi npm[5301]: zigbee2mqtt:error 2019-10-11T21:57:35: Error while starting zigbee-herdsman
Oct 12 08:57:35 raspberrypi npm[5301]: zigbee2mqtt:error 2019-10-11T21:57:35: Error: SRSP - SYS - version after 6000ms
Oct 12 08:57:35 raspberrypi npm[5301]:     at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
Oct 12 08:57:35 raspberrypi npm[5301]:     at ontimeout (timers.js:436:11)
Oct 12 08:57:35 raspberrypi npm[5301]:     at tryOnTimeout (timers.js:300:5)
Oct 12 08:57:35 raspberrypi npm[5301]:     at listOnTimeout (timers.js:263:5)
Oct 12 08:57:35 raspberrypi npm[5301]:     at Timer.processTimers (timers.js:223:10)
Oct 12 08:57:35 raspberrypi npm[5301]: zigbee2mqtt:error 2019-10-11T21:57:35: Failed to start zigbee
Oct 12 08:57:35 raspberrypi npm[5301]: zigbee2mqtt:error 2019-10-11T21:57:35: Exiting...
Oct 12 08:57:35 raspberrypi npm[5301]: zigbee2mqtt:error 2019-10-11T21:57:35: Error: SRSP - SYS - version after 6000ms
Oct 12 08:57:35 raspberrypi npm[5301]:     at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
Oct 12 08:57:35 raspberrypi npm[5301]:     at ontimeout (timers.js:436:11)
Oct 12 08:57:35 raspberrypi npm[5301]:     at tryOnTimeout (timers.js:300:5)
Oct 12 08:57:35 raspberrypi npm[5301]:     at listOnTimeout (timers.js:263:5)
Oct 12 08:57:35 raspberrypi npm[5301]:     at Timer.processTimers (timers.js:223:10)
Oct 12 08:58:11 raspberrypi npm[5301]: npm ERR! code ELIFECYCLE
Oct 12 08:58:11 raspberrypi npm[5301]: npm ERR! errno 1
Oct 12 08:58:11 raspberrypi npm[5301]: npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
Oct 12 08:58:11 raspberrypi npm[5301]: npm ERR! Exit status 1
Oct 12 08:58:11 raspberrypi npm[5301]: npm ERR!
Oct 12 08:58:11 raspberrypi npm[5301]: npm ERR! Failed at the zigbee2mqtt@1.6.0 start script.
Oct 12 08:58:11 raspberrypi npm[5301]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Oct 12 08:58:11 raspberrypi npm[5301]: npm ERR! A complete log of this run can be found in:
Oct 12 08:58:11 raspberrypi npm[5301]: npm ERR!     /home/pi/.npm/_logs/2019-10-11T21_58_11_173Z-debug.log
Oct 12 08:58:11 raspberrypi systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Oct 12 08:58:11 raspberrypi systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Oct 12 08:58:11 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Oct 12 08:58:11 raspberrypi systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Oct 12 08:58:11 raspberrypi systemd[1]: Stopped zigbee2mqtt.
Oct 12 08:58:11 raspberrypi systemd[1]: Started zigbee2mqtt.
Oct 12 08:58:12 raspberrypi npm[5337]: > zigbee2mqtt@1.6.0 start /opt/zigbee2mqtt
Oct 12 08:58:12 raspberrypi npm[5337]: > node index.js
Oct 12 08:58:14 raspberrypi npm[5337]: zigbee2mqtt:info  2019-10-11T21:58:14: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-10-12.08-58-14'
Oct 12 08:58:14 raspberrypi npm[5337]: zigbee2mqtt:info  2019-10-11T21:58:14: Starting zigbee2mqtt version 1.6.0 (commit #ba92f73)
Oct 12 08:58:14 raspberrypi npm[5337]: zigbee2mqtt:info  2019-10-11T21:58:14: Starting zigbee-herdsman...
Oct 12 08:58:27 raspberrypi npm[5337]: zigbee2mqtt:error 2019-10-11T21:58:27: Error while starting zigbee-herdsman
Oct 12 08:58:27 raspberrypi npm[5337]: zigbee2mqtt:error 2019-10-11T21:58:27: Error: SRSP - SYS - version after 6000ms
Oct 12 08:58:27 raspberrypi npm[5337]:     at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
Oct 12 08:58:27 raspberrypi npm[5337]:     at ontimeout (timers.js:436:11)
Oct 12 08:58:27 raspberrypi npm[5337]:     at tryOnTimeout (timers.js:300:5)
Oct 12 08:58:27 raspberrypi npm[5337]:     at listOnTimeout (timers.js:263:5)
Oct 12 08:58:27 raspberrypi npm[5337]:     at Timer.processTimers (timers.js:223:10)
Oct 12 08:58:27 raspberrypi npm[5337]: zigbee2mqtt:error 2019-10-11T21:58:27: Failed to start zigbee
[... cut ...]
    at Timer.processTimers (timers.js:223:10)
zigbee2mqtt:error 2019-10-11T22:05:51: Failed to start zigbee
zigbee2mqtt:error 2019-10-11T22:05:51: Exiting...
zigbee2mqtt:error 2019-10-11T22:05:51: Error: SRSP - SYS - version after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)

fefris, 2019-10-11:
well something worked, on reboot we are up and running again.

closed 2019-10-11 by fefris

g3rb3n, 2020-01-19:
Power cycling the TI CC2531 did the trick for me. A reboot does not power cycle the USB I guess.