Koenkk/zigbee2mqtt #2123: Timeout error on HUE bulbs

sti0, 2019-10-12:
I'm running zigbee2mqtt edge addon on HASSIO right now. All my connected hue bulbs (really every single one) are raising this error every few minutes:

```
zigbee2mqtt:error 2019-10-09T20:52:48: Failed to setup reporting for '0x00178801029116bd' - Error: Timeout - 45157 - 11 - 128 - 7 after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
```

All bulbs are online and can be turned on and off, color changing ... This error didn't appear on zigbee-shepherd. Other devices (Xiaomi sensors, Tradfri bulbs, Hue motion sensors) didn't raise this error.

Koenkk, 2019-10-12:
Should be fixed in latest dev.

sti0, 2019-10-12:
Installed zigbee2mqtt edge version again but still same error messages. Is it possible to see if it's the right build? Need the edge version to be reinstalled or is it sufficient to restart?

```
error 2019-10-10T21:29:29: Failed to setup reporting for '0x0017880102b9305d' - Error: Timeout - 12875 - 11 - 31 - 7 after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
error 2019-10-10T21:29:29: Failed to setup reporting for '0x00178801029116bd' - Error: Timeout - 45157 - 11 - 33 - 7 after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
info  2019-10-10T21:29:30: MQTT publish: topic 'zigbee2mqtt/zigbee2mqtt_router/availability', payload 'offline'
error 2019-10-10T21:29:40: Failed to setup reporting for '0x0017880102ec32f1' - Error: Timeout - 7572 - 11 - 35 - 7 after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
error 2019-10-10T21:29:40: Failed to setup reporting for '0x001788010285dac9' - Error: Timeout - 47223 - 11 - 36 - 7 after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
```

Koenkk, 2019-10-12:
I expect that it only occurs on startup now an not every minute.

sti0, 2019-10-12:
No, unfortunately not. Messages still printed out every 2 minutes

Koenkk, 2019-10-12:
Can you update again? From commit till new version can take up to 2 hours for hassio.

sti0, 2019-10-12:
Do I need to uninstall and reinstall the addon?

Koenkk, 2019-10-12:
I've never used hassio but just use the same mechanism as you already used to update.

james-fry, 2019-10-12:
@sti0 you do need to uninstall the edge addon and then reinstall

sti0, 2019-10-12:
Problem solved with latest dev branch.

closed 2019-10-12 by sti0