Koenkk/zigbee2mqtt #2126: Cannot pair MCCGQ11LM door sensors

krzyk, 2019-10-13:
# Bug Report
I can't pair Aqara door sensors (I have two) - MCCGQ11LM

## What happened
Nothing happens in the zigbee2mqtt logs, so I assume no pairing happens.

## What did you expect to happen
Door sensor have paired.

## How to reproduce it (minimal and precise)
1. Install zigbee2mqtt 1.6 on RPi 4
2. Flash CC2531 with 20190608 firmware
3. npm start
4. Hold the reset button on door sensor until it starts flashed (I also tried clicking the button afterwards with 1 second intervals, as was suggested in other bug reports)

## Debug Info
zigbee2mqtt version: 1.6.0 (e26ad2a)
CC253X firmware version: 20190608

Logs at https://pastebin.com/cgSnjYgW

krzyk, 2019-10-13:
I tried with two sensors that I have.
I moved CC2531 to a different USB port (2.0 and 3.0), restarted coordinator, and RPi 4.
I also removed several times the battery from sensors, and still nothing.
I don't have any other coordinator/gateway so I can't test if something wrong with sensors.

I also have bluetooth sensors and 2.4GHz wifi network, but I try pairing with CC2531 just next to it (1cm distance).

tanjipon, 2019-10-23:
I have the same issue pairing with Xiaomi Smart Plug. Nothing happened. Does someone know what the problem is?

Syoll, 2019-11-07:
I have exactly the same issue with the MCCGQ11LM and with all the same characteristics but on a RPI3. One month ago I paired with success 2 xiaomi door sensor. Today I have received new ones but impossible to pair... even all instructions found followed as krzyck describes. I also tried to repair my old door sensor and it doesn't work anymore. I will try to uninstall/install zigbee2mqtt.

Koenkk, 2019-11-07:
Switching to the dev branch should greatly improve the xiaomi pairing behavior (#1888). https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

Syoll, 2019-11-07:
Thank you, I tried but it didn't solve the issue. I don't know what can I do more.

Koenkk, 2019-11-07:
Can you share your log when pairing it?

Syoll, 2019-11-07:
This is my logs, nothing happens when pairing it (in debug mode or not) :

Debug mode : https://pastebin.com/CRnEia0S
Normal mode : https://pastebin.com/13yDbRqd

Maybe I sould try a sniff method. I will try it later.
Anyway thank you for your help.

myroad94, 2019-12-29:
Hi, I have the exact same problem. Tried with zigbee2mqtt 1.7.0 and 1.8.0. I keep pushing reset button for more than 5s, I can see the light blinking, but that's all. Even at a few cm from the coordinator. Setting with zigbee assistant to DEBUG did not change anything and I cannot see  any log related to this.

myroad94, 2019-12-29:
No improvement with latest-dev https://hub.docker.com/layers/koenkk/zigbee2mqtt/latest-dev/images/sha256-53b7b31d719c4173e32fa12f810b1d2e7c97b4636e5eb12fdf7255271a0df832

I also applied a technique of pairing by another buyer (source [here](https://www.gearbest.com/access-control/pp_626703.html?lkid=11394713#anchorGoodsReviews)), without success

> Pairing with zigbee2mqtt is super easy: after allowing pairing, just press the button on the door sensor for around 4s until the LED starts to blink, then press it again every second and check your zigbee2mqtt logs until it appears there and you're good. don't worry if there is a warning at first, that seems to be normal.

myroad94, 2019-12-29:
Tried with another device same model. Same issue. For info, it seems to me, reading some comments, that there are 2 generations of the model, one with a reset as a pinhole, one as a button. I have 2 as a button. Maybe it worked before with another generation. Anyone for help? @Koenkk ? Thanks!

myroad94, 2019-12-29:
Also, last info because I am giving up now, I used firmware CC2531_DEFAULT_20190608.zip as coordinator.

septeven, 2020-01-01:
Same thing here - I have just ordered 8 of those sensors - none of them have been able to paired with zigbee2mqtt. Nothing on logs, exactly same as OP

Any chance to know how to fix it - I would hate having bought those sensors for nothing.
Thanks

gunrunner20, 2020-01-03:
I had the same issue with a CC2531 on CC2531_20190425 and a RPi 3 on 1.7.1.
First I updated from 1.7.1 to 1.8.0 without any further success. Then I updated the CC2531 to CC2531_DEFAULT_20190608 and now the pairing works.

myroad94, 2020-01-03:
Ooookay, I rebooted my computer and now it pairs. No idea why!

septeven, 2020-01-04:
@gunrunner20 
thanks, I updated to 20190608 and everything works now.

krzyk, 2020-01-11:
@septeven Strange, my issue is reported for 20190608, so for me it doesn't work there at all :(

septeven, 2020-01-12:
@krzyk I'm using zigbee2mqtt 1.8 dev branch, did you try with it ?

myroad94, 2020-01-12:
@krzyk I would suggest: 1/ reboot the machine using the zigbee2mqtt, see if this works 2/ reset the state of persistence and readd all zigbee devices
1.8 should be enough.

krzyk, 2020-01-12:
I did try 1.8, since the report the device has been rebooted multiple times.

I don't have any devices added, the door sensor would be the first of it ever worked.

IronButterfly, 2020-01-14:
The same issue with MCCGQ11LM. Can't pair. 
The version of z2m is 1.8.0 on RPI4
Reboot didn't help.

myroad94, 2020-01-14:
I reproduced the issue and what worked for me was not a reboot, but to take the CC2531 off and replug it. I don't know why, after sometime it cannot add devices anymore.

cirolaferrara, 2020-01-23:
I have 20190608 and 1.9.0. Can't pair MCCGQ01LM. All other Xiaomi devices works

Koenkk, 2020-01-23:
@cirolaferrara please provide the herdsman debug logging when pairing the device. 

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

cirolaferrara, 2020-01-23:
This is the log. Nothing happens if i hold reset button on the device for +- 5 seconds.

I tried with another non-already-paired with xiaomi gateway MCCGQ11LM sensor and it works

```
zigbee2mqtt:info  2020-01-23 16:44:47: Connecting to MQTT server at mqtt://localhost
zigbee2mqtt:info  2020-01-23 16:44:47: Connected to MQTT server
zigbee2mqtt:info  2020-01-23 16:44:47: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82,254,1,69,203,254,113,254,3,69,182,0,0,0,240] +58ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82,254,1,69,203,254,113,254,3,69,182,0,0,0,240] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0} +59ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,69,203,254,113,254,3,69,182,0,0,0,240] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 2 - 5 - 203 - [254] - 113 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - permitJoinInd - {"duration":254} +67ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [0,0,0] - 240 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0} +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
zigbee2mqtt:info  2020-01-23 16:44:47: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.9.0","commit":"287f843","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'
```

Koenkk, 2020-01-23:
@cirolaferrara if any other zigbee network is still powered on (e.g. xiaomi hub) try powering that down first. If that does't help try replacing the batteries.

krzyk, 2020-01-23:
@Koenkk Let me jump in, I tried replacing batteries, and I don't have any other zigbee networks. There is something strange with those aquara door sensors. I also tried them with deCONZ and same, they are not seen. Is there a specific Xiaomi protocol that they require?

cirolaferrara, 2020-01-23:
> @cirolaferrara if any other zigbee network is still powered on (e.g. xiaomi hub) try powering that down first. If that does't help try replacing the batteries.

Xiaomi hub was power off. I will try to remove battery

Koenkk, 2020-01-24:
@krzyk maybe range? 

- https://www.zigbee2mqtt.io/information/FAQ.html#why-dont-my-devices-pair-when-using-a-raspberry-pi-4
- https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html

derekoharrow, 2020-02-10:
Just FYI, I bought an Aqara QBKG11LM switch and had this exact same issue - I couldn't get it to pair no matter what I tried. Eventually, I tried unplugging the CC2531 from my Pi (while live), re-inserting it, restarting zigbee2mqtt and hey presto it then paired just fine. Hope this helps someone.

mikfoo, 2020-03-31:
I had this issue as well and after fighting with it for 20 minutes I tried rebooting my server. Fixed the problem and it paired right away!

Range was definitely not an issue as I tried pairing it a couple of centimeters from the device as well.

It should be noted I have a Hue hub in the vicinity as well and I did not try powering it off.

krzyk, 2020-03-31:
OK, it worked for me.
What I did was:
1. Add a 2m USB extension cable - I thought at first that this is just a snake oil but apparently it worked :)
2. Upgrade to newest zigbee2mqtt version

closed 2020-03-31 by krzyk

BergenLarsen, 2021-06-25:
In case anyone stumbled across this thread  and needed to get their sensor paired. 
What worked for me:

1. Unplug the USB Zigbee gateway from Pi, Move it to my pc,  
2. Installed Z2M on the PC (linux on a laptop, but I guess it should matter if you are using windows/mac)
3. Took the same config file from the PI, moved it to the folder on my PC, (so I kept the same network key). 
4. Tried the standard pairing process (hold down button for ~4 seconds till the light starts blinking)
5. Pairing success!
6. Shut down Z2M on PC
7. Copy config file, db file and state file across to the PI's data folder for Z2M
8. Plug USB Zigbee gateway back into Pi.
9. Restart Zigbee, 
10. Great success, it's nice.

Avamander, 2021-09-15:
I didn't have to do anything except moving the MCCGQ11LM basically right next to the antenna, hold down the button until it started flashing and then release after two flashes.

Then it turns off for a bit and then flashes three times, after that you can press the button for a short time and it'll report its battery status and you can see it in the logs.

pete-simmons, 2025-01-26:
> I didn't have to do anything except moving the MCCGQ11LM basically right next to the antenna, hold down the button until it started flashing and then release after two flashes.
>
> Then it turns off [...]

This was the solution for me as well, thanks @Avamander