Koenkk/zigbee2mqtt #2129: CC2530 CC2591 router modelID undefined

cslehel, 2019-10-13:
Hi, 
first i would like to thank you for your work.

I have a CC2531 USB coordinator flashed with 2019-06-08 default firmware. 
I flashed the CC2530 CC2591 modele with the 2019-02 router firmware. 

When i tried to pair i am getting the following messages every minutes: 

zigbee2mqtt:warn 10/13/2019, 4:42:22 PM Device with modelID 'undefined' is not supported.
zigbee2mqtt:warn 10/13/2019, 4:42:22 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

I am using the Home Assistant addon of zigbee2mqtt v1.6.0. 

Can i set the modelID in the devices.yaml or the problem is with flashing ?

Thank you for your time.

AustinTurner, 2019-10-16:
I just assembled my CC2530 + CC2591 router as well, and I am having the same issues. Any updates on this would be appreciated. 

Router Firmware is also 2019-02.
zigbee2mqtt version is also 1.6.0

AustinTurner, 2019-10-16:
I have resolved the issue. For myself, at least.

I went through the re-pairing process for it. And now it is completely identified within my system.

"Re-pairing

CC2530, CC2531: Power on, wait 2 seconds, power off, repeat this cycle three times. CC2531: Power on, press and hold down the S2 button for 5 seconds."

So I just unplugged and plugged it back in about 3 different times, and now everything is working as intended. Could have just been a fluke. 

I hope this information helps you out @cslehel

cslehel, 2019-10-17:
Hello AustinTurner, 
did you removed the router from the devices list before the plug unplug precedure ?

vhristev, 2019-10-17:
Devices:

Old Pi3 + HassOS + HassIO + CC2531 without antenna ( coordinator)
New Pi4 + Rasberian + HassIO + CC2531 WITH antenna (coordinator)
Zigbee Router ( CC2530 + CC2591 )

Hello, guys recently I move from Pi3-CC2531 to Pi4 - CC2531- with an antenna. With Pi3 I configure CC2530 + CC2591 based on documentation and when I plug it into the wall socket after few minutes it appear

the problem is when i move to new Pi4 with new CC2531 with antenna i cannot see CC2530 + CC2591 ( router )

I check the voltage and I can see that CC2530 + CC2591 board get 5V which means power is fine

Then i realize a friend of mine gave me 5V 

HLK-PM01 = 5V
HLK-PM03 = 3V

I start from scratch, so I re-flash it successfully and then solder it to 3V but still does not show. 

I also tried the procedure below: 
Re-pairing CC2530/CC2530 router

СС2530 can be re-paired if you power on/power off the device three times (power on, wait 2 seconds, power off, repeat this cycle three times). 

I'm wondering is it possible to work fine with 5V at some point to become partially bricked. I'm still wondering if it can pass re-flash successful and it's partially bricked. Any idea how to troubleshoot. It seems i need to start with CC sniffing traffic.

cslehel, 2019-10-18:
Hello vhristev, 

you can try the coordinator with the antenna with the RPI3 to see if its working, you dont need to re-pair your devices. Maybe you have a problem with the coordinator.

vhristev, 2019-10-18:
Thank you for the proposal I plan tonight is to put CC with an antenna to my Pi3 to see what is going on if it's not working I will connect my old CC without antenna which will bring everything back how it was and if it's not connecting do you have any recommendations for next steps. As I said i don't think is dead because it was working fine with 5V. 

Could you give some troubleshooting ideas for "problem with coordinator" because i connect 5-6 EndDevices but for example 'vibrant' sensors i cannot add it.

cslehel, 2019-10-18:
I had a problem with the source routing version of coordinator firmware. 
The coordinator had 5 direct connections and i could not pair anything else with it.

vhristev, 2019-10-18:
OK maybe everytime i hit different scenarios which are a combination of cases. I will try to re-flash everything from scratch. 

- Re-flash  CC2531 ( default firm ) 
- Configure it with Zigbee2Mqtt
- Try to connect the router ( already flashed )
- Try to connect other devices 

For router, i tried both 'diag' and 'std' firmware. 
router-cc2530-cc2591-diag.hex
router-cc2530-cc2591-std.hex

For coordinator: CC2531ZNP-Prod.hex

cslehel, 2019-10-20:
Today i had time to try the plug unplug precedure and its workin correctly. 
I did not removed the router from the dveices list before the repair.

Thank you @AustinTurner

vhristev, 2019-10-22:
Tried 2 different USB's with default and router firmware on CC2531 and still router is not connecting. Not sure if its still alive what happen i dont know strange is that firmware check pass successfull 2 times and still cannot connect it.

cslehel, 2019-10-23:
How many zigbee devices do you have ?

vhristev, 2019-10-23:
I remove DB and State reinstall Zigbee2Mqtt addon for HASsIO and nothing router is not coming up and pairing devices is a hell. I'm trying to downgrade to 1.5.1 and use older HassIO version to check if its a software problem. With PI3 everything was working like a charm following step by step guides. When I move to Pi4 and upgrade to the latest HassIO and latest Zigbee2mqtt 1.6 version I hit those issues. 

All my sensors but for my working solution only 3 was paired ( 1 temp and 2 body sensors ) 
2 Body sensors
2 Temp sensors
2 Door
1 Vibration
2 Swithes

cslehel, 2019-10-23:
I have the latest HassIO and zigbee2mqtt on a virtual machine with the latest coordinator firmware (not the source routing) for cc2531 and the CC2530 CC2591 router with the latest firmware. Maybe the raspberry 4 has some problems with the coordinator connected to the usb.

vhristev, 2019-10-23:
I have some ideas to stop Philips hue hub and my wifi and then try again. They are a lot of noise where I live. Back in the days, i was not able to pair my NEST for some reason, so i unplug Hue hub and power off all my light bulbs then it was fine. Currently, i lose my apple trackpad connectivity to my mac very often and i think its cause by the noise around my flat. I scan with "NetStumber" and switch my 2.4 Wifi to channel 9 . Maybe its a good idea to start from reducing the noise but ... my automation was working on Pi3 

About Pi4 and USB: I tried 3 different ports issues are the same. I'm not sure but i will try with older firmware. I Also order another CC2530+CC2591. 

Back in the days i tried Osram smart switch to act as a router but it was not working correctly. End devices were connected to Osram plug but dont know why it didnt pass the message to CC2531.  I will find what is the root cause its a matter of time. 
```

core-ssh:~# ls -l /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0014B941BA-if00
crw-rw----    1 root     dialout   166,   0 Oct 23 02:21 /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0014B941BA-if00
```

vhristev, 2019-10-23:
Other question: 

What re-paring for USB CC2531 means? Why do i need to repair it? To what to re-pair it that is my coordinator 

I tried power off/on procedure for CC2530+ CC2591 but not sure if i trigger it correctly. 
What im doing is: 
CC2530+ CC2591 is connected to HLK-PM03
I disconnect the female pin from 3.3V HLK-PM03 ( power is off). I dont unplug it from a power socket. 
1st time   - Connect it back again wait 2 sec. and disconnect it 
2nd time - Connect it back again wait 2 sec. and disconnect it 
3rd time - Connect it back again wait 2 sec. and disconnect it  ( do i need to stop disconnecting here ) 
Connect it back again. 
I'm I doing it correct ( its super clear what i need to do i dont think i'm doing it wrong. 

Next step is: I have 2 CC2531 and one of them will become a sniffer and i will Wireshark the network I need to know if the router is working or not it seems there is not other way. 

Any high level steps how to start from scratch. 
Remove zigbee2mqtt HassIO addon
Remove everything from /share/zigbee2mqtt/ folder ( if its still there )
Flash CC2531 with default firmware
Flash CC2530+ CC2591 with router firmware 
Install zigbee2mqtt hassIO addon
Plug CC2531 into USB.  - Its a USB i dont need to reboot PI4 
Configure zigbee2mqtt ( config file with connection to broker )
Start zigbee2mqtt and check if coordinator is up
`  zigbee2mqtt:info 10/23/2019, 2:45:52 AM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"unknown","coordinator":20190608,"log_level":"info","permit_join":false}'`

My current config: 

```
{
  "data_path": "/share/zigbee2mqtt",
  "devices": "devices.yaml",
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": true,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://192.168.1.173:1883",
    "user": "MYCOOLUSER",
    "password": "MYSTRONGPASS"
  },
  "serial": {
    "port": "/dev/ttyACM0"
  },
  "advanced": {
    "pan_id": 6754,
    "channel": 11,
    "network_key": [
      1,
      3,
      5,
      7,
      9,
      11,
      13,
      15,
      0,
      2,
      4,
      6,
      8,
      10,
      12,
      13
    ],
    "availability_blacklist": []
  },
  "ban": [],
  "whitelist": [],
  "queue": {},
  "socat": {
    "enabled": false,
    "master": "pty,raw,echo=0,link=/dev/ttyZ2M,mode=777",
    "slave": "tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5",
    "restartdelay": 1,
    "initialdelay": 1,
    "options": "-d -d",
    "log": false
  }
}
``` 

I also generate my own key. 

Any advice for different channel instead of 11 like 25 ( i know it depends of my noise )

cslehel, 2019-10-23:
I think you have to re-pair or re-flash the coordinator if you change the network key, channel or you want to reset the routes. 
The on/off procedure is correct. Did you disconnected the vcc pin ? After the procedure you have to wait a bit, a minute or two.
I have the same config except the socat part and the permit_join. For the permit join i have a switch and a counter. 
If you move to wifi channel 9 you dont have to change the zigbee channel 11. 
https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html

I am sorry i couldn't help you more, i only started with zigbee 4 months ago.

vhristev, 2019-10-23:
I will create CC sniffer and see what is going on right now I'm blind i didnt even know if router is working on not. 

I think you have to re-pair or re-flash the coordinator if you change the network key, channel or you want to reset the routes. - I change my key at the moment i create my config it was long time ago. 

I didnt know i need to wait 1-2 mins after power off / on again. Actually I'm not sure if i disconnect the VCC pin but i also tried it a couple of times with power socket off / on. I will wait couple of mis and I will see what is going on. 

The main question is why end devices are so hard to pair now. with Pi 3 and old Coordinator was a matter of click now its a hell. I think with a Sniffer i can see if there is too much noise in the house.

cslehel, 2019-10-23:
If you power on/off from mains the power module can provide for a little time enough power to keep the router alive.

vhristev, 2019-10-23:
It seems people are suffering from same issue and i think i just hit all corner cases in my different variants. 

https://github.com/danielwelch/hassio-zigbee2mqtt/issues/226

vhristev, 2019-10-24:
I reflash CC2531 and power on my old Pi 3 with HassOS and HassIO - zigbee2mqtt 1.5.1 and with a press of a reset button on my Xiaomi temp and body sensor, everything is working like a charm paired instantly.  

It seems there is something weird with Pi 4 USB ports and CC2531 may be a driver ... I'm not sure. 

I also create a Sniffer and start sniffing the network this days i will spend more time there. 

Other issues: 
https://community.home-assistant.io/t/raspberry-pi-4-cc2531-zigbee2mqtt-xiaomi-sensors-setup-help-needed/133646/8 
https://github.com/danielwelch/hassio-zigbee2mqtt/issues/226

vhristev, 2019-10-24:
I will that is what we discuss in other thread.

closed 2019-10-24 by cslehel

vhristev, 2019-11-02:
**Solved** with USB extender and moving away CC2531 from Pi4

Mother of god.... I put extender and move away from the CC2531 USB and now everything is working like a charm. Even I able to connect my router at first it didnt show up but after 3 power off / on ( re-pair ) it shows up.

` zigbee2mqtt:info 11/3/2019, 12:22:04 AM New device 'lumi.router' with address 0x00124b00199ddad2 connected!
`

I connect 3 devices to my CC2531 USB and they pair with a single try its incredible how this extender fixes all my issues. It seems PI4 interference of radio signals and electrical components is very strong ... stronger than Pi3. 

Thanks for the help.

cslehel, 2019-11-03:
I am glad it worked out. 
Thank you for sharing your experience.

Esperk, 2019-11-09:
> **Solved** with USB extender and moving away CC2531 from Pi4
>
> Mother of god.... I put extender and move away from the CC2531 USB and now everything is working like a charm. Even I able to connect my [...]

It was turning me crazy! First time installing zigbee2mqtt on fresh hassio installation on RPI4. An usb extension cable did the trick, who would've thought that!

vhristev, 2019-11-13:
> It was turning me crazy! First time installing zigbee2mqtt on fresh hassio installation on RPI4. An usb extension cable did the trick, who would've thought that!

Me too I spend weeks troubleshooting even flash one of my drives and start watching Zigbee packegs with Wireshark.