Koenkk/zigbee2mqtt #2130: QBKG12LM decoupled	Button doesn’t control any relay no work

VOVAiS, 2019-10-13:
# Bug Report
## What happened
send to topic zigbee2mqtt/0x00158d0002d7a022/system/set
{"operation_mode": {
    "button": "left",
    "state": "decoupled"}
}
system log

> debug 2019-10-13T16:34:12: Received Zigbee message from '0x00158d00024862df', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 2 with groupID 0
at work (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:283:12))
at MqttClient._handlePacket (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:351:12)
at MqttClient._handlePublish (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:1162:12)
at MqttClient.emit (events.js:198:13)
at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.js:86:14)
at MQTT.emit (events.js:198:13)
at Controller.onMQTTMessage (/opt/zigbee2mqtt/lib/controller.js:239:14)
at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:320:44)
at EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/entityPublish.js:55:36)
at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:173:25)
error 2019-10-13T16:33:57: Failed to call 'EntityPublish' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Postfix 'system' is given but device has no such endpoint
}'
}
"state": "decoupled"
"button": "left",
"operation_mode": {
{
debug 2019-10-13T16:33:57: Received MQTT message on 'zigbee2mqtt/0x00158d0002d7a022/system/set' with data '
info 2019-10-13T16:33:51: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload 

## What did you expect to happen
decoupled button left 
## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.6.0 commit "4abafa9"
CC253X firmware version: 20190706

VOVAiS, 2019-10-21:
> info  2019-10-21T10:19:10: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"0527b89","coordinator":{"type":"zStack30x","meta":{"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190425}},"log_level":"debug","permit_join":false}'
debug 2019-10-21T10:19:32: Received MQTT message on 'zigbee2mqtt/0x00158d0002d7a022/system/set' with data '
{
"operation_mode": {
"button": "left",
"state": "decoupled"
}
}'
error 2019-10-21T10:19:32: Failed to call 'EntityPublish' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Postfix 'system' is given but device has no such endpoint
    at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:173:25)
    at EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/entityPublish.js:55:36)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:320:44)
    at Controller.onMQTTMessage (/opt/zigbee2mqtt/lib/controller.js:239:14)
    at MQTT.emit (events.js:198:13)
    at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.js:86:14)
    at MqttClient.emit (events.js:198:13)
    at MqttClient._handlePublish (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:1162:12)
    at MqttClient._handlePacket (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:351:12)
    at work (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:283:12))
debug 2019-10-21T10:19:59: Received Zigbee message from '0x00158d0002d7a022', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2019-10-21T10:19:59: MQTT publish: topic 'zigbee2mqtt/0x00158d0002d7a022', payload '{"state_left":"OFF","linkquality":73,"power":0,"consumption":0,"temperature":30,"state_right":"OFF"}'
info  2019-10-21T10:19:59: MQTT publish: topic 'zigbee2mqtt/0x00158d0002d7a022', payload '{"state_left":"OFF","linkquality":73,"power":0,"consumption":0,"temperature":30,"state_right":"OFF","click":""}'
debug 2019-10-21T10:20:01: Received Zigbee message from '0x00158d00024862df', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 2 with groupID 0
info  2019-10-21T10:20:01: MQTT publish: topic 'zigbee2mqtt/0x00158d00024862df', payload '{"state_left":"OFF","linkquality":57,"state_right":"OFF","power":0,"consumption":242.41,"temperature":27}'
info  2019-10-21T10:20:01: MQTT publish: topic 'zigbee2mqtt/0x00158d00024862df', payload '{"state_left":"OFF","linkquality":57,"state_right":"OFF","power":0,"consumption":242.41,"temperature":27,"click":""}'
debug 2019-10-21T10:20:24: Received Zigbee message from '0x00158d00024862df', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2019-10-21T10:20:24: MQTT publish: topic 'zigbee2mqtt/0x00158d00024862df', payload '{"state_left":"OFF","linkquality":57,"state_right":"OFF","power":0,"consumption":242.41,"temperature":27}'
info  2019-10-21T10:20:24: MQTT publish: topic 'zigbee2mqtt/0x00158d00024862df', payload '{"state_left":"OFF","linkquality":57,"state_right":"OFF","power":0,"consumption":242.41,"temperature":27,"click":""}'
debug 2019-10-21T10:20:25: Received MQTT message on 'zigbee2mqtt/bridge/config/log_level' with data 'error'
info  2019-10-21T10:20:25: Switching log level to 'error'

VOVAiS, 2019-10-26:
Dear @Koenkk have you seen my message?

Koenkk, 2019-10-26:
Is this with the latest dev branch?

VOVAiS, 2019-10-27:
yes

> info  2019-10-27T05:30:53: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"cb3ec45","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}},"log_level":"debug","permit_join":false}'
debug 2019-10-27T05:30:58: Received MQTT message on 'zigbee2mqtt/0x00158d00024862df/system/set' with data '
{
"operation_mode": {
"button": "left",
"state": "decoupled"
}
}'
error 2019-10-27T05:30:58: Failed to call 'EntityPublish' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Postfix 'system' is given but device has no such endpoint
    at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:174:25)
    at EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/entityPublish.js:55:36)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:320:44)
    at Controller.onMQTTMessage (/opt/zigbee2mqtt/lib/controller.js:239:14)
    at MQTT.emit (events.js:198:13)
    at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.js:86:14)
    at MqttClient.emit (events.js:198:13)
    at MqttClient._handlePublish (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:1162:12)
    at MqttClient._handlePacket (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:351:12)
    at work (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:283:12))

Koenkk, 2019-10-27:
Instead of system, can you try with left or right?

VOVAiS, 2019-10-27:
debug 2019-10-27T05:31:04: Received MQTT message on 'zigbee2mqtt/0x00158d00024862df/system/set' with data '
{
"operation_mode": {
"button": "right",
"state": "control_right_relay"
}
}'
error 2019-10-27T05:31:04: Failed to call 'EntityPublish' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Postfix 'system' is given but device has no such endpoint
    at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:174:25)
    at EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/entityPublish.js:55:36)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:320:44)
    at Controller.onMQTTMessage (/opt/zigbee2mqtt/lib/controller.js:239:14)
    at MQTT.emit (events.js:198:13)
    at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.js:86:14)
    at MqttClient.emit (events.js:198:13)
    at MqttClient._handlePublish (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:1162:12)
    at MqttClient._handlePacket (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:351:12)
    at work (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:283:12))
debug 2019-10-27T05:31:16: Received MQTT message on 'zigbee2mqtt/bridge/config/log_level' with data 'error'
info  2019-10-27T05:31:16: Switching log level to 'error'

matlab22, 2019-10-27:
I tried to get the state but got the same error:

```
Oct 27 11:46:00 loxberry npm[1321]: zigbee2mqtt:error 2019-10-27T10:46:00: Failed to call 'EntityPublish' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Postfix 'system' is given but device defines no endpoints
Oct 27 11:46:00 loxberry npm[1321]:     at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:172:25)
Oct 27 11:46:00 loxberry npm[1321]:     at EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/entityPublish.js:55:36)
Oct 27 11:46:00 loxberry npm[1321]:     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:320:44)
Oct 27 11:46:00 loxberry npm[1321]:     at Controller.onMQTTMessage (/opt/zigbee2mqtt/lib/controller.js:239:14)
Oct 27 11:46:00 loxberry npm[1321]:     at MQTT.emit (events.js:209:13)
Oct 27 11:46:00 loxberry npm[1321]:     at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.js:86:14)
Oct 27 11:46:00 loxberry npm[1321]:     at MqttClient.emit (events.js:209:13)
Oct 27 11:46:00 loxberry npm[1321]:     at MqttClient._handlePublish (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:1162:12)
Oct 27 11:46:00 loxberry npm[1321]:     at MqttClient._handlePacket (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:351:12)
Oct 27 11:46:00 loxberry npm[1321]:     at work (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:283:12))
```

Koenkk, 2019-10-27:
Publish to zigbee2mqtt/0x00158d00024862df/right/set instead of zigbee2mqtt/0x00158d00024862df/system/set

VOVAiS, 2019-10-27:
if you give the command 
zigbee2mqtt/0x00158d0002d7a022/left/set

{
"operation_mode": {
"button": "left",
"state": "decoupled"
}
}
the left key is decoupled from the relay

if you give the command 
zigbee2mqtt/0x00158d0002d7a022/right/set
{
"operation_mode": {
"button": "right",
"state": "decoupled"
}
}

error error 2019-10-27T11:30:01: Publish 'set' 'operation_mode' to '0x00158d0002d7a022' failed: 'Error: Timeout - 59097 - 2 - 139 - 4 after 10000ms'

VOVAiS, 2019-10-27:
when controlling the relay I sometimes get errors

> error 2019-10-27T08:50:32: Publish 'set' 'state' to '0x00158d0002d7a022' failed: 'Error: Timeout - 59097 - 2 - 119 - 11 after 15000ms'
> error 2019-10-27T09:00:16: Publish 'set' 'state' to '0x00158d0002d7a022' failed: 'Error: Timeout - 59097 - 2 - 252 - 11 after 15000ms'
> error 2019-10-27T09:28:34: Publish 'set' 'state' to '0x00158d0002d7a022' failed: 'Error: Timeout - 59097 - 1 - 232 - 11 after 15000ms'
> error 2019-10-27T09:40:01: Publish 'set' 'state' to '0x00158d0002d7a022' failed: 'Error: Data request failed with error: 'MAC no ack' (233)'
> error 2019-10-27T09:44:37: Publish 'set' 'state' to '0x00158d0002d7a022' failed: 'Error: Data request failed with error: 'MAC no ack' (233)'

Koenkk, 2019-10-28:
When these errors happen, does the relay still changes it state?

VOVAiS, 2019-10-28:
on error
Publish 'set' 'state' to '0x00158d0002d7a022' failed: 'Error: Timeout - 59097 - 1 - 232 - 11 after 15000ms'
the relay changes its state with a large delay

on error
Publish 'set' 'state' to '0x00158d0002d7a022' failed: 'Error: Data request failed with error: 'MAC no ack' (233)'
not ready to answer.

Koenkk, 2019-10-28:
What do you mean with large delay, 1 second, 10 seconds?

Is your CC2531 connected through am extension cable? (https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html)

VOVAiS, 2019-10-29:
a delay of about 7-10 seconds. Yes I use an extension cable

directman66, 2019-11-16:
Hey. QBKG03LM switches could previously be controlled through the topics 
zigbee2mqtt/0x00158d00022bc3ab/left/set by sending the payload {"**state_left**": "OFF"} and zigbee2mqtt/0x00158d00022bc3ab/right/set by sending the payload {"**state_left**": "OFF"}. 

Everything worked well. But after updating to version 1.7.0 the management topics changed. Now you can control it only by sending zigbee2mqtt/0x00158d00022bc3ab/right/set by sending the payload {"**state**": "OFF"}. I have already adapted the code and control methods when the result of the command corresponds to the command being sent. I do not like that sending the state parameter I see the result in the state_right topic. Can I add an old control option in addition to the existing one?

HAKostya, 2019-11-23:
> Publish to zigbee2mqtt/0x00158d00024862df/right/set instead of zigbee2mqtt/0x00158d00024862df/system/set

how would the topic look like for a single button QBKG11LM switch? Documentation https://www.zigbee2mqtt.io/devices/QBKG11LM.html still refers to zigbee2mqtt/[FRIENDLY_NAME]/system/get
and it's not working:
`error 2019-11-23T20:12:24: Failed to call 'EntityPublish' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Postfix 'system' is given but device defines no endpoints
    at Zigbee.resolveEntity (/app/lib/zigbee.js:173:25)
    at EntityPublish.onMQTTMessage (/app/lib/extension/entityPublish.js:55:36)
    at Controller.callExtensionMethod (/app/lib/controller.js:320:44)
    at Controller.onMQTTMessage (/app/lib/controller.js:239:14)
    at MQTT.emit (events.js:198:13)
    at MQTT.onMessage (/app/lib/mqtt.js:86:14)
    at MqttClient.emit (events.js:198:13)
    at MqttClient._handlePublish (/app/node_modules/mqtt/lib/client.js:1162:12)
    at MqttClient._handlePacket (/app/node_modules/mqtt/lib/client.js:351:12)
    at work (/app/node_modules/mqtt/lib/client.js:283:12))`

VOVAiS, 2019-11-24:
zigbee2mqtt/[FRIENDLY_NAME]/single/get

HAKostya, 2019-11-24:
Tried that - no luck :(
zigbee2mqtt/kidsroom_switch/single/get
getting this in z2m logs:

> info  2019-11-24T18:07:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":{"friendly_name":"kidsroom_switch/single"}}'
error 2019-11-24T18:07:43: Entity 'kidsroom_switch/single' is unknown

VOVAiS, 2019-11-24:
i assume on QBKG11LM, there is no endpoint
`endpoint: (device) => {
            return {'system': 1, 'default': 2};`
https://github.com/Koenkk/zigbee-herdsman-converters/blob/4b55f49968e6f9c9b96e2a6ee3326d93927c9e31/devices.js#L474

I guess on QBKG12LM, supposed to look like endpoint
```
endpoint: (device) => {
            return {'system': 1, 'left': 2, 'right': 3};
```
https://github.com/Koenkk/zigbee-herdsman-converters/blob/4b55f49968e6f9c9b96e2a6ee3326d93927c9e31/devices.js#L504

HAKostya, 2019-11-24:
is this a bug? or not yet implemented? 
Interesting thing is that I'm able to set 'decoupled' mode like this:

> zigbee2mqtt/kidsroom_switch/set
{
    "operation_mode": {
    "button": "single",
    "state": "decoupled"
    }
}
>
but I'm not able to get current mode.

VOVAiS, 2019-11-25:
it's a bug. Waiting for when the author will correct

Koenkk, 2019-11-25:
You should be able to get the mode by publishing to `zigbee2mqtt/kidsroom_switch/get` payload `{"operation_mode": ""}`

HAKostya, 2019-11-25:
> {"operation_mode": ""}

no dice :(

> error 2019-11-25T20:06:28: Publish 'get' 'operation_mode' to 'kidsroom_switch' failed: 'Error: Cluster 'genBasic' has no attribute '65314''

> info  2019-11-25T20:06:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'get' 'operation_mode' to 'kidsroom_switch' failed: 'Error: Cluster 'genBasic' has no attribute '65314''","meta":{"friendly_name":"kidsroom_switch"}}'

Koenkk, 2019-11-26:
That was a bug, should be fixed in latest zigbee2mqtt dev version.

VOVAiS, 2019-11-27:
no work

`error 2019-11-27 16:47:37: Failed to call 'EntityPublish' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Postfix 'system' is given but device has no such endpoint
    at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:175:25)
    at EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/entityPublish.js:55:36)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:320:44)
    at Controller.onMQTTMessage (/opt/zigbee2mqtt/lib/controller.js:239:14)
    at MQTT.emit (events.js:198:13)
    at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.js:86:14)
    at MqttClient.emit (events.js:198:13)
    at MqttClient._handlePublish (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:1162:12)
    at MqttClient._handlePacket (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:351:12)
    at work (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:283:12))
error 2019-11-27 16:47:42: Failed to call 'EntityPublish' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Postfix 'system' is given but device has no such endpoint
    at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:175:25)
    at EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/entityPublish.js:55:36)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:320:44)
    at Controller.onMQTTMessage (/opt/zigbee2mqtt/lib/controller.js:239:14)
    at MQTT.emit (events.js:198:13)
    at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.js:86:14)
    at MqttClient.emit (events.js:198:13)
    at MqttClient._handlePublish (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:1162:12)
    at MqttClient._handlePacket (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:351:12)
    at work (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:283:12))`

version:"1.7.1+dev","commit":"2fcddb5"

Koenkk, 2019-11-27:
@VOVAiS use either left or right instead of system, for QBKG11LM  use nothing.

VOVAiS, 2019-11-27:
in the left works.
In the right error
> error 2019-11-27 17:39:51: Publish 'set' 'operation_mode' to '0x00158d00024862df' failed: 'Error: Timeout - 35125 - 2 - 155 - 0 - 4 after 10000ms'

Koenkk, 2019-11-27:
Sorry I was wrong, should work with

`zigbee2mqtt/[FRIENDLY_NAME]/get`, payload 

```js
{"operation_mode": {"button": "right"}}
```

VOVAiS, 2019-11-28:
Thank you, it works . need to update instruction www.zigbee2mqtt.io

closed 2019-11-28 by VOVAiS

Koenkk, 2019-11-28:
@VOVAiS done, thanks all!

captaindaveman, 2019-12-12:
hi there, is this issue also resolved for the QBKG04LM single switch? I'm still having issues using prod or dev branches?