Koenkk/zigbee2mqtt #2131: Adding support for new Philips Hue Filament (Bluetooth)

jordy55, 2019-10-13:
Was looking for adding support for Philips hue filament G93 (LWO001).
Unfortunately I can not find anything about this lamp regarding the settings for device.js.
There are also several versions. So you also have the ST64 and A60.

Maybe someone here has already done it?

Koenkk, 2019-10-13:
Its already supported in the dev branch.

jordy55, 2019-10-13:
My excuse, I didn't know this. Is there a way that I can check this next time? I am currently doing the following: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js

Then I look for the device. Currently I only see the A60 there

Koenkk, 2019-10-14:
For now you need to check: https://github.com/Koenkk/zigbee-shepherd-converters/blob/zigbee-herdsman-converters/devices.js

jordy55, 2019-10-27:
Sorry, but the link above doesn't work.

Today I removed and reinstalled Zigbee. Unfortunately, it does not yet recognize the Philips Hue Filament lamp (G93 / LWO001). I just looked in the link that I sent, but I don't see it there either. Do you have an idea when it will work?

Koenkk, 2019-10-28:
I was wrong it's not supported yet, can you share your `database.db` when paired?

jordy55, 2019-10-31:
Sorry, didn't see the post.
Hereby my database.db:

```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b001936797c","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,11],"status":"online","joinTime":1569178143,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}}}}},"_id":"7AsSp4ia9CLLLVaL"}
{"id":2,"type":"Router","ieeeAddr":"0x00124b001c85f438","nwkAddr":4129,"manufId":0,"manufName":"3A Smart Home DE","powerSource":"Mains (single phase)","modelId":"NUET56-DL27LX1.1","epList":[11,13],"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"swBuildId":"1.0.2","status":"offline","joinTime":null,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attributes":{}},"genIdentify":{"dir":{"value":1},"attributes":{}},"genGroups":{"dir":{"value":1},"attributes":{}},"genScenes":{"dir":{"value":1},"attributes":{}},"genOnOff":{"dir":{"value":1},"attributes":{}},"genLevelCtrl":{"dir":{"value":1},"attributes":{}},"lightingColorCtrl":{"dir":{"value":1},"attributes":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attributes":{}}}}},"dateCode":"20160422","_id":"Fpt2bYLsp9815m3H"}
{"id":3,"type":"Router","ieeeAddr":"0x00124b001c88a0bc","nwkAddr":22981,"manufId":0,"manufName":"3A Smart Home DE","powerSource":"Mains (single phase)","modelId":"NUET56-DL27LX1.1","epList":[11,13],"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"swBuildId":"1.0.2","status":"offline","joinTime":null,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attributes":{}},"genIdentify":{"dir":{"value":1},"attributes":{}},"genGroups":{"dir":{"value":1},"attributes":{}},"genScenes":{"dir":{"value":1},"attributes":{}},"genOnOff":{"dir":{"value":1},"attributes":{}},"genLevelCtrl":{"dir":{"value":1},"attributes":{}},"lightingColorCtrl":{"dir":{"value":1},"attributes":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attributes":{}}}}},"dateCode":"20160422","_id":"i7EHIzqZGBTNwj8q"}
{"id":4,"type":"Router","ieeeAddr":"0x00124b001c88935e","nwkAddr":45062,"manufId":0,"manufName":"3A Smart Home DE","powerSource":"Mains (single phase)","modelId":"NUET56-DL27LX1.1","epList":[11,13],"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"swBuildId":"1.0.2","status":"offline","joinTime":null,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attributes":{}},"genIdentify":{"dir":{"value":1},"attributes":{}},"genGroups":{"dir":{"value":1},"attributes":{}},"genScenes":{"dir":{"value":1},"attributes":{}},"genOnOff":{"dir":{"value":1},"attributes":{}},"genLevelCtrl":{"dir":{"value":1},"attributes":{}},"lightingColorCtrl":{"dir":{"value":1},"attributes":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attributes":{}}}}},"dateCode":"20160422","_id":"5AWR6yvFBfkFNoNT"}
{"id":5,"type":"EndDevice","ieeeAddr":"0x00158d00032c104d","nwkAddr":16048,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion.aq2","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,65535,1030,1024,1280,1,3],"outClusterList":[0,25],"clusters":{"msIlluminanceMeasurement":{"attributes":{"measuredValue":0}},"msOccupancySensing":{"attributes":{"occupancy":1}},"genBasic":{"attributes":{"65281":{"1":3015,"3":23,"4":17320,"5":17,"6":[0,12],"10":4129,"11":0,"100":0}}}}}},"appVersion":5,"stackVersion":2,"hwVersion":1,"dateCode":"20170627","swBuildId":"3000-0001","zclVersion":1,"interviewCompleted":true,"meta":{},"_id":"vHNsFo4elEP8h5SN"}
{"id":6,"type":"EndDevice","ieeeAddr":"0x00158d00027aee41","nwkAddr":30543,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.weather","epList":[1],"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":30,"swBuildId":"3000-0001","status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,65535,1026,1027,1029],"outClusterList":[0,4,65535],"clusters":{"genBasic":{"dir":{"value":3},"attributes":{"65281":{"1":2945,"4":5032,"5":8,"6":[0,1],"10":0,"100":2043,"101":6601,"102":101611}}},"genIdentify":{"dir":{"value":1},"attributes":{}},"genGroups":{"dir":{"value":2},"attributes":{}},"msTemperatureMeasurement":{"dir":{"value":1},"attributes":{"measuredValue":2046}},"msPressureMeasurement":{"dir":{"value":1},"attributes":{"16":10161,"20":- [... cut ...] "epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8,4096,64514],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LWO001","manufacturerName":"Philips","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":2,"stackVersion":1,"zclVersion":2,"interviewCompleted":false,"meta":{}}
```

The last added lamp is the filament bulb (LWO001)

andy778, 2019-11-01:
I have one also and this is from database.db..  

{"id":3,"type":"Router","ieeeAddr":"0x0017880106d0f413","nwkAddr":52044,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"LWA001","epList":[11,242],"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":1,"swBuildId":"1.53.3_r27175","status":"online","joinTime":1572635546,"endpoints":{"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8,4096,64514],"outClusterList":[25],"clusters":{"64514":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{"9":255,"10":{},"11":"www.meethue.com","65533":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":1,"manufacturerName":"Philips","modelId":"LWA001","dateCode":"20190122","powerSource":1,"appProfileVersion":0,"swBuildId":"1.53.3_r27175"}},"genIdentify":{"dir":{"value":1},"attrs":{"65533":1,"identifyTime":178}},"genGroups":{"dir":{"value":1},"attrs":{"65533":1,"nameSupport":0}},"genScenes":{"dir":{"value":1},"attrs":{"65533":1,"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"16387":1,"65533":1,"onOff":1,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"15":0,"16384":254,"65533":1,"currentLevel":254,"remainingTime":0}},"genOta":{"dir":{"value":2},"attrs":{}},"lightLink":{"dir":{"value":1},"attrs":{"65533":1}}}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{"33":{"dir":{"value":2},"attrs":{}}}}},"dateCode":"20190122","_id":"oHgnaoriPbdG5NC9"}

Koenkk, 2019-11-01:
@andy778 `LWA001` is already supported in the dev branch

@jordy55 yours has also been added 4 days ago (https://github.com/Koenkk/zigbee-herdsman-converters/commit/072126952e952c58b0677f600f0d7078d31086c3). Also supported in latest dev branch.

closed 2019-11-01 by Koenkk