Koenkk/zigbee2mqtt #2132: Gledopto GL-C-008 does not get configured

Kryzek, 2019-10-13:
When starting Z2MQTT (dev #4abafa9) I get this
```
zigbee2mqtt:debug 2019-10-13T17:14:25: Received Zigbee message from '0x000b57fffe24fd6a', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2019-10-13T17:14:26: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe24fd6a', payload '{"brightness":172,"linkquality":0,"state":"OFF"}'
zigbee2mqtt:error 2019-10-13T17:14:32: Failed to configure '0x00158d0001d0f272', attempt 1 (Error: AREQ - ZDO - bindRsp after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10))
zigbee2mqtt:debug 2019-10-13T17:14:32: Setup reporting for '0x00124b001d3ab1b9' - 11 - genOnOff
zigbee2mqtt:error 2019-10-13T17:14:43: Failed to setup reporting for '0x00124b001d3ab1b9' - Error: Timeout - 14071 - 11 - 2 - 7 after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
zigbee2mqtt:debug 2019-10-13T17:15:22: Received Zigbee message from '0x000b57fffe8be58f', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2019-10-13T17:15:22: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe8be58f', payload '{"state":"ON","linkquality":21,"brightness":70}'
zigbee2mqtt:debug 2019-10-13T17:15:22: Received MQTT message on 'zigbee2mqtt/0x00158d0001d0f272/set' with data '{"state": "OFF"}'
zigbee2mqtt:debug 2019-10-13T17:15:22: Publishing 'set' 'state' to '0x00158d0001d0f272'
zigbee2mqtt:error 2019-10-13T17:15:23: Publish 'set' 'state' to '0x00158d0001d0f272' failed: 'Error: Data request failed with error: 'No network route' (205)'
```

Device can be controlled as normal.

Koenkk, 2019-10-14:
Do you have a custom zigbee-herdsman-converters? I don't see a configure section here: https://github.com/Koenkk/zigbee-shepherd-converters/blob/zigbee-herdsman-converters/devices.js#L2722

Kryzek, 2019-10-14:
Device 0x00124b001d3ab1b9 Zigbee LED controller RGB + CCT (GL-C-008)

I do not have custom converters for this device, it is exactly like described above, and there is indeed no configure section. But still gives I get error "Failed to setup reporting for '0x00124b001d3ab1b9'" on Z2M startup.

I double checked this from logs:
```
0x00124b001d3ab1b9 (0x00124b001d3ab1b9): GL-C-008 - Gledopto Zigbee LED controller RGB + CCT (Router)
Setup reporting for '0x00124b001d3ab1b9' - 11 - genOnOff
Failed to setup reporting for '0x00124b001d3ab1b9' - Error: Timeout - 14071 - 11 - 2 - 7 after 10000ms
```

Koenkk, 2019-10-15:
Gledopto devices don't support reporting, I've disabled it in the latest dev. Should be fixed now.

closed 2019-10-17 by Koenkk