Koenkk/zigbee2mqtt #2133: Can't pair Xiaomi devices

JHeijting, 2019-10-13:
I am new to HASS.io and zigbee2mqtt. I have spend the day installing zigbee2mqtt and trying to pair devices. Unfortunately I am unable to pair any device and I have tried lots of fixes I have found on the internet. Like trying another USB port, using a usb cable, pairing while really close to the coordinator, reflashing the CC2531 USB stick etc.

I hope someone can assist me in pairing my Xiaomi devices.

I have the following setup:
- hass.io 0.100.2 on a Debian VM
- Mosquitto broker v5.1
- zigbee2mqtt 1.6.0.
- CC2531 is flashed with "coordinator":20190608

Also I have flashed the CC2531 with the sniffer once to test if it is able to see the Xiaomi switches. This works.

In the configuration.yaml I have the following settings:
`mqtt:
    broker: core-mosquitto
    #port: 1883
    username: username
    password: password
    discovery: true
    discovery_prefix: homeassistant
    birth_message:
        topic: 'hass/status'
        payload: 'online'
    will_message:
        topic: 'hass/status'
        payload: 'offline'`

The zigbee2mqtt config is as follows:
`{
  "data_path": "/share/zigbee2mqtt",
  "devices": "devices.yaml",
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": true,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://core-mosquitto",
    "user": "user",
    "password": "password"
  },
  "serial": {
    "disable_led": true,
    "port": "/dev/ttyACM0"
  },
  "advanced": {
    "log_level": "debug",
    "pan_id": 6754,
    "channel": 11,
    "network_key": [
      1,
      3,
      5,
      7,
      9,
      11,
      13,
      15,
      0,
      2,
      4,
      6,
      8,
      10,
      12,
      13
    ],
    "availability_blacklist": []
  },
  "ban": [],
  "whitelist": [],
  "queue": {}
}`

This is the log:
./run.sh: line 16: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
2019-10-13T17:59:07: PM2 log: Launching in no daemon mode
2019-10-13T17:59:07: PM2 log: App [npm:0] starting in -fork mode-
2019-10-13T17:59:07: PM2 log: App [npm:0] online
> zigbee2mqtt@1.6.0 start /zigbee2mqtt-1.6.0
> node index.js
  zigbee2mqtt:info 10/13/2019, 5:59:08 PM Logging to directory: '/share/zigbee2mqtt/log/2019-10-13.17-59-07'
  zigbee2mqtt:debug 10/13/2019, 5:59:08 PM Removing old log directory '/share/zigbee2mqtt/log/2019-10-13.16-59-40'
  zigbee2mqtt:debug 10/13/2019, 5:59:08 PM Loaded state from file /share/zigbee2mqtt/state.json
  zigbee2mqtt:debug 10/13/2019, 5:59:08 PM Saving state to file /share/zigbee2mqtt/state.json
  zigbee2mqtt:info 10/13/2019, 5:59:08 PM Starting zigbee2mqtt version 1.6.0 (commit #unknown)
  zigbee2mqtt:info 10/13/2019, 5:59:08 PM Starting zigbee-shepherd
  zigbee2mqtt:debug 10/13/2019, 5:59:08 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/share/zigbee2mqtt/database.db","coordBackupPath":"/share/zigbee2mqtt/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:info 10/13/2019, 5:59:10 PM zigbee-shepherd started
  zigbee2mqtt:info 10/13/2019, 5:59:10 PM Coordinator firmware version: '20190608'
  zigbee2mqtt:debug 10/13/2019, 5:59:10 PM zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":"11","panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"ieeeAddr":"0x00124b000be8a7f9","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20190608},"startTime":1570989550,"joinTimeLeft":0}
  zigbee2mqtt:info 10/13/2019, 5:59:10 PM Currently 0 devices are joined:
  zigbee2mqtt:warn 10/13/2019, 5:59:10 PM `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 10/13/2019, 5:59:10 PM Allowing new devices to join.
  zigbee2mqtt:warn 10/13/2019, 5:59:10 PM Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 10/13/2019, 5:59:10 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 10/13/2019, 5:59:10 PM Connecting to MQTT server at mqtt://core-mosquitto
  zigbee2mqtt:info 10/13/2019, 5:59:10 PM zigbee-shepherd ready
  zigbee2mqtt:debug 10/13/2019, 5:59:10 PM Mounted the cieApp (epId 11)
  zigbee2mqtt:info 10/13/2019, 5:59:11 PM Connected to MQTT server
  zigbee2mqtt:info 10/13/2019, 5:59:11 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 10/13/2019, 5:59:11 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"unknown","coordinator":20190608,"log_level":"debug","permit_join":true}'
  zigbee2mqtt:info 10/13/2019, 6:01:51 PM Successfully reenabled joining

JHeijting, 2019-10-13:
Just tried the dev version of zigbee2mqtt. Unforntunately also unable to pair Xiaomi switches with this version.

JHeijting, 2019-10-14:
I am feeling pretty stupid right now... Obviously I had to disconnect the offical Aqarq hub since it is real close to my setup with Z2B. After this, I am receiving warning messages and error immidiately when trying to pair! So this is good :)

Weird thing is though, I am trying to pair the Xiaomi WXKG03LM (Aqara single key wireless wall switch), which should be supported. However I receive these messages in Z2B log:
zigbee2mqtt:info  2019-10-14T06:45:09: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"3e75108741a0d7328b278a8c26daaf8a5adf0934","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}'
zigbee2mqtt:info  2019-10-14T06:45:09: MQTT publish: topic 'homeassistant/sensor/0x00158d000249d1a4/click/config', payload '{"icon":"mdi:toggle-switch","value_template":"{{ value_json.click }}","state_topic":"zigbee2mqtt/0x00158d000249d1a4","json_attributes_topic":"zigbee2mqtt/0x00158d000249d1a4","name":"0x00158d000249d1a4_click","unique_id":"0x00158d000249d1a4_click_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d000249d1a4"],"name":"0x00158d000249d1a4","sw_version":"Zigbee2mqtt 1.6.0","model":"Aqara single key wireless wall switch (WXKG03LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2019-10-14T06:45:09: MQTT publish: topic 'homeassistant/sensor/0x00158d000249d1a4/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x00158d000249d1a4","json_attributes_topic":"zigbee2mqtt/0x00158d000249d1a4","name":"0x00158d000249d1a4_battery","unique_id":"0x00158d000249d1a4_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d000249d1a4"],"name":"0x00158d000249d1a4","sw_version":"Zigbee2mqtt 1.6.0","model":"Aqara single key wireless wall switch (WXKG03LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2019-10-14T06:45:09: MQTT publish: topic 'homeassistant/sensor/0x00158d000249d1a4/action/config', payload '{"icon":"mdi:gesture-double-tap","value_template":"{{ value_json.action }}","state_topic":"zigbee2mqtt/0x00158d000249d1a4","json_attributes_topic":"zigbee2mqtt/0x00158d000249d1a4","name":"0x00158d000249d1a4_action","unique_id":"0x00158d000249d1a4_action_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d000249d1a4"],"name":"0x00158d000249d1a4","sw_version":"Zigbee2mqtt 1.6.0","model":"Aqara single key wireless wall switch (WXKG03LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2019-10-14T06:45:09: MQTT publish: topic 'homeassistant/sensor/0x00158d000249d1a4/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00158d000249d1a4","json_attributes_topic":"zigbee2mqtt/0x00158d000249d1a4","name":"0x00158d000249d1a4_linkquality","unique_id":"0x00158d000249d1a4_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d000249d1a4"],"name":"0x00158d000249d1a4","sw_version":"Zigbee2mqtt 1.6.0","model":"Aqara single key wireless wall switch (WXKG03LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2019-10-14T06:45:15: MQTT publish: topic 'zigbee2mqtt/0x00158d000249d1a4', payload '{"click":"single","linkquality":81}'
zigbee2mqtt:info  2019-10-14T06:45:15: MQTT publish: topic 'zigbee2mqtt/0x00158d000249d1a4', payload '{"linkquality":81,"click":""}'
zigbee2mqtt:warn  2019-10-14T06:45:22: Device '0x00158d000249d1a4' left the network
zigbee2mqtt:info  2019-10-14T06:45:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x00158d000249d1a4"}}'
zigbee2mqtt:info  2019-10-14T06:45:23: Device '0x00158d000249d1a4' joined
zigbee2mqtt:info  2019-10-14T06:45:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00158d000249d1a4"}}'
zigbee2mqtt:info  2019-10-14T06:45:23: Starting interview of '0x00158d000249d1a4'
zigbee2mqtt:info  2019-10-14T06:45:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00158d000249d1a4"}}'
zigbee2mqtt:warn  2019-10-14T06:45:24: Received message from unsupported device with Zigbee model 'lumi.sensor_86sw1'
zigbee2mqtt:warn  2019-10-14T06:45:24: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2019-10-14T06:45:25: Received message from unsupported device with Zigbee model 'lumi.sensor_86sw1'
zigbee2mqtt:warn  2019-10-14T06:45:25: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2019-10-14T06:45:27: Received message from unsupported device with Zigbee model 'lumi.sensor_86sw1'
zigbee2mqtt:warn  2019-10-14T06:45:27: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn  2019-10-14T06:45:30: Received message from unsupported device with Zigbee model 'lumi.sensor_86sw1'
zigbee2mqtt:warn  2019-10-14T06:45:30: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:error 2019-10-14T06:45:35: Failed to interview '0x00158d000249d1a4', device has not successfully been paired

tb-killa, 2019-10-14:
It's on https://github.com/Koenkk/zigbee-shepherd-converters/blob/74b426e30e6f9db6ec954beb927b6ae0cb2fe936/devices.js#L412
@Koenkk doesn't you regex the name part? If I add the name themselve as standalone like ``lumi.sensor_86sw1`` it work for me too.

L3H0, 2019-10-14:
I have the same problem, i don`t remember but with 0.6 or older version herdsman-dev all was ok.
But now i it's impossible to pair those switch.

JHeijting, 2019-10-14:
So what's the best thing to do now? I suppose I shouldn't add the device as 'new device' following these instructions: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices_on_hassio.html

Since the device should be supported. Is there anything I can do to help fixing this? Also regarding the new herdsman?

Koenkk, 2019-10-14:
Fixed in the latest dev branch. @JHeijting just update to the latest dev and it should automatically work.

closed 2019-10-14 by Koenkk

JHeijting, 2019-10-14:
Hi Koenkk, thank you for your fast reply and update! You just beat me to it. I have reïnstalled the production plugin and the switch made a connection at first try! So at this moment I am not using the DEV version. Is it possible to use them along side each other? Just by stopping one and start the other?

Koenkk, 2019-10-14:
Yes thats possible