Koenkk/zigbee2mqtt #2135: Restart Z2M from HA ?

Bastiencc, 2019-10-14:
Hi

Is there a way to restart zigbee2mqtt from HA ?

Thanks

lolorc, 2019-10-14:
as far as I know there's no mqtt command to stop z2m, it would be nice to add one, in that case systemd would restart it.
but in the mean time can't you control systemd services from HA ?

Koenkk, 2019-10-14:
Why do you want to restart it?

Kryzek, 2019-10-15:
Only reason to restart (to me) is when I make changes to devices.js to add new devices or change device configuration. Those changes are not applied on-the-fly?

Bastiencc, 2019-10-15:
> 
> 
> Only reason to restart (to me) is when I make changes to devices.js to add new devices or change device configuration. Those changes are not applied on-the-fly?

yes that's the main reason, when I have to remove and re-paire device that stop working (mainly hue dimmer switch ou Xiaomi motion sensor). Pairing seems easier after a restart

tb-killa, 2019-10-15:
What about dynamic reload of converters by load them dynamic if convert is needing?
With this you could modify the converters and after next zigbee message to need to proceed this would load them directly.
I don't like the idea of restart the whole Z2M instance remotely by mqtt.

Kryzek, 2019-10-15:
> Pairing seems easier after a restart

I've noticed this too. I'm been re-pairing my whole network last couple of days and today I just could not get my last two devices Trådfri Motion sensor (E1525) and Trådfri Led bulb (LED1623G12) paired. After restarting Z2M both joined immediately.

carldebilly, 2019-10-26:
I can see another reason for a remote update: force Z2M to republish all topics when they are not set as "retain".

closed 2020-01-01 by stale[bot]