Koenkk/zigbee2mqtt #2136: MQTT message structure for ThingsBoard

mcpicoli, 2019-10-14:
[Thingsboard](https://github.com/thingsboard/thingsboard) is an open source platform for collection, processing and display of IoT data. However, while it uses MQTT as its primary IoT protocol, its message format and API is somewhat different from the one used by HomeAssistant.

The issue is that I cannot find anywhere how to perform/implement changes or configuration of the Zigbee2MQTT bridge as to conform to the Thingsboard message payload format and API.

In the sense that Zigbee2MQTT acts as a bridge between the Zigbee network and an IoT broker, the most sensible API to use in Thingsboard for this integration would be the Gateway API described [here](https://thingsboard.io/docs/reference/gateway-mqtt-api/).

Among the most significant differences is that the device identifier (ID or friendly_name) is contained inside the message payload and not in the topic.

If there is a way to reconfigure Zigbee2MQTT's message payload format (a configuration file?), it would be possible to at least implement the basic telemetry functions easily.

Did anyone had any success in using other platforms (not HomeAssistant) as the consumer for Zigbee2MQTT's data?

Thanks

Koenkk, 2019-10-15:
Currently zigbee2mqtt supports json and attribute output. A new output type can be implemented here: https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/controller.js#L282

mcpicoli, 2019-10-15:
That is not related to the problem. Thingsboard uses JSON as output, but the payload structure is quite different.

In other words, same data, organized in different way, at the same time being output as JSON.

Koenkk, 2019-10-15:
That's why a new output mode should be introduced called `thingsboard` which is still JSON but adds the missing data.

mcpicoli, 2019-10-15:
Undestood.

Is there any addittional documentation on how to do this?

The lines you've pointed deal with the outputting of the message, but I couldn't find where the actual payload object is constructed. For attributes, L273-L274 seems to do it, but for device telemetry?

I have to admit I am quite a novice regarding this language, therefore I apologize if the answer is right in front of me and not being seen. If I have to implement this, it will most likely be _ugly_, but worth a try...

(And while at this, by the same reasoning, shouldn't the current "json" output mode be renamed "homeassistant"?)

Koenkk, 2019-10-16:
`messagePayload` is the object containing all the properties, you can add extra properties to the object and they will be published to MQTT.

`json` shouldn't be renamed to `homeassistant` as the json output is generic and contains no Home Assistant specific information (it can also be used by other applications)

closed 2019-12-22 by stale[bot]

michcfr, 2025-06-10:
Hi, do you know if ThingsBoard is supported now? If not, what needs to be implemented now that the Zigbee2MQTT structure has changed (controller.ts instead of controller.js)?"