Koenkk/zigbee2mqtt #2137: Can't install zigbee2mqtt anymore.

TilmanK, 2019-10-14:
zigbee2mqtt version: Current dev branch
CC253X firmware version: CC2531_DEFAULT_20190608

Hey,

I was having a lot of problems with routers (OSRAM Plug-01) disconnecting, so I wanted to try out the current dev version with herdsman. I tried updating zigbee2mqtt and afterwards also deleted everything, always ending up with the same error when calling "sudo npm install":

`sudo npm install

> @serialport/bindings@8.0.1 install /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install EACCES: permission denied, mkdir '/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build'
gyp ERR! configure error
gyp ERR! stack Error: EACCES: permission denied, mkdir '/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build'
gyp ERR! System Linux 4.15.0-65-generic
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
gyp ERR! node -v v10.16.3
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/zigbee-herdsman/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@8.0.1 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the @serialport/bindings@8.0.1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/tilman/.npm/_logs/2019-10-14T21_33_16_202Z-debug.log`

I followed the guide from https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html, except that i switched branches and I sudo'ed the npm install command. I restarted several times, purged the folder and started over again, reflashed the stick.

Any suggestions?

TilmanK, 2019-10-14:
I can install zigbee2mqtt when changing owner of /opt/zigbee2mqtt and running without root rights.

zoltantakacshu, 2019-10-15:
I can confirm - i was running to into the same issue - same errors.
Following the instructions - and running the **npm install** command without root permission should be ok.

TilmanK, 2019-10-15:
> I can confirm - i was running to into the same issue - same errors.
> Following the instructions - and running the **npm install** command without root permission should be ok.

Yes, indeed. Following the instructions does solve the problem. Nevertheless, I think it might be useful that superuser rights will be used for folder creation when running with sudo, right?

zoltantakacshu, 2019-10-15:
yes.
if you follow the steps - the git clone command is executed with sudo, therefore it is created - then you change the folder ownership to pi user...
it took me a while yesterday until i could figure it out where did i made the mistake with the steps...

closed 2019-12-21 by stale[bot]

rstrawsb, 2021-09-11:
I had the same issue. The instruction involving chowning the files with pi:pi didn't work for me as there wasn't a pi user on the openhabian os.  I had other issues as well but it seems I have resolved them all.

movy, 2021-11-07:
@rstrawsb , any post-morterm analysis how you managed to fix your issues? I encounter problems installing on armbian as well.