Koenkk/zigbee2mqtt #2139: Unable to add support for new devices

indubwestep, 2019-10-14:
zigbee2mqtt version: 1.6.0
CC2652R firmware version: CC26X2R1_20191011

Previously the same devices.js, fromZigbee.js, and homeassistant.js was working and support for custom devices that was added was working properly. I understand the folder structure has changed and I am wondering whether I have the correct structure. In opt/zigbee2mqtt I ran 'npm install zigbee-shepherd-converters' which added a 'node_modules' folder which contains zigbee-shepherd converters. Permissions are correct. 

In devices.js the following was added:
`    {
        zigbeeModel: ['SLC605-LEVELCTR'], 
        model: 'SLC605-LEVELCTR', 
        vendor: 'OWON', 
        description: 'Zigbee Switch', 
        supports: 'on/off', 
        fromZigbee: [fz.SLC605, fz.ignore_onoff_change, fz.ignore_metering_change, fz.state], 
        toZigbee: [tz.on_off, tz.ignore_transition],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),          
                (cb) => device.bind('genBasic', coordinator, cb),
                (cb) => device.bind('genGroups', coordinator, cb),
                (cb) => device.bind('genIdentify', coordinator, cb),
                (cb) => device.report('seMetering', 'instantaneousDemand', 1, 100, 1, cb),
            ];

            execute(device, actions, callback);
        },
    },
`

In fromZigbee.js the following was added:
`    SLC605: {
        cid: 'seMetering',
        type: 'attReport',
        convert: (model, msg, publish, options) => {
            return {power: precisionRound(msg.data.data['instantaneousDemand'], 2)};
        },    
    },
`

In homeassistant.js the following was added (is this file supposed to be located in opt/zigbee2mqtt/data/lib/extension ?):
`    'SLC605-LEVELCTR': [configurations.switch, configurations.sensor_power, configurations.sensor_linkquality],`

However the log is still showing lack of support:
`zigbee2mqtt:debug 2019-10-14T22:02:49: Received Zigbee message from '0x000d6f000d8155c1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0

zigbee2mqtt:warn  2019-10-14T22:02:49: Received message from unsupported device with Zigbee model 'SLC605-LEVELCTR'

zigbee2mqtt:warn  2019-10-14T22:02:49: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

zigbee2mqtt:debug 2019-10-14T22:07:42: Saving state to file /app/data/state.json`

I noticed that in the logs 'zigbee-herdsman started' replaced 'zigbee-shepherd-converters started' and I am wondering whether that could be the issue? 

As a side note I am getting some npm warnings and wondering whether 'npm audit fix' should be run:
` === npm audit security report ===

# Run  npm install zigbee-herdsman@0.9.2  to resolve 1 vulnerability
┌───────────────┬──────────────────────────────────────────────────────────────┐
│ Moderate      │ Memory Exposure                                              │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Package       │ bl                                                           │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Dependency of │ zigbee-herdsman                                              │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Path          │ zigbee-herdsman > dissolve > bl                              │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ More info     │ https://npmjs.com/advisories/596                             │
└───────────────┴──────────────────────────────────────────────────────────────┘

# Run  npm update eslint-utils --depth 2  to resolve 1 vulnerability
┌───────────────┬──────────────────────────────────────────────────────────────┐
│ Critical      │ Arbitrary Code Execution                                     │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Package       │ eslint-utils                                                 │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Dependency of │ eslint [dev]                                                 │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Path          │ eslint > eslint-utils                                        │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ More info     │ https://npmjs.com/advisories/1118                            │
└───────────────┴──────────────────────────────────────────────────────────────┘
`

Koenkk, 2019-10-15:
You need to look at node_modules/zigbee-herdsman-converters instead of zigbee-shepherd-converters.

indubwestep, 2019-10-15:
But now with zigbee-herdsman I'm getting the error 'Error: SRSP - SYS - version after 6000ms'

indubwestep, 2019-10-17:
Update: somehow my dev/tty device swapped to the second port of the CC2652 so changing to the other try fixed the above particular error.  However, there is still the issue where adding entries to devices.js in zigbee2mqtt/node_modules/zigbee-herdsman-converters produces no change and I still receive errors of device unsupported for every device I try to add. Ive done npm install in that folder and permissions are correct. Hit a dead end..

Koenkk, 2019-10-17:
Can you post the content of your `devices.js`? (post it on pastebin.com and link it here)

indubwestep, 2019-10-17:
@Koenkk here you go : https://pastebin.com/JgViBb1Y

Koenkk, 2019-10-18:
This is a zigbee-shepherd-converts `devices.js`.

I would recommand to
- go to zigbee2mqtt directory
- execute `rm -rf node_modules && npm install`
- in `node_modules` you shouldn't see a `zigbee-shepherd-converters` folder, only ` zigbee-herdsman-converters` folder
- now edit the `devices.js` inside `zigbee-herdsman-converters`.

ayushjain4484, 2019-12-04:
Hey Koenkk,
I am getting same issue and also not able to locate devices.js file.
under node_module i can see below dir:
zigbee-herdsman/            zigbee-shepherd-converters/

Please let me know if i am looking under correct path:
/opt/zigbee2mqtt/node_modules

Koenkk, 2019-12-05:
@ayushjain4484 first make sure you are on the latest zigbee2mqtt (1.7.1), after that do:

```
cd /opt/zigbee2mqtt
rm -rf node_modules && npm install
```

Now you will find the `devices.js` in `/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices.js`.

closed 2020-02-10 by stale[bot]