Koenkk/zigbee2mqtt #2143: Please support IKEA Tradfri filament LED1842G3 (704.556.76)

bartplessers, 2019-10-15:
# Bug Report
## What happened
New bulb paired with zigbee2mqtt, but does not response to publishing messages

## What did you expect to happen
response

## Case
  - I paired a new bulb
  - bulb seems to be recognized in zigbee2mqtt as device '0x000d6ffffee83304':
    - it's ID appears in configuration.yaml:
[image]

    - item is displayed in Zigbee map
[image]

But bulb does not respond to mqtt publishing messages. As you can see here: my mqtt client receives the published message, but there is no response like in normal situation:

my new IKEA bulb (0x000d6ffffee83304)
[image]

if I compare this with another (working) IKEA bulb (0xd0cf5efffe0bb71a)
[image]
The bulb publishes it's new state here (second line)

also: bulb does not appear in hass.

debug log attached
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3730079/log.txt)

Solution is clearly documented in debug log: :-)
[image]

But it would be geat if this device would be supported natively!

thanx in advance,
kind regards,
bart

## Debug Info
Zigbee2mqtt on bare-metal RPI3
CC2531_USB-stick (using an USB extension cable)
latest version: CC2531_DEFAULT_20190608.zip

bartplessers, 2019-10-15:
Ok, 

adding following lines to
`node_modules/zigbee-shepherd-converters/devices.js`

config:
```
    {
        zigbeeModel: ['TRADFRI bulb E27 WW clear 250lm'],
        model: 'LED1842G3',
        vendor: 'IKEA',
        description: 'TRADFRI bulb E27 WW clear 250lm, dimmable, white spectrum, clear',
        extend: generic.light_onoff_brightness_colortemp,
    },

```

did the trick.

Will testing some things tonight. I don't think the bulb support colortemp, so I will update config if necessary.

Kind regards,
Bart

sjorge, 2019-10-18:
The 'WW' in the model hits that it does not.
WW -> Warm White
WS -> White Spectrum (colortemp)

Koenkk, 2019-10-18:
This device is already supported in the dev branch (https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L783) and indeed doesn't support color temperature.

bartplessers, 2019-10-18:
super! I switched indeed to dev branch few day ago. It seems to be much faster  (pairing) too!
Many many many thanx for this project @Koenkk, having so much fun with it !

Koenkk, 2019-10-18:
Good to read you are seeing the improvements :smile: