Koenkk/zigbee2mqtt #2144: why does "homeassistant: true" sends an empty mqtt message

bartplessers, 2019-10-16:
# Bug Report
## What happened
When you have following entry in your configuration.yaml:
`homeassistant: true`

the messages of a remote are immediately followed by a "clearing" message.
In example:
```
{"linkquality":0,"battery":60,"action":"arrow_right_hold"}
{"linkquality":0,"battery":60,"action":""}
{"linkquality":0,"battery":47,"action":"arrow_right_release","duration":4.932}
{"linkquality":0,"battery":47,"action":""}
```

## What did you expect to happen
no clearing message:

```
{"linkquality":0,"battery":60,"action":"arrow_right_hold"}
{"linkquality":0,"battery":47,"action":"arrow_right_release","duration":4.932}
```

What is the purpose of this clearing message?
it seems that HA works perfectly well without, except for new instances of HA.
So I would keep this setting to true, but without the clearing message.

kind regards,
Bart

## Debug Info
Zigbee2mqtt on bare-metal RPI3
CC2531_USB-stick (using an USB extension cable)
latest version: CC2531_DEFAULT_20190608.zip

Koenkk, 2019-10-16:
See https://github.com/Koenkk/zigbee2mqtt/issues/959#issuecomment-480341347 and https://github.com/Koenkk/zigbee2mqtt/issues/1457#issuecomment-485401419

closed 2019-10-16 by Koenkk

bartplessers, 2019-10-17:
ok, thanx for clearing this out.
So, @Koenkk , if I understand this correctly:
- without the clearing message: there is no change between 2 identical consecutive "on click" events (i.e. "single")
- following automation would not work, because there is no change between the first and second "single"

```
   trigger:
      platform: state
      entity_id: sensor.my_switch_click
      to: 'single'
```

- Because I use the mqtt variant:
```
   trigger:
      platform: mqtt
      topic: 'zigbee2mqtt/<FRIENDLY_NAME'
    condition:
      condition: template
      value_template: '{{ "single" == trigger.payload_json.click }}'
```
I don't really need the clearing message (but it does no harm either).

Am I correct?

Kind regards,
Bart

Koenkk, 2019-10-18:
Yes exactly.

```yaml
   trigger:
      platform: state
      entity_id: sensor.my_switch_click
      to: 'single'
```

is the preferred way to integrate (because it's simpler).

> I don't really need the clearing message (but it does no harm either).

Yes