Koenkk/zigbee2mqtt #2147: LLKZMK11LM do not respond

LCerebo, 2019-10-16:
# Bug Report
## What happened
When updating to latest-dev version of the docker container all the xiaomi aqara relay (LLKZMK11LM) stop working. When send a command from home assistant, the device turn on but the state is not registered.
[logs](https://pastebin.com/embed_js/BvigayaE)

## How to reproduce it (minimal and precise)
update to latest-dev docker container and try to turn on Xiaomi aqara relay (LLKZMK11LM)

## Debug Info
zigbee2mqtt version: version 1.6.0 (commit #7ffe255)
CC2531 firmware version: 20190608

Koenkk, 2019-10-17:
- Does this always happen? 
- What was your previous version?

Koenkk, 2019-10-17:
Can you sniff the network when controlling the device and getting the error? (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html). I need to know if this error is at the device or zigbee-herdsman side.

LCerebo, 2019-10-17:
I could arrange a second CC2531 in a couple of day. I will write back when I have the results.

LCerebo, 2019-10-18:
@Koenkk here you can find the sniffed packet some from xiaomi relay and one from the osram smart-plug.
[packets.zip](https://github.com/Koenkk/zigbee2mqtt/files/3745766/packets.zip)
If you need some more information just tell me.

Koenkk, 2019-10-20:
Sorry, I forgot to mention that while sniffing I also need the corresponding log (I need to compare the used transaction numbers)

LCerebo, 2019-10-20:
Hi, unfortunately I'don't have the logs anymore (only previous day logs saved in the container). Now that I've done the re-pair the devices are working again, so I can't reproduce the situation. I'm closing the issue. If it happen again during the next update, I will open another issue and provide both logs.
Thanks for your support.

closed 2019-10-20 by LCerebo