Koenkk/zigbee2mqtt #2149: Tradfri Switch on/off not working reliable - message inconsistency

njordan77, 2019-10-17:
Hi,
i'm using the latest stable version (installed few days ago) on a Raspi4 that works nicely and is connectedt to Mosquitto and Openhab.

Tradfri Remote Control works great, the Switch on/off device gives me some hard times....
Whenever i press ON or OFF i receive frequently a message that does not have the CLICK part in JSON structure, which ends up in Openhab not receiving anything as it waits for the click part.....

15.10.2019 18:45:51 /z2m/Tradfri_Switch01 {“battery”:60,“linkquality”:0}

15.10.2019 18:45:s51 /z2m/Tradfri_Switch01 {“battery”:60,“linkquality”:0,“click”:“off”}

Any idea what is wrong here, is this a mistake/bug???? Never seen this from any other device i have installed. Kind Regards Norbert

Koenkk, 2019-10-17:
This is not a bug, the tradfri switch occasionally reports it's battery status and these messages don't have a `click`.

njordan77, 2019-10-17:
Thanks for the fast reply. That was my 2nd thought so it would unsolicitated provide feedback about its state. The bad thing only is that it always comes in front of a click message.....so Openhab takes the first one which does not do anything and for whatever reason ignores the 2nd message with the click content. This does not happen with Tradfri RemoteController....

All of it not your problem of course, will try to find a solution on the other end of the communications channel.

(btw, do you know about Tradfri Open/Close Remote. looks similar to ON/OFF switch and comes with their shutter series....tried to bind it but zigbee2mqtt does not support it yet....let me please know if i can help to get it working)

Kind Regards Norbert

Koenkk, 2019-10-17:
The open close remote is supported in the latest dev branch.

njordan77, 2019-10-17:
ah OK, very up to date. big like and closing this threat.

closed 2019-10-17 by njordan77