Koenkk/zigbee2mqtt #2153: zigbee2mqtt silent on pairing failure

wassfila, 2019-10-17:
I assume that the message response with the modelid can get lost after pairing a device, therefore zigbee2mqtt gets no modelId from shepherd, keeps logging it as undefined and not reporting it on mqtt. The device is then unresponsive from the user point of view.
My suggestion would be to add a hint to the user to pair the device again or to retrigger request of the model id reading from shepherd so that the device does not stay as a zombie.

Attached two logs, one where the issue happened and one where I added a log in lib\controller.js line 133
`        // Retrieve modelId from message
        if (device && device.modelId) {
            mappedDevice = zigbeeShepherdConverters.findByZigbeeModel(device.modelId);
        }
        else{
            logger.warn("No device.modelId in device:",device);
        }
`
The funny is that the second log also start by reporting the device modelId as undefined until this message comes which fixes every thing. And this message is missing in the first log where the issue persisted
`Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"LUMI","modelId":"lumi.sensor_motion.aq2","powerSource":3}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002753dec) of endpoint 1`

[missing model-id in pairing - log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3740774/missing.model-id.in.pairing.-.log.txt)
[pairing with model ide as second msg - log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3740775/pairing.with.model.ide.as.second.msg.-.log.txt)

Koenkk, 2019-10-17:
Did you try the latest dev branch? This greatly improves this (#1888)

wassfila, 2019-10-18:
Ow, sorry, Yeah, I'll be testing with latest dev branch rather, thanks for
the hint.

wassfila, 2019-10-18:
I'll close it for now as this is hard to reproduce and if it doesn't on latest dev then it's fine.

closed 2019-10-18 by wassfila