Koenkk/zigbee2mqtt #2154: Failed to configure <device>, attempt 2 (Error: AREQ - ZDO - bindRsp after 10000ms

timdonovanuk, 2019-10-17:
Thought I'd try the new underlying herdsmans bit of zigbee2mqtt. I have an Ikea E1524/E1810. It appears to function, but I'm getting the following error in the log:

```
zigbee2mqtt:error 2019-10-17T18:44:53: Failed to configure 'ikea_remote_1', attempt 2 (Error: AREQ - ZDO - bindRsp after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10))
```

This device is simply configured as:

```
  '0x90fd9ffffe45c50f':
    friendly_name: ikea_remote_1
    retain: false
```

## Debug Info
zigbee2mqtt version: latest dev (from like literally 13 mins ago :))
CC253X firmware version: Don't know now, sorry.

Koenkk, 2019-10-17:
The device is probably sleeping when the configure is happening, to fix it keep pressing a button on the device while starting zigbee2mqtt (until you see successfully configured).

arteck, 2019-11-15:
same problem here with a hue bulb gu10

```
Failed to execute LQI for '0x0017880103dcc9d2'. "Error: AREQ - ZDO - mgmtLqiRsp after 10000ms\n    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/iobroker/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)\n    
at ontimeout (timers.js:436:11)\n    
at tryOnTimeout (timers.js:300:5)\n    
at listOnTimeout (timers.js:263:5)\n    
at Timer.processTimers (timers.js:223:10)"
```

sjorge, 2019-11-15:
I managed to get 2 pair but they are a bit annoying, what worked for me was to push the back button once every 3 seconds or so until it paired.

Koenkk, 2019-11-15:
So what happens:
- zigbee2mqtt configures the devices on startup, note that zigbee2mqtt 1.7 >= only does this once, once it succeeds it will remember it. This was not the case for pre zigbee2mqtt 1.6 (it would do it on every startup).
- As probably the configure never went successfully while on zigbee2mqtt 1.7, it will try the configure on every startup
- The configure fails because the device is sleeping, to fix it, while starting zigbee2mqtt keep pressing buttons on the switch until you see `Succesfully configured .....` for this device.
- Next time you restart zigbee2mqtt you won't run into this problem again as it already configured successfully once.

closed 2020-01-21 by stale[bot]

Patrix2911, 2020-05-08:
Hello, if i try to bind my IKEA E1524/E1810 (i'm not shure, which of the two it is) with the IKEA LED1732G11 i receive the followed Error Messages

May 08 23:23:27 Schlafzimmer npm[299]: zigbee2mqtt:error 2020-05-08 23:23:27: Failed to bind cluster 'genLevelCtrl' from '0x680ae2fffeaaa57a' to '0x680ae2fffe8032c1' (Error: Bind 0x680ae2fffeaaa57a/1 genLevelCtrl from '0x680ae2fffe8032c1/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms))
May 08 23:23:27 Schlafzimmer npm[299]: zigbee2mqtt:info  2020-05-08 23:23:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_bind_failed","message":{"from":"0x680ae2fffeaaa57a","to":"0x680ae2fffe8032c1","cluster":"genLevelCtrl"}}'

I use an CC2531. Can anyone help me to fix the Problem?

ooii, 2020-10-30:
Hi, I'm having this issue with 3 dimmers (I'm adding them for the nth time).
The three are in the same room, far from the bridge by 2 to 3 meters. I set the `transmit_power` to its maximum value, 19.
One of the dimmers blinks green all the time but nothing is reported on z2m logs.
The 2 others do not blink.
Tried keeping a button pressed while starting z2m, it's still unable to configure the three dimmers.
Running z2m (1.15.0) as an add-on of HA (0.1.117) using a CC2652R zzh usb stick with `znp_CC26X2R1_LAUNCHXL_tirtos_ccs.hex`firmware.