Koenkk/zigbee2mqtt #2155: Mesh Network?

aihysp, 2019-10-18:
Hi All long time ago i think i heard that zigbee is a mesh network 
but i don't see it in real life, i am using RPI and the CC2531 flashed and Zigbee2MQTT ...

is there any way to enable it?
anyway to check what's wrong?

james-fry, 2019-10-18:
It becomes a mesh only if there are router devices in the network - these tend to be powered devices such as bulbs, switches etc.
Battery powered end devices are not likely to be routers due to the fact they sleep most of the time.
Do you have only sensors in your network?

aihysp, 2019-10-18:
yes for now only sensor thank you!!!
seems to be the case, for AC power I will rather use wifi

closed 2019-10-18 by aihysp

james-fry, 2019-10-18:
BTW if you want to extend your network range you can add additional CC2530/CC2531 devices that are flashed with [router firmware](https://github.com/Koenkk/Z-Stack-firmware/tree/master/router). I have found that a CC2531 inserted into an old phone USB charger is a very cheap way to add routers.

aihysp, 2019-10-19:
Wow very good to know !!!

so if I get you right, I can even put this on my Let's say AVR receiver /TV anything with a USB
do you know what is the required Miliamps? for CC2530/CC2531