Koenkk/zigbee2mqtt #2157: Docker latest-dev: error starting when USB device is not named /dev/ttyACM(0,1)

airdrummingfool, 2019-10-18:
I'm using Docker to run the dev version of zigbee2mqtt. I pulled the latest version a couple hours ago, and as soon as I did, z2m stopped starting, and would instead throw the following error:
```
zigbee2mqtt_1    | Using '/app/data' as data directory
zigbee2mqtt_1    |
zigbee2mqtt_1    | > zigbee2mqtt@1.6.0 start /app
zigbee2mqtt_1    | > node index.js
zigbee2mqtt_1    |
zigbee2mqtt_1    | zigbee2mqtt:info  2019-10-18T10:17:09: Logging to directory: '/app/data/log/2019-10-18.10-17-08'
zigbee2mqtt_1    | zigbee2mqtt:debug 2019-10-18T10:17:09: Removing old log directory '/app/data/log/2019-10-18.09-18-14'
zigbee2mqtt_1    | zigbee2mqtt:debug 2019-10-18T10:17:09: Loaded state from file /app/data/state.json
zigbee2mqtt_1    | zigbee2mqtt:info  2019-10-18T10:17:09: Starting zigbee2mqtt version 1.6.0 (commit #f219e5d)
zigbee2mqtt_1    | zigbee2mqtt:info  2019-10-18T10:17:09: Starting zigbee-herdsman...
zigbee2mqtt_1    | zigbee2mqtt:debug 2019-10-18T10:17:09: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extenedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","backupPath":"/app/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/zigbee"}}'
zigbee2mqtt_1    | zigbee2mqtt:error 2019-10-18T10:17:09: Error while starting zigbee-herdsman
zigbee2mqtt_1    | zigbee2mqtt:error 2019-10-18T10:17:09: Failed to start zigbee
zigbee2mqtt_1    | zigbee2mqtt:error 2019-10-18T10:17:09: Exiting...
zigbee2mqtt_1    | zigbee2mqtt:error 2019-10-18T10:17:09: Error: Path '/dev/zigbee' does not exist
zigbee2mqtt_1    |     at Object.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/serialPortUtils.js:31:19)
zigbee2mqtt_1    |     at Generator.next (<anonymous>)
zigbee2mqtt_1    |     at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/serialPortUtils.js:5:58)
zigbee2mqtt_1    |     at process._tickCallback (internal/process/next_tick.js:68:7)
zigbee2mqtt_1    | npm ERR! code ELIFECYCLE
zigbee2mqtt_1    | npm ERR! errno 1
zigbee2mqtt_1    | npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
zigbee2mqtt_1    | npm ERR! Exit status 1
zigbee2mqtt_1    | npm ERR!
zigbee2mqtt_1    | npm ERR! Failed at the zigbee2mqtt@1.6.0 start script.
zigbee2mqtt_1    | npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
zigbee2mqtt_1    |
zigbee2mqtt_1    | npm ERR! A complete log of this run can be found in:
zigbee2mqtt_1    | npm ERR!     /root/.npm/_logs/2019-10-18T10_17_09_288Z-debug.log
```
Note that I had not touched my `docker-compose.yaml` or `zigbee2mqtt/configuration.yaml` file since the previous version had started successfully.

Here's my original configuration (that no longer works):
docker-compose.yaml
```yaml
  zigbee2mqtt:
    image: koenkk/zigbee2mqtt:latest-dev
    restart: unless-stopped
    devices:
      - /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B00015A7695-if00:/dev/zigbee
[...]
```

zigbee2mqtt configuration.yaml:
```yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://mqtt'
serial:
  port: /dev/zigbee
```

After lots of trial and error, I found that I could get zigbee2mqtt working again if I used `/dev/ttyACM0` or `/dev/ttyACM1` instead of `/dev/zigbee`. However, other similar device names do not work; so far I've tried `/dev/ttyACM2` and `/dev/ttyACM11` - neither works.

**The only thing I have to do to fix the problem is change the name of the device in both `configuration.yaml` and `docker-compose.yaml`**.

Here's my configuration that works now:
docker-compose.yaml
```yaml
  zigbee2mqtt:
    image: koenkk/zigbee2mqtt:latest-dev
    restart: unless-stopped
    devices:
      - /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B00015A7695-if00:/dev/ttyACM0
[...]
```

zigbee2mqtt configuration.yaml
```yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://mqtt'
serial:
  port: /dev/ttyACM0
```

Koenkk, 2019-10-18:
Does it work with

```js
serial:
  port: null
```

airdrummingfool, 2019-10-18:
No, there's a different error (doesn't matter if the device is set to `/dev/ttyACM0` or `/dev/zigbee` in `docker-compose.yaml`):
```
zigbee2mqtt_1    | zigbee2mqtt:info  2019-10-18T11:17:28: Starting zigbee-herdsman...
zigbee2mqtt_1    | zigbee2mqtt:debug 2019-10-18T11:17:28: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extenedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","backupPath":"/app/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":null}}'
zigbee2mqtt_1    | zigbee2mqtt:error 2019-10-18T11:17:28: Error while starting zigbee-herdsman
zigbee2mqtt_1    | zigbee2mqtt:error 2019-10-18T11:17:28: Failed to start zigbee
zigbee2mqtt_1    | zigbee2mqtt:error 2019-10-18T11:17:28: Exiting...
zigbee2mqtt_1    | zigbee2mqtt:error 2019-10-18T11:17:28: Error: No path provided and failed to auto detect path
zigbee2mqtt_1    |     at Function.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/adapter.js:51:27)
zigbee2mqtt_1    |     at Generator.next (<anonymous>)
zigbee2mqtt_1    |     at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/adapter.js:5:58)
zigbee2mqtt_1    |     at process._tickCallback (internal/process/next_tick.js:68:7)
```

Koenkk, 2019-10-18:
What adapter or you using with what firmware?

airdrummingfool, 2019-10-18:
Adapter: CC2531
Firmware: CC2531_DEFAULT_20190608

Koenkk, 2019-10-18:
Below https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/serialPortUtils.ts#L11 can you add a `console.log(devices)`? (you should find the `serialPortUtils.js` under `node_modules/zigbee-herdsman/dist/adapter`.

airdrummingfool, 2019-10-18:
```
zigbee2mqtt_1    | [ { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyACM0' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyACM1' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyS0' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyS1' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyS10' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyS11' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyS12' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyS13' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyS14' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
[... 145 lines cut ...]
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyS6' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyS7' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyS8' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyS9' } ]
```

airdrummingfool, 2019-10-18:
The correct path is `/dev/ttyACM0`

Koenkk, 2019-10-18:
Can you run the container with:

```
privileged: true
```

plasticrake, 2019-10-18:
Okay this is funny, I just posted a [related issue](Koenkk/zigbee-herdsman#77) with a similar workaround. However I COULDN'T use `ttyACM0` and had to switch to `ttyUSB0`.

airdrummingfool, 2019-10-19:
With `privileged: true` and `/dev/ttyACM0` in `docker-compose.yaml`, and `port: null` in z2m `configuration.yaml`, the error appears to be the same:
```
zigbee2mqtt_1    | zigbee2mqtt:info  2019-10-19T04:58:25: Starting zigbee-herdsman...
zigbee2mqtt_1    | zigbee2mqtt:debug 2019-10-19T04:58:25: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extenedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","backupPath":"/app/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":null}}'
zigbee2mqtt_1    | [ { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyACM0' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyACM1' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyS0' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyS1' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyS10' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyS11' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyS12' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyS13' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyS14' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
[... 153 lines cut ...]
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyS7' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyS8' },
zigbee2mqtt_1    |   { manufacturer: undefined,
zigbee2mqtt_1    |     serialNumber: undefined,
zigbee2mqtt_1    |     pnpId: undefined,
zigbee2mqtt_1    |     locationId: undefined,
zigbee2mqtt_1    |     vendorId: undefined,
zigbee2mqtt_1    |     productId: undefined,
zigbee2mqtt_1    |     path: '/dev/ttyS9' } ]
zigbee2mqtt_1    | zigbee2mqtt:error 2019-10-19T04:58:25: Error while starting zigbee-herdsman
zigbee2mqtt_1    | zigbee2mqtt:error 2019-10-19T04:58:25: Failed to start zigbee
zigbee2mqtt_1    | npm ERR! code ELIFECYCLE
zigbee2mqtt_1    | zigbee2mqtt:error 2019-10-19T04:58:25: Exiting...
zigbee2mqtt_1    | zigbee2mqtt:error 2019-10-19T04:58:25: Error: No path provided and failed to auto detect path
```

airdrummingfool, 2019-10-19:
as @plasticrake guessed, my Docker host machine has `/dev/ttyACM[0-1]`, but not `/dev/ttyACM2` or `/dev/ttyACM11` (other things I had tried).

airdrummingfool, 2019-10-19:
And with `privileged: true` and `/dev/zigbee` in `docker-compose.yaml`, and `port: dev/zigbee` in z2m `configuration.yaml`, the error is slightly different (but matches @plasticrake's in https://github.com/Koenkk/zigbee-herdsman/issues/77#issuecomment-544012715): 

```
zigbee2mqtt_1    | zigbee2mqtt:info  2019-10-19T05:06:15: Starting zigbee2mqtt version 1.6.0 (commit #f219e5d)
zigbee2mqtt_1    | zigbee2mqtt:info  2019-10-19T05:06:15: Starting zigbee-herdsman...
zigbee2mqtt_1    | zigbee2mqtt:debug 2019-10-19T05:06:15: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extenedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","backupPath":"/app/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/zigbee"}}'
zigbee2mqtt_1    | zigbee2mqtt:error 2019-10-19T05:06:15: Error while starting zigbee-herdsman
zigbee2mqtt_1    | zigbee2mqtt:error 2019-10-19T05:06:15: Failed to start zigbee
zigbee2mqtt_1    | zigbee2mqtt:error 2019-10-19T05:06:15: Exiting...
zigbee2mqtt_1    | zigbee2mqtt:error 2019-10-19T05:06:15: Error: ENOENT: no such file or directory, lstat '/dev/zigbee'
zigbee2mqtt_1    |     at Object.realpathSync (fs.js:1461:7)
zigbee2mqtt_1    |     at Object.realpathSync [as RealpathSync] (/app/node_modules/zigbee-herdsman/dist/utils/realpathSync.js:10:25)
zigbee2mqtt_1    |     at Function.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/adapter.js:56:50)
zigbee2mqtt_1    |     at Generator.next (<anonymous>)
zigbee2mqtt_1    |     at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/adapter.js:5:58)
zigbee2mqtt_1    | npm ERR! code ELIFECYCLE
zigbee2mqtt_1    | npm ERR! errno 1
zigbee2mqtt_1    | npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
zigbee2mqtt_1    | npm ERR! Exit status 1
zigbee2mqtt_1    | npm ERR!
zigbee2mqtt_1    | npm ERR! Failed at the zigbee2mqtt@1.6.0 start script.
zigbee2mqtt_1    | npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
zigbee2mqtt_1    |
zigbee2mqtt_1    | npm ERR! A complete log of this run can be found in:
zigbee2mqtt_1    | npm ERR!     /root/.npm/_logs/2019-10-19T05_06_15_798Z-debug.log
```

airdrummingfool, 2019-10-19:
@Koenkk By manually building Docker images of different commits, I've tracked this issue down to commit https://github.com/Koenkk/zigbee2mqtt/commit/4164f01a6b24754bfd121bd31d3356413ee1feec, but the actual relevant code change probably lies in `zigbee-herdsman` between 0.9.3 and 0.10.0. My guess is https://github.com/Koenkk/zigbee-herdsman/commit/bd7dfff1d697ed5e2a40df7005e4389913bc428e.

Koenkk, 2019-10-20:
Should be fixed in the latest dev branch. Old configuration should work again now.

To make use of port auto detection the container needs to be run with `-v /run/udev:/run/udev:ro --privileged=true` (will add this to the docs).

airdrummingfool, 2019-10-21:
I can confirm that I can again use a custom device name (e.g. `/dev/zigbee`) instead of being forced to use an existing USB device name (i.e. `/dev/ttyACM0`).

I don't use port auto-detection, so I cannot report on that functionality.

Thank you!

closed 2019-10-21 by airdrummingfool