Koenkk/zigbee2mqtt #2158: Zigbee gateway limits and requirements

choeflake, 2019-10-18:
First some background. I started a few months ago with the CC2531, ordered some sticks at AliExpress. When the sticks arrived, I started flashing the sticks. But which firmware? Generally I had to choose between Zigbee 2 and 3. That's a no brainer! Of course Zigbee 3, more secure, more reliable, newer, better support. But... hey wait... the maximum number of devices is 40! My network is currently 60+ devices.

Replacing all bulbs in a standard house will touch the 50+ easily. And think next, we (zigbee2mqtt users) want to add dozens of other devices like temperature meters, blinds etc etc in the near future. Zigbee products are becoming more and more payable for everyone. So networks will grow and reach the limits of the hardware. 

Looking at https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator, shows that the max with the best hardware is currently 100 devices. Personally I want a gateway with a limit around 250 at least (at the moment). 
I'm curious how other zigbee2mqtt users see this.

kirovilya, 2019-10-18:
My opinion is that it is not necessary to create one large network. You can make several small ones (in the house you can, for example, one by floor). This will increase fault tolerance and speed of work (the smaller the network, the faster it works).
Although the new tested chips 2652 and 2538 will allow you to connect more devices (probably up to 200).

wassfila, 2019-10-18:
Very interesting. Just two hints, I used the dongle with PA+LNA and that gives a huge range difference, important in big networks.
I do run 3 zigbee2mqtt instances myself, just experimenting, important is to think about selected channels not to conflict with each other and push them far from wifi channels.
Routers are not power cycles resilients I think, should re-pair which is no practicle.

kirovilya, 2019-10-18:
panID must be different too

choeflake, 2019-10-18:
Multiple smaller networks have some advantages like you described. Another advantage is testing upgrades per network.
I see also some disadvantages. Multiple channels and far from wifi is not very realistic, see https://support.metageek.com/hc/en-us/articles/203845040-ZigBee-and-WiFi-Coexistence. So the networks have to deal with more noise. 
More routers in the network (like bulbs) decreases the need for 'high' energy radio signals than one large network (in one network the senders and receivers are mostly close together). Which decreases the radio noise from/to neighbours. And nobody knows exactly what effects for health are.
Just my two cents about multiple networks...

wassfila, 2019-10-23:
@choeflake actually the reference you mention is a good one and after clicking continue reading on the second page on the section "Channel Planning", it mentions exactly what I meant by allocating the zigbee on a channel far from wifi.
You cannot control your neighbors wifi, but yours at least which is closer would be good to help avoid further noise.
I also understand and agree about the multiple routers network that it will require less power, I do not know to which extent all devices do adapt the power to the link quality.
When it comes to effects on health, it's hard to conclude on any evidence, it's a difficult topic. The low power devices do have a low db compared to wifi, and staying in the sun 10 mn would transfer enough energy through electromagnetic waves to warm someones up more than one could get from a life time wifi electromagnetic waves exposure I guess.

closed 2019-12-29 by stale[bot]