Koenkk/zigbee2mqtt #2159: Update fails

Fabenissimo, 2019-10-18:
# Bug Report
## What happened
I was trying to update zigbee2mqtt which I installed in a venv
## What did you expect to happen
The updte to finish without problems
## How to reproduce it (minimal and precise)
Use the official guide (https://www.zigbee2mqtt.io/information/virtual_environment.html)
## Debug Info
zigbee2mqtt version: latest
CC253X firmware version: Should be not needed for this problem?
```
(zigbee2mqtt) pi@raspberrypi:/opt/zigbee2mqtt $ npm install

> @serialport/bindings@2.0.8 install /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=10.15.1 runtime=node arch=arm libc= platform=linux)
Aborted
Aborted
gyp: Call to 'node -e "require('nan')"' returned exit status 134 while in binding.gyp. while trying to load binding.gyp
gyp ERR! configure error
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/opt/zigbee2mqtt/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:345:16)
gyp ERR! stack     at ChildProcess.emit (events.js:189:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Linux 4.19.58-v7+
gyp ERR! command "/opt/zigbee2mqtt/bin/node" "/opt/zigbee2mqtt/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
gyp ERR! node -v v10.15.1
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
Aborted
npm WARN @typescript-eslint/eslint-plugin@1.12.0 requires a peer of eslint@^5.0.0 but none is installed. You must install peer dependencies yourself.
npm WARN @typescript-eslint/parser@1.12.0 requires a peer of eslint@^5.0.0 but none is installed. You must install peer dependencies yourself.
npm WARN bl@0.7.0 requires a peer of stream-browserify@* but none is installed. You must install peer dependencies yourself.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/zigbee-shepherd-converters/node_modules/zigbee-herdsman/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/zigbee-herdsman/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

npm ERR! code ELIFECYCLE
npm ERR! errno 134
npm ERR! @serialport/bindings@2.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 134
npm ERR!
npm ERR! Failed at the @serialport/bindings@2.0.8 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-10-18T19_52_30_007Z-debug.log
Aborted
```

I hope I didn't miss something, if I did though please contact me, I'll give you all the information you'll need.

Thanks!

Fabian

Koenkk, 2019-10-19:
`Failed at the @serialport/bindings@2.0.8 install script` this doesn't seem to be caused by zigbee2mqtt itself. Something goes wrong while installing the serialport dependency. Perhaps you can try without venv?

Fabenissimo, 2019-10-19:
Thanks for the quick reply! I first tried creating a new venv, as I didn't try that yesterday. That didnt' work either, so I did the bare metal install as you suggested. It works! That's a good workaround, but don't you think there's a possibilty to do this in venv again? Any suggestions on what I could do?

Koenkk, 2019-10-19:
Im not sure on how to fix this, perhaps you can ask this to the venv guys.

closed 2019-12-25 by stale[bot]