Koenkk/zigbee2mqtt #2160: Is my database gone after branch change

fefris, 2019-10-18:
Ran the branch change instructions here

https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

including the backup and restore parts.

after running the 

sudo systemctl start zigbee2mqtt

the Shepard would not start (timeout error) so i rebooted the pi

On reboot ran

sudo systemctl start zigbee2mqtt

and the Shepard started with only the coordinator and no devices.

Database shows no devices accept coordinator. configuration.yaml still shown the customization of the devices though.

Is my database backup now gone and am i now stuck with repairing everything.?

LCerebo, 2019-10-19:
Hi, for some weeks now I'm using the latest-dev docker container and I confirm that you can't switch back to the latest stable branch and keep the db. Even if the db.yml is there it's some how incompatible and will be erased. The only solution that I found so far is to keep a version of each database and switch them manually.

fefris, 2019-10-19:
thanks for confirming. :-(

closed 2019-10-19 by fefris

Morphy99, 2021-01-04:
I think I came across this issue, for some reason the db gets wiped going back to stable but it creates a copy called database.db.backup. I restored this and seem to work for me.