Koenkk/zigbee2mqtt #2161: Zboss cant create account.... how are you performing packet sniffing on windows?

CamHaug, 2019-10-19:
According to the how-to here:
https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html
I should use zboss on windows for packet sniffing, however I cannot create an account. 
Yes I have checked spam filtering.... :\

Any one have a copy of it?

Cheers 

-Cam

CamHaug, 2019-11-14:
sooo no one does any packet sniffing on windows?

sjorge, 2019-11-14:
I only every did sniffing on windows once as a test using a nRF52840-Dongle following https://infocenter.nordicsemi.com/index.jsp?topic=%2Fstruct_nrftools%2Fstruct%2Fnrftools_nrfsniffer.html&cp=6_4, but it never worked well.

I had more succes on my mac using both a nRF dongle or a ti dongle.

Kryzek, 2019-11-16:
I did also try to download zboss firmware for sniffing for sniffing in Windows. Same problem, can't get newly made account to login there. Guess I need to pull out my linux laptop then and do sniffing on linux.

zi-l, 2019-12-13:
In may case, I cannot log in until I have activated my account through an activation email. Maybe you should wait for the account activation mail.

closed 2020-02-18 by stale[bot]

rmhomecouk, 2020-03-13:
Having the same issue - unable to receive the activation email. Does anyone have the file they could kindly share?

zladukas, 2020-03-13:
Found mine in spam folder yesterday.