Koenkk/zigbee2mqtt #2164: Cannot read property 'white_value' of undefined

bochelork, 2019-10-19:
GledOpto GL-C-007 used to work fine, But can't control it anymore. This is the message:

```
zigbee2mqtt:debug 2019-10-19T15:42:08: Received MQTT message on 'zigbee2mqtt/CCTSTRIP1W/set' with data '{"state": "Off"}'
zigbee2mqtt:debug 2019-10-19T15:42:08: Publishing 'set' 'state' to 'CCTSTRIP1W'
zigbee2mqtt:error 2019-10-19T15:42:08: Publish 'set' 'state' to 'CCTSTRIP1W' failed: 'TypeError: Cannot read property 'white_value' of undefined'
zigbee2mqtt:info  2019-10-19T15:42:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'CCTSTRIP1W' failed: 'TypeError: Cannot read property 'white_value' of undefined'","meta":{"friendly_name":"CCTSTRIP1W"}}'

```

Koenkk, 2019-10-19:
Zigbee2mqtt version?

bochelork, 2019-10-19:
Sorry. Here it is.

```
zigbee2mqtt:info  2019-10-19T18:24:41: Starting zigbee2mqtt version 1.6.0 (commit #7ffe255)
zigbee2mqtt:info  2019-10-19T18:24:43: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}}'
```

Koenkk, 2019-10-20:
Can you provide me the database.db?

bochelork, 2019-10-20:
[database.zip](https://github.com/Koenkk/zigbee2mqtt/files/3747892/database.zip)

Koenkk, 2019-10-20:
Can you check if this is fixed in the latest dev?

bochelork, 2019-10-23:
Everything is working now. Don't know what you did, but my network is pretty stable for a few days now.

closed 2019-10-24 by Koenkk