Koenkk/zigbee2mqtt #2166: Cannot npm start the 1.6.0 version on my Raspi

juergenschubert, 2019-10-19:
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.9.0
3 info using node@v10.16.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.6.0~prestart: zigbee2mqtt@1.6.0
6 info lifecycle zigbee2mqtt@1.6.0~start: zigbee2mqtt@1.6.0
7 verbose lifecycle zigbee2mqtt@1.6.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.6.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/ga
mes:/usr/games
9 verbose lifecycle zigbee2mqtt@1.6.0~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.6.0~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.6.0~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.6.0~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.6.0 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:198:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:198:13)
13 verbose stack     at maybeClose (internal/child_process.js:982:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.6.0
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.19.75-v7l+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.16.3
19 verbose npm  v6.9.0
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.6.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.6.0 start script.
23 error This is probably not a problem with npm.

Koenkk, 2019-10-19:
Duplicate of #2163

didiht, 2019-10-19:
I also installed z2m on a Raspi4 the last days and it's running well.

The versions of npm and node are:
node: v10.16.3
npm: 6.12.0

Maybe update of npm helps

juergenschubert, 2019-10-19:
nope npm 6.12.0 has the same issue.
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.12.0
3 info using node@v10.16.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.6.0~prestart: zigbee2mqtt@1.6.0
6 info lifecycle zigbee2mqtt@1.6.0~start: zigbee2mqtt@1.6.0
7 verbose lifecycle zigbee2mqtt@1.6.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.6.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/ga
mes:/usr/games
9 verbose lifecycle zigbee2mqtt@1.6.0~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.6.0~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.6.0~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.6.0~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.6.0 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:198:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:198:13)
13 verbose stack     at maybeClose (internal/child_process.js:982:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.6.0
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.19.75-v7l+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.16.3
19 verbose npm  v6.12.0
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.6.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.6.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

juergenschubert, 2019-10-19:
npm start

> zigbee2mqtt@1.6.0 start /opt/zigbee2mqtt
> node index.js

internal/modules/cjs/loader.js:638
    throw err;
    ^

Error: Cannot find module 'zigbee-herdsman/dist/deprecated/ziee'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/zapp/cie.js:6:14)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-10-19T21_38_35_581Z-debug.log

juergenschubert, 2019-10-20:
git checkout npm-shrinkwrap.json
git checkout package.json
rm -rf node_modules && npm install
npm start

solved the problem !

closed 2019-10-20 by juergenschubert