Koenkk/zigbee2mqtt #2169: Unable to pair a repeater from the Fyrtur blinds

genehand, 2019-10-19:
# Bug Report
## What happened
I just got the Fyrtur blinds released yesterday in the US.  The repeater is labeled as the same E1746 that is already supported, but I'm wondering if there's something different about it now like a new firmware.

`Failed to interview '0x000d6ffffe2f0914', device has not successfully been paired`

You can see it trying to interview a few times until the repeater stops the pairing process:
https://pastebin.com/0s7WVTKe

I have 6 Xioami devices paired without issue, the CC2531 is on an extension cable and I've tried different USB ports for good measure.  I'm also putting the repeater right next to it (~2 cm).  Happy to provide any other debugging info.

## What did you expect to happen
Devices pairs successfully

## How to reproduce it (minimal and precise)
- Permit joining
- Plug in the repeater and press the reset button for 5s

## Debug Info
zigbee2mqtt version: 1.6.0 using hassio-zigbee2mqtt-edge and verified I have the latest container image:

```dwelch2101/zigbee2mqtt-edge-amd64        test                sha256:04393720ddf29aa75fa2d2ade28a11bc8b008e9da567d1d2e5aa8e709e43f30c   79323748e702        13 hours ago        360MB``` 

CC253X firmware version: 20190608

I also noticed this issue which makes me wonder if something is different with the Fyrtur repeaters: https://github.com/home-assistant/home-assistant/issues/27923

Koenkk, 2019-10-20:
The pairing procedure consists of 2 stages:
1. Joining the network: the device (e.g. TRADFRI signal repeater) announces that it wants to join the network. When `pemit_join: true` the adapter (e.g. CC2531) will share the network keys with the device. Once this is finished the device is part of the network and, in case it is a router, will route messages in the network. This can be seen in the zigbee2mqtt log as `Device '[ieeeAddr]' joined. This stage went successful.
2. Interviewing:  after the device joined, we need need to get detailed information about the device. An example is the `modelID` which is used by zigbee2mqtt to determine the device type and the device definition to be used. Note that when a part of this still goes right, it could happen that it works perfectly fine (the most crucial thing for zigbee2mqtt is the `modelID`). A part of this procedure failed in your case.

As the TRADFRI signal repeater cannot be controlled, it will work perfectly fine when only stage 1 has been completed. I can look into what went wrong, but for that I need to have the device myself. 

Let's keep this issue open to see if more people encounter this issue with this device. If so, I will dive into it.

james-fry, 2019-10-20:
I have found stage #2 is failing often. Maybe now with 1.6 dev it’s it properly being logged etc. One device I had the issue with recently Is the trust remote.

Koenkk, 2019-10-20:
The point at where it fails can be determined by running zigbee2mqtt with `DEBUG=zigbee-herdsman:controller:* npm start`.

james-fry, 2019-10-20:
Do you know if this is possible with hassio? 
Maybe I can log into the container and change start script? I’ll have a look...

jdtsmith, 2019-12-07:
Can you mention how pairing the signal repeater to `z2m` would be advantageous?  I'm having difficulty coming up with a pairing scheme among the 4 Zigbee devices: 

1. `z2m` on a CC253X
1. up/down remote
1. Fyrtur blinds
1. TRADFRI signal repeater

I have no difficulty pairing **2+3+4 to 1**, or **3 to 1**, or  **2 to 4+3**.   But I can't seem to accomplish pairing **3 to 1 and 2 to 4+3**, which is what would give the desired direct up/down remote control + MQTT control.  Thanks for `z2m`!

closed 2020-02-12 by stale[bot]