Koenkk/zigbee2mqtt #2173: Understanding Zigbee2mqtt and viewing the Zigbee graph

wassfila, 2019-10-20:
I love this project on which I'm basing my home automation projects now.
Yet I find it still very difficult to apprehend and require a high level of understanding to be able to contribute.
So here I share my efforts to help understanding zigbee2mqtt better, and hopefully quicker growing a community of contributors.

Diagrams are available as svg files here : https://github.com/HomeSmartMesh/raspi/tree/master/zigbee

[image]

[image]

A small helper tool also here : https://github.com/HomeSmartMesh/raspi/tree/master/zigbee/graph_view

[image]

Same announce on hackaday
https://hackaday.io/project/20388-home-smart-mesh/log/170103-understanding-zigbee2mqtt-and-viewing-the-zigbee-graph

wassfila, 2019-10-20:
@Koenkk I wanted to label this ticket as info or announcement, I was not sure which channel could help to share such info. I find the doc as a user very amazing, so here I try to document info for the developers/contributors and not only for the users.

Koenkk, 2019-10-20:
Thanks for the graphs, but currently due to #1888 there is a very big difference between zigbee2mqtt dev and 1.6. Especially the herdsman graph will look completely different.

wassfila, 2019-10-20:
Nice !
Another reason then for me to focus on the new dev to point out the difference.
Parsing the code and building such a graph does not take me much time (~ 1h), and both svg and graphml are text formats so could be versioned as well.
I actually had some issues installing the dev with compiling the serial port on raspberry pi, but I'll give it another try.
Looking forward for the improvements.

wassfila, 2019-12-19:
idea was to publish thif info and make it visible on a search, I can close this issue now.

closed 2019-12-19 by wassfila