Koenkk/zigbee2mqtt #2174: IKEA LED1732G11

towerhand, 2019-10-20:
# Bug Report
## What happened
I purchased 3 IKEA light bulbs model# LED1732G11 E26 1000lm 91lm/W 11W 114mA CCT TRADFRI WS. I have tried multiple times to get them properly paired and they do pair but keep coming up as not supported. I have restarted zigbee2mqtt and removed the device info from database.db multiple times, always the same result.

According to this they should be supported but I did notice is listed as E27.
https://www.zigbee2mqtt.io/devices/LED1732G11.html

This are the logs during the pairing process.
```
zigbee2mqtt    | zigbee2mqtt:info  2019-10-20T19:01:22: Device '0x000d6ffffe150270' joined
zigbee2mqtt    | zigbee2mqtt:info  2019-10-20T19:01:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x000d6ffffe150270"}}'
zigbee2mqtt    | zigbee2mqtt:info  2019-10-20T19:01:22: Starting interview of '0x000d6ffffe150270'
zigbee2mqtt    | zigbee2mqtt:info  2019-10-20T19:01:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000d6ffffe150270"}}'
zigbee2mqtt    | zigbee2mqtt:debug 2019-10-20T19:01:25: Device '0x000d6ffffe150270' announced itself
zigbee2mqtt    | zigbee2mqtt:debug 2019-10-20T19:01:32: Received Zigbee message from '0x000d6ffffe150270', type 'readResponse', cluster 'genBasic', data '{"modelId":"TRADFRI bulb E26 WS opal 1000lm","manufacturerName":"IKEA of Sweden","powerSource":1}' from endpoint 1 with groupID 0
zigbee2mqtt    | zigbee2mqtt:warn  2019-10-20T19:01:32: Received message from unsupported device with Zigbee model 'TRADFRI bulb E26 WS opal 1000lm'
zigbee2mqtt    | zigbee2mqtt:warn  2019-10-20T19:01:32: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt    | zigbee2mqtt:debug 2019-10-20T19:01:32: Received Zigbee message from '0x000d6ffffe150270', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2,"appVersion":32,"stackVersion":98}' from endpoint 1 with groupID 0
zigbee2mqtt    | zigbee2mqtt:warn  2019-10-20T19:01:32: Received message from unsupported device with Zigbee model 'TRADFRI bulb E26 WS opal 1000lm'
zigbee2mqtt    | zigbee2mqtt:warn  2019-10-20T19:01:32: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt    | zigbee2mqtt:debug 2019-10-20T19:01:33: Received Zigbee message from '0x000d6ffffe150270', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"20190308","swBuildId":"2.0.022"}' from endpoint 1 with groupID 0
zigbee2mqtt    | zigbee2mqtt:warn  2019-10-20T19:01:33: Received message from unsupported device with Zigbee model 'TRADFRI bulb E26 WS opal 1000lm'
zigbee2mqtt    | zigbee2mqtt:warn  2019-10-20T19:01:33: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt    | zigbee2mqtt:info  2019-10-20T19:01:33: Successfully interviewed '0x000d6ffffe150270', device has successfully been paired
zigbee2mqtt    | zigbee2mqtt:warn  2019-10-20T19:01:33: Device '0x000d6ffffe150270' with Zigbee model 'TRADFRI bulb E26 WS opal 1000lm' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
zigbee2mqtt    | zigbee2mqtt:info  2019-10-20T19:01:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x000d6ffffe150270","supported":false}}'
```
## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.6.0 latest-dev in docker compose.
CC253X firmware version: 20190608

towerhand, 2019-10-20:
Tried reading the "how_to_support_new_devices.html" and couldn't make sense of it.

Koenkk, 2019-10-20:
What is the model number of this bulb? You can probably find it on the box and should be similar to something like LED1732G11.

towerhand, 2019-10-20:
Sadly I don’t have the box anymore.

https://www.ikea.com/us/en/p/tradfri-led-bulb-e26-1000-lumen-wireless-dimmable-white-spectrum-opal-70408492/

In the light bulb:
1000lm 91lm/W    LED1732G11    E26
11W 114mA  CCT TRADFRI  WS
Made in China.  21633   1916

Koenkk, 2019-10-20:
Supported in latest dev branch, thanks!

closed 2019-10-20 by Koenkk

towerhand, 2019-10-20:
WOW THANKS for taking care of it so quickly. How long should I wait to try pairing it again? Anything special I need to do to use the latest dev branch? (using zigbee2mqtt version 1.6.0 commit #2076bba)

Sorry for the probably dumb questions.