Koenkk/zigbee2mqtt #2175: Store last_seen in state.json?

airdrummingfool, 2019-10-21:
Working in Docker `latest-dev`.

I've been restarting my z2m instance a lot lately as I add/remove devices and change configuration code, and I noticed that, while all other attributes of a connected device get restored after a restart, "last seen" ends up getting reset to 1970-01-01 (i.e. epoch time 0). It looks like the last_seen attribute isn't stored in `state.json`, while the other attributes are.

If I was using `last_seen` in automations (to warn me about possibly disconnected devices, etc), then restarting z2m could cause a lot of false alerts. This is a small thing (since in normal operation, one wouldn't be restarting z2m much), but I do think storing `last_seen` in `state.json` would be an improvement.

In `state.json`:
```json
    "0x00xxxxxxxxxxxxx": {
        "battery": 100,
        "voltage": 3005,
        "linkquality": 81,
        "water_leak": false
    },
```

In HA after a restart:
```
battery			100
voltage			3005
linkquality		81
water leak		false
last seen		1970-01-01T00:00:00.000Z
```

wassfila, 2019-10-22:
I do see the last_seen in my state.json.
it can be missing for old reports saved before activating the last_seen feature.
I do restart my zigbee2mqtt quite often as well, note that the "retain:true" feature would also have the last_seen stored in the mqtt broker throughout zigbee2mqtt restart or even when zigbee2mqtt is completely shutdown and a client connects to the broker afterwards.

[image]

airdrummingfool, 2019-10-22:
@wassfila are you using a release version or a dev version of zigbee2mqtt?

I don't have `last_seen` entries for any devices - whether they were paired before or after I turned on `last_seen`. All of my devices have reported since I turned it on as well.

wassfila, 2019-10-23:
Yeah weired, I don't know, here's my version log and sxtract of state.json
info: Starting zigbee2mqtt version 1.6.0 (commit #e26ad2a)

        "battery": 100,
        "voltage": 3045,
        "linkquality": 33,
        "water_leak": false,
        "last_seen": "2019-10-23T07:12:03+02:00"
    },

        "battery": 100,
        "voltage": 3025,
        "linkquality": 65,
        "angle": 27.42,
        "from_side": 3,
        "to_side": 1,
        "last_seen": "2019-10-23T06:30:22+02:00",
        "side": 1
    },
        "battery": 100,
        "voltage": 3005,
        "linkquality": 41,
        "last_seen": "2019-10-23T07:07:12+02:00"
    },

airdrummingfool, 2019-10-23:
@wassfila Ah, that's probably it - you're running 1.6.0 stable, while I'm running the newest dev version (which includes some major changes by switching from `zigbee-shepherd` to `zigbee-herdsman`).

@Koenkk `last_seen` not getting stored in `state.json` looks like a change (possibly a regression?) since zigbee2mqtt 1.6.0.

Koenkk, 2019-10-23:
@airdrummingfool I think this can easily be fixed by moving https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/controller.js#L277 before https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/controller.js#L249 and instead of `messagePayload.last_seen =` do `payload.last_seen =`

Koenkk, 2019-10-24:
Merged

PR #2197 (2019-10-24): Store last_seen in state.json
This PR fixes #2175 based on @Koenkk's recommendations. Note that I had to add a null check also, because `entity.device.lastSeen` is `null` when zigbee2mqtt first starts up, before it receives any messages.
closed 2019-10-24 by Koenkk