Koenkk/zigbee2mqtt #2177: Coordinator initialize had an error: Error: SRSP - UTIL - getDeviceInfo after 6000ms

belidzs, 2019-10-21:
# Bug Report
## What happened
* Zigbee2mqtt suddenly stopped connecting to my CC2531 device and thus my zigbee network went down
* The sniffer was blinking red
* After replugging to the USB port the green led returned, but Zigbee2mqtt can't connect to it anymore
* In device manager I can still see the COM port assigned

### Notes
* I have tried on a separate computer which also worked with this setup, same symptoms
* I don't have CCDebugger device, so I don't know if there's a way to reset the chip itself apart from reconnecting to the USB port

## What did you expect to happen
The sniffer shouldn't have started blinking red and then stopped working

## How to reproduce it (minimal and precise)
I can't tell

## Debug Info
zigbee2mqtt version: 1.6.0
CC253X firmware version:  CC2531_DEFAULT_20190608
log output: https://pastebin.com/7h47yRJ5

belidzs, 2019-10-21:
I've upgraded to the dev branch to see if this was a bug and thus maybe it was fixed, but unfortunately not. The startup failed a lot quicker this time.

Here's the console output: https://pastebin.com/EwGFY7w2

ciotlosm, 2019-10-21:
This is pretty similar to https://github.com/danielwelch/hassio-zigbee2mqtt/issues/226

belidzs, 2019-10-21:
Thanks @ciotlosm I'm gonna try reflashing via Arduino (even if it's a pain) and then report back.  

It's worth noting that 1.6.0 was the only version I've ever used before the crash since my whole zigbee network was created just yesterday, so in my case it probably doesn't have to do anything with upgrade rather than a probable bug in 1.6.0

belidzs, 2019-10-21:
So after flashing the stick again with the same exact firmware it is now working again.

I'm not sure if the stick got corrupted because itself, the host or the software however the issue you linked suggests that it has to do with Zigbee2mqtt

Export33, 2019-10-22:
I had this error several times in the last days. Reflashing always helped, but the waf -> 0.

Today the devices wouldn´t reconnect after reflashing:
```null failed with error Error: AF data request fails, status code: 205.```
Disconnecting them from power/ using the real switch helped.

fheinz, 2019-11-03:
I'm also seeing this issue, same versions. I'l admit that the many references to reflashing don't sound very comforting... why would you ever need to reflash except for an upgrade?

Do we have any leads? Any debugging info I might be able to collect?

Koenkk, 2019-11-04:
- How much devices are in your network?
- Do you have the availability or report feature enabled?

belidzs, 2019-11-04:
When the incident happened I had 3 bulbs, a motion sensor and 3 steering devices in my zigbee network. All of them made by IKEA

I had the report feature specifically enabled. I don't know about the availability feature but it wasn't explicitly enabled in the config

Koenkk, 2019-11-04:
I also recommend trying with the latest dev branch as it uses zigbee-herdsman instead of zigbee-shepherd (#1888)

belidzs, 2019-11-04:
@Koenkk after I reflashed the stick I've also upgraded to dev and since then the issue haven't happened again.

Koenkk, 2019-11-04:
Great, then I will close this for now, feel free to re-open once you run into problems again.

closed 2019-11-04 by Koenkk