Koenkk/zigbee2mqtt #2178: Failed to configure OSRAM Smart+ Plug (AC10691)

rsprit, 2019-10-21:
OSRAM Smart+ Plug AC10691 is correctly identified during pairing, but can't be configured
by zigbee2mqtt (neither after pairing nor on zigbee2mqtt startup).

```
zigbee2mqtt:info  2019-10-21T11:41:52: Configuring 'plug-5b3b'
zigbee2mqtt:error 2019-10-21T11:41:52: Failed to configure 'plug-5b3b', attempt 1 (TypeError: Cannot read property 'bind' of undefined
    at bind (/app/node_modules/zigbee-herdsman-converters/devices.js:17:24)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices.js:1823:19)
    at DeviceConfigure.configure (/app/lib/extension/deviceConfigure.js:66:32)
    at DeviceConfigure.onZigbeeStarted (/app/lib/extension/deviceConfigure.js:41:28))
```

Nevertheless the device can be controlled afterwards (on/off).

zigbee2mqtt version: docker image koenkk/zigbee2mqtt:latest-dev
CC253X firmware version: 2.6.3 (20190608)

Koenkk, 2019-10-21:
Can you share your `database.db`?

rsprit, 2019-10-21:
`database.db`: [https://hastebin.com/raw/tamoxoveto](https://hastebin.com/raw/tamoxoveto)

Koenkk, 2019-10-21:
Should be fixed in latest dev.

rsprit, 2019-10-21:
works like a charm, even linkquality is now reported in MQTT.

thx a lot

closed 2019-10-21 by Koenkk

alwashe, 2019-10-30:
How can i install dev?