Koenkk/zigbee2mqtt #2179: the network ZigBee and ICTC-G-1

VOVAiS, 2019-10-21:
# Bug Report
## What happened
after connecting to the network ICTC-G-1
ICTC-G-1 using (left/right rotation) zigbee network does not work.Other ZigBee devices do not respond to commands.
## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.6.0 commit 0527b89
CC253X firmware version: 20190425

ciotlosm, 2019-10-22:
@VOVAiS Can you check if you reached the device limit? Can you enable debugging and post the log?

VOVAiS, 2019-10-22:
log https://pastebin.com/qLCKP8N2
4 router and 2 end device

VOVAiS, 2019-10-22:
when the dimmer is connected to the router (QBKG12LM), the router does not respond and does not respond to all end devices connected to the router. When connecting the dimmer to the coordinator the network works better

VOVAiS, 2019-10-23:
left in networks two lamp LED1650R5 and dimmer ICTC-G-1. lamps bind to dimmer(zigbee2mqtt / bridge / bind). LED1650R5 are controlled by dimmer not simultaneously. they may not react at all
> digraph G { node[shape=record]; "0x00124b001938efc7" [style="bold, filled", fillcolor="#e04e5d", fontcolor="#ffffff", label="{0x00124b001938efc7|0x00124b001938efc7 (0)|Last seen: 00:00:00s ago}"]; "0x00124b001938efc7" -> "0x000d6ffffeff3d06" [penwidth=0.5, weight=0, color="#994444", label="77"] "0x00124b001938efc7" -> "0x000d6ffffeff2e15" [penwidth=0.5, weight=0, color="#994444", label="66"] "0x00158d0002235cb3" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{0x00158d0002235cb3|0x00158d0002235cb3 (51006)failed: lqi|Xiaomi Aqara double key wired wall switch without neutral wire. Doesn't work as a router and doesn't support power meter (QBKG03LM)|Last seen: unknown}"]; "0x00158d0002235cb3" -> "0x00124b001938efc7" [penwidth=0.5, weight=0, color="#994444", label="91"] "0x00158d0002d7a022" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{0x00158d0002d7a022|0x00158d0002d7a022 (16111)failed: lqi|Xiaomi Aqara double key wired wall switch (QBKG12LM)|Last seen: unknown}"]; "0x00158d00024862df" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{0x00158d00024862df|0x00158d00024862df (38647)failed: lqi|Xiaomi Aqara double key wired wall switch (QBKG12LM)|Last seen: unknown}"]; "0x000d6ffffeff3d06" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{0x000d6ffffeff3d06|0x000d6ffffeff3d06 (21346)|IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (LED1650R5)|Last seen: 00:02:12s ago}"]; "0x000d6ffffeff3d06" -> "0x000d6ffffeff2e15" [penwidth=0.5, weight=0, color="#994444", label="129"] "0x000d6ffffeff2e15" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{0x000d6ffffeff2e15|0x000d6ffffeff2e15 (7006)|IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (LED1650R5)|Last seen: 00:02:12s ago}"]; "0x000d6ffffeff2e15" -> "0x00124b001938efc7" [penwidth=0.5, weight=0, color="#994444", label="29"] "0x000d6ffffeff2e15" -> "0x000d6ffffeff3d06" [penwidth=0.5, weight=0, color="#994444", label="130"] "0x000d6ffffe2abca2" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{0x000d6ffffe2abca2|0x000d6ffffe2abca2 (35598)|IKEA TRADFRI wireless dimmer (ICTC-G-1)|Last seen: 00:04:23s ago}"]; "0x000d6ffffe2abca2" -> "0x00124b001938efc7" [penwidth=1, weight=0, color="#994444", label="1"] }

VOVAiS, 2019-10-28:
And can be turn off have ICTC-G-1 publishing messages in MQTT? I use it through the Binding function

VOVAiS, 2019-11-03:
added debounce: 10 for the dimmer. Work better

closed 2019-11-03 by VOVAiS