Koenkk/zigbee2mqtt #2180: Please add new Gledopto device GL-MC-001

gualostly, 2019-10-21:
https://m.aliexpress.com/item/4000061143937.html?trace=wwwdetail2mobilesitedetail&spm=a2g0n.store_allProduct.allProducts_11442818.4000061143937
Their product USB mini RGB+CCT controller appears as model 'GL-MC-001'. It does not appear in the latest devices.js. Controls should be identical to other gledopto RGB+CCT controllers. Could confirm it works after I added the model to the existing zigbeeModel:[] list in shepherd. (I'm not running Dev branch)
Please include this model. Thanks!

Koenkk, 2019-10-21:
Added, will be supported in the dev branch soon. thanks!

closed 2019-10-21 by Koenkk

GleWapTo, 2023-12-20:
Hello,
could you also add the GL-MC-002P as synonym to the GL-MC-001P?
Seems to be pretty much the same controller just different generation.

I’m a beginner and can’t get it to work using an external converter. Thank you!

GLEDOPTO ZIGBEE3.0 USB 5V RGBWW/RGBCCT Mini LED Streifen Controller RF Fernbedienung APP Sprachsteuerung Kompatibel mit hu*e Tuya SmartThings Amazon Echo Plus ZIGBEE2MQQT https://amzn.eu/d/bDx1ei2