Koenkk/zigbee2mqtt #2183: Loss of my zigbee xiaomi sensors (temperature, movement, door)

kwetnico, 2019-10-21:
Hello,
I have a raspberry pi4 with the key zigbeemqtt that works well except the night .. every night I have more reassembled on your zigbee devices under Jeedom.
Would you have a solution? (I have the latest firmware for the month of June on my cc2531 key)

If i reboot my Jeedom software everything works again

Thank you

wassfila, 2019-10-22:
Hi,
Do you have a log file?
Do you use debug mode ?
It could also be a Jeedom problem, as I use zigbee2mqtt alone without Jeedom and I don't have this issue.

kwetnico, 2019-10-22:
Hello another question :

 I tried as stated on the website zigbeemqtt increase the sensitivity of the vibration sensor command line but apparently I have an error someone has already done or could help me?  Thank you
 Info here: https://www.zigbee2mqtt.io/information/ ... or-djt11lm

 My command : mosquitto_pub -t zigbee2mqtt/Capteur_Vibration/set -m '{"sensitivity": "high"}'

Journal :
zigbee2mqtt:info 22/10/2019 à 15:06:06 Zigbee publish to device 'Capteur_Vibration', genBasic - write - [{"attrId":65293,"dataType":32,"attrData":1}] - {"manufSpec":1,"disDefaultRsp":1,"manufCode":4447} - 1
oct. 22 15:06:12 raspberrypi npm[16761]: zigbee2mqtt:error 22/10/2019 à 15:06:12 Zigbee publish to device 'Capteur_Vibration', genBasic - write - [{"attrId":65293,"dataType":32,"attrData":1}] - {"manufSpec":1,"disDefaultRsp":1,"manufCode":4447} - null failed with error Error: AF data request fails, status code: 240. MAC transaction expired.
oct. 22 15:06:12 raspberrypi npm[16761]: zigbee2mqtt:info 22/10/2019 à 15:06:12 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 240. MAC transaction expired.","meta":{"entity":{"ID":"0x00158d000312eb6e","type":"device","friendlyName":"Capteur_Vibration"},"message":"{\"sensitivity\": \"high\"}"}}'

Kryzek, 2019-10-23:
Probably end device is sleeping and end devices can not receive commands if they are sleeping. I would imagine that vibration sensor would wake up to report vibration when it is shaken gently. If not, then maybe there is some button you can press to wake it up to receive this setting. So try shaking device and then issuing your command.

kwetnico, 2019-10-23:
I pressed the button just before sending the order and I got that answer.  the order for you and she good?  thank you.

kwetnico, 2019-10-23:
I confirm that even by pressing to wake him up, when I run the same command error message.
This is the correct method to send the hight parameter? Thank you

wassfila, 2019-10-23:
Hi,
I think as you changed the topic you can edit the name of the ticket to something related to the config setting.
I just tested this config with the same DJT11LM and I admit that they are quite a pain to pair, so make sure the pairing is not lost somehow, check that it is correctly reporting when you move it for example.
Then as suggested in the comments above, you really have to press its tiny button to wake it up right before you send the mqtt config command.
in my case it is same as you local host 
`mosquitto_pub 'zigtest/motion test/set' -m '{"sensitivity":"high"}'`

The first time I tried, I did have an error like you did
`Oct 23 20:09:36 zigtest/bridge/log {"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 240. MAC transaction expired.","meta":{"entity":{"ID":"0x00158d0002a549ff","type":"device","friendlyName":"motion test"},"message":"{\"sensitivity\":\"medium\"}"}}
`

but then it worked out three times in a row, I wake it with the button right before sending the command.
Then actually there is no confirmation back, so no guarantees that it actually worked other than the absence of an error.
I also do no know any way how I can check if the config is really the one I want, it does seem a bit more sensitive when I set it on high compared to low, but I have no exact test for that. So assume it is working.

closed 2019-12-29 by stale[bot]