Koenkk/zigbee2mqtt #2184: HOW: power-on state PHILIPS hue?

dedepene, 2019-10-21:
# Bug Report
## What happened
How do I set the power on behavior on a hue bulb???

Sending { "hue_power_on_behavior": "off" } to zigbee2mqtt/HUE White Ambiance/set does not change the power-on behavior. No errors in the log.

## What did you expect to happen
Once the behavior has changed, I would expect the bulb to stay off when powered up after a power cut or simply flicking the light switch.
## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 0.2.0-dev.153+Branch.master.Sha.4519f4ed0c99909bad9ee9178acc23a2fca411c8
CC2531 firmware version: 20190425

sjorge, 2019-10-21:
You're firmware might be to old, I recently got one of the E14 color/white ambiant bulbs, manual was from 2019 but firmware is from 20170908 1.29.0_r21169, no hue hub to update the firmware.

So check your firmware, it might be to old as wel.

dedepene, 2019-10-21:
> You're firmware might be to old, I recently got one of the E14 color/white ambiant bulbs, manual was from 2019 but firmware is from 20170908 1.29.0_r21169, no hue hub to update the firmware.
> 
> So check your firmware, it might be to old as wel.

Makes sense. How did you check the fw version?

sjorge, 2019-10-21:
It's available in the output when you query all devices using `zigbee2mqtt/bridge/config/devices/get` (https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html)

dedepene, 2019-10-21:
> It's available in the output when you query all devices using `zigbee2mqtt/bridge/config/devices/get` (https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html)

This is all I got:

`"modelID":"LTW001","hardwareVersion":20,"softwareBuildID":"5.50.1.16698","dateCode":"20151116"`

sjorge, 2019-10-21:
So it's a firmware from 20151116 with version  5.50.1.16698, seems older than the one I have(2015) so it's probbaly to old for the poweron feature.

dedepene, 2019-10-21:
dang. no way to update without a hue hub, huh?

sjorge, 2019-10-21:
Not via zigbee2mqtt AFAIK, I think decobee (I can't spell it, the german software for zigbee and zwave devices) can also do updates.

dedepene, 2019-10-22:
@Koenkk, would it be possible at some point to flash hue devices the way they can with deCONZ? - [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/270](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/270)

Koenkk, 2019-10-22:
Maybe, see https://github.com/Koenkk/zigbee-herdsman/issues/60

I will close this for now, thanks @sjorge for helping!

closed 2019-10-22 by Koenkk