Koenkk/zigbee2mqtt #2189: Support of Philips Hue LWA004 - Single Filament Bulb

ovhelemann, 2019-10-22:
Hi, 

at first: I am new to ioBroker and Github so it is still not easy for me to work with it.
Thanks for your understanding.

Today I bought a Philips Hue Single Filament Bulb and tried to integrate it. At first I paired it without any changes to the adapter. I recognized a LWA004, but it is not supported.

I tried to follow the How-To-Add device guide and tried to change the device.js file. I added following under Philips section:

 ```
   {
        zigbeeModel: ['LWA004'],
        model: '551796',
        vendor: 'Philips',
        description: 'Hue Single Filament',
        extend: hue.light_onoff_brightness,
    },
```
As the Bulb supports on/off and dimming, I supposed this would work. But unfortunately my log tells me:

> "LWA004" not described in statesMapping

Is there any step I missed?
I successfully turned it on and off with "send functional command". I did not manage to dimm it. Writing on genLevelCtrl/currentLevel gives me a 

> readOnly {"msg":[{"status":136,"attrId":0}]}

Would be happy to get some help from you guys!

Thanks in advance
OVH

Koenkk, 2019-10-23:
This device is already supported by zigbee-herdsman-converters (https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L1087). Iobroker needs to update to the latest one. Closing this as it is not a zigbee2mqtt issue.

closed 2019-10-23 by Koenkk

ovhelemann, 2019-10-23:
Alright, did not check for new version. My fault
Thanks anyway.