Koenkk/zigbee2mqtt #2190: Cant pair IKEA Tradfri ON/OFF Switch E1743

Zellerfeld, 2019-10-22:
# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

lolorc, 2019-10-23:
I paired 2 of them 2 days ago, I followed the pairing instructions and it worked, but I had to move the switches very near to the coordinator, it didn't worked when the switch was 2 meters away. (but still, my debug logs were showing them trying to pair)

I'm using the dev branch.

Zellerfeld, 2019-10-23:
Thank you for your fast replay. 

I hold the Buttons 2cm away from the coordinator but it didn't worked. Any other tips?

Zellerfeld, 2019-10-24:
OK I tryed this https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html
and try to pair agin, but no chance. 
How exactly do you connect it? Is the link the right way to change to dev version?

Im sorry men. That sounds not good.

lolorc, 2019-10-25:
hi,

what's wrong ? I can't see the issue you're having in this log ?
You might want to upgrade your fw to the source routing one, but I don't really know if this is needed, I'm basing my comments on the other github issues, I'm using a cc2652r on my side

Zellerfeld, 2019-10-25:
Witz this log i wann To Show That there is ne pairing.

lolorc, 2019-10-25:
looks like you've lost the pairing while upgrading, what about trying to pair the e1743 switch ?

fredrikgk, 2019-10-26:
What do you do on the switch to initiate the pairing?

sjorge, 2019-11-02:
I managed to get mine paired after 3 attempts, I have to keep clicking the on/off button to keep the device awake during the interview process. Not had much lock with binding bulbs to it though :(

```
zigbee2mqtt:info  2019-11-02T10:50:13: Successfully bound cluster 'genOnOff' from '0x14b457fffeca351b' to 'bedroom/night_light'
zigbee2mqtt:info  2019-11-02T10:50:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_bind","message":{"from":"0x14b457fffeca351b","to":"bedroom/night_light","cluster":"genOnOff"}}'
zigbee2mqtt:info  2019-11-02T10:50:14: MQTT publish: topic 'zigbee2mqtt/0x14b457fffeca351b', payload '{"linkquality":52,"click":"on"}'
zigbee2mqtt:info  2019-11-02T10:50:15: Successfully bound cluster 'genLevelCtrl' from '0x14b457fffeca351b' to 'bedroom/night_light'
zigbee2mqtt:info  2019-11-02T10:50:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_bind","message":{"from":"0x14b457fffeca351b","to":"bedroom/night_light","cluster":"genLevelCtrl"}}'
```

Seems OK (I unbound the coordinator first) but the bulbs doesn't do anything. Will play with it some more.

sjorge, 2019-11-02:
Good news, I got it to work.

- pair device (keep clicking on/off alternating until interview is complete, might take a few tries)
- create a group with id 99
- add lamps to group 99
< works fine >
- remove lamps from group 99
- unbind group 99
- delete group 99
- create group 123
- add lamp to group 123 (tried both trådfri and hue)
- bind group 123 to remote
< works fine with lamps in group 123 >

lolorc, 2019-11-02:
I also managed to pair with group 99 created, I just unbound switch from 99. (didn't add/remove bulb to/from 99, didn't not remove the group)

Zellerfeld, 2019-11-05:
Thx sjorge. I tryed but nothing is showed in the log. How log did you pressed the pair button?

sjorge, 2019-11-05:
I counted to 4 and then pressed the on or off button. I think the pairing usually fails because the remote goes back to sleep before the interview is complete.

If you spam the button it also does not seem to complete the interview process either.

Also I don't think creating the group 99 is needed when using `dev` but I am not sure, would be nice of @Koenkk could onfirm that.

Koenkk, 2019-11-05:
In the dev branch the remote is not added to a group anymore indeed.

sjorge, 2019-11-05:
Nice, that would also explain my failed unbind's for that group.

Koenkk, 2019-11-05:
Yes (as it is not bound to anything by default). HOWEVER when upgrading from a zigbee2mqtt 1.6 installation you need to do this, zigbee2mqtt 1.6 bound it to the group.

To start fresh once on the dev branch, repair (factory reset) the device to let it forget its bindings.

sjorge, 2019-11-05:
OK, than that mirrors my experience yesterday... I punted a lot on trying to unbind the group that was never bound to begin with.

Zellerfeld, 2019-11-17:
I installed now version 1.7.1 and still not able to pair. I dont know what im doing wrong. I tryed this instruction [https://www.zigbee2mqtt.io/devices/E1743.html](url) and try many other paring methodes, but nothing seams to work.

Zellerfeld, 2019-11-17:
OK it works now. I dont know how but after a couple of restarts i can pair it.
1. 4 times press the connect button
2. wait until the lights stop flashing
3. light start glowing
4. wait until the light stopo glowing

now they are paired

closed 2019-11-17 by Zellerfeld

Alestrix, 2019-12-29:
Hi @sjorge 

> * unbind group 99

How did you unbind a group? I can see in the docs how to add/remove devices to/from a group and how to bind/unbind one device to another, but how do you unbind from a group?

I can't get the Tradfri switch to work. It pairs fine (provided "interview successful" means that pairing was successful), but I cannot see any state updates when pressing on/off. I do not intent to bind it to some other device or group, only to use it as a switch in my Home Automation setup.

Regards
 Alex

sjorge, 2019-12-29:
I create a group with ID 99, without devices.

Then I used the binding docs to unbind the group from the remote.

~ sjorge

Alestrix, 2019-12-29:
> I create a group with ID 99, without devices. Then I used the binding docs to unbind the group from the remote.

Thanks for the info @sjorge , helped me to get slightly farther (didn't fail with "unknown group" this time), but it failed to unbind the switch from the group and I still cannot see switch state changes in the logs or MQTT. I guess I'll have to open another issue.

Alestrix, 2019-12-30:
Just fyi, after upgrading the controller firmware to the current version, everything works for me.